tobiasfaust/SolaxModbusGateway #3: Wiki link not found

uham999, 2023-01-12:
The wiring wiki link page https://github.com/tobiasfaust/SolaxModbusGateway/blob/master/wiring-the-circuit comes up with a 404 error.
I was checking it because the last time I looked, you had changed added information about the GND pin which was useful. The pin number was correct for the Solax X1 Hybrid but the ground pin is different for other inverters. It is important to point this out for obvious reasons.

tobiasfaust, 2023-01-13:
Where did you find this link?
In navigation bar the link works well:
https://github.com/tobiasfaust/SolaxModbusGateway/wiki/wiring-the-circuit

i specified the solax connection a little bit clearer at this page

uham999, 2023-01-13:
I see it on the wiki page. That looks good. The link error is on is on
https://github.com/tobiasfaust/SolaxModbusGateway 'How to wire the circuit'
readme.md.

I'm just trying to set up an RS485 link to my Growatt SPF3000 to test.
The battery uses the RS485 port so to make it work with an additional RS485
device, it looks like I will need to make a special cable. This will need
to connect an additional RJ45 plug in series with the existing RJ45 battery
connection cable to effectively make a field bus. Do you know a better way?
Steve

tobiasfaust, 2023-01-13:
I think they are using different wires from RJ45. Please check the documentation.

commit 3f214c8a (2023-01-13): fix #3
closed 2023-01-13 by tobiasfaust