tobiasfaust/SolaxModbusGateway #4: Document truncated and writing to registers

uham999, 2023-01-18:
I noticed that the document called Solax-Hybrid-X1X3-G4-ModbusTCPRTU-V3.21-English_0622-public-version.pdf
in the docs directory is truncated.

I have been using register definitions in Solax-Hybrid X1&X3-G3 ModbusTCP&RTU V3.21- English.pdf to write to registers using simple test code for function 0x06 for single register and multiple registers using function code 0x10 and it works.
For some time, it kept returning a slave fault response abnormal code 4 and I suspected it was some kind of security issue. I haven't found anyone at Solax who really knows anything about any of this. I then asked my inverter installer engineer for the administration 4 digit code that needs entering into the front panel on the inverter so I could look at the advanced settings not usually available to a user. I hoped that this might allow me to write data. Once I was able to access advanced settings on the inverter control panel, the code for writing data to registers started to work without returning an error.
I had not changed the test code so think you might need the advanced settings 4 digit code number that a normal user would not necessarily have. I haven't seen an error since and can't reproduce the errors now.
I'm not 100% confident off my findings but thought that you may have experienced the same problem if you tried to write data to registers using function code 0x06 or 0x10? Have you tried it yet?

tobiasfaust, 2023-01-22:
No, i didn´t try it yet. I implemented the basics to do it for the first testing purpose by anyone

steffenk1981, 2023-02-05:
Hi, 
Can someone give me an example of how to write the values here?

"set": [
{
"name": "BatSOC",   --> MQTT Topic?
"request" : ["#ClientID", "0x06", "0x00", "0x83"]
"request" : ["#ClientID", **[write command]**, **[position]**, **[value]**]   --> is this correct?
}
		]

Thanks

tobiasfaust, 2023-02-08:
MQTTBasePath/MQTTRoot/set/command

command = name in json

"request" : ["#ClientID", [write command], [position High], [Position Low]] 

example to set BatSOC to 55%:
home/solax/set/BatSOC 55

please note: this feature is untestet. Please set debuglevel at least to 4 and monitor serial output

tobiasfaust, 2024-02-07:
closed due inactivity

closed 2024-02-07 by tobiasfaust