tobiasfaust/SolaxModbusGateway #5: Changing the baud rate

GKXTRONIC, 2023-01-20:
Hello,
You have realized a great project!
Unfortunately, changes in the Modbus settings are not accepted!
I have tested the recommended RS485 adapter with a Modbus logger software and the hardware works with 9600 baud!
Unfortunately I'm not a programmer. I hope the baud rate can be changed in a future version.

tobiasfaust, 2023-01-20:
Hi,

i just checkd this but a cannot see any problems.
Changing the baudrate to 9600 in WebUI, the serial modbus is going to reinitialize.

You will see that in serial console after changing it:
`Start Hardwareserial 1 on RX (4), TX(15), RTS(5)
Init Modbus to Client 0x01 with 9600 Baud`

Did you try a reboot after that change? By default it´s not necasssary

GKXTRONIC, 2023-01-20:
Hi,

the change is displayed in the WEB front end. But the baud rate is always the same, no matter what I set." If I look at the TX data on the oscilloscope, this is confirmed. I also restarted and the change is displayed in the monitoring.

With the "RMMS-Modbus-Simulator" I can read out my SOLAX-X3 correctly, but with 9600 baud because of the cable length.

For some reason the baud rate stays constant...!?
I'm using an ESP32-WROOM-32, which should be fine...

tobiasfaust, 2023-01-20:
Hi
Ok, the change of baud Rate is only affected After ESP restart? It is correct?
Btw: My Solax x1 works with 19200 baud sucessfully.

GKXTRONIC, 2023-01-20:
No, the baud rate is not changed!

Here 2 oscilloscope screenshots, with 9600 and 19200 baud, both times the same section. After changing the baud rate, a reset was performed!

[image]

[image]

tobiasfaust, 2023-01-20:
Ok, i will Check this next days.
But i dont believe that there are any Code Problems because i know some Users with a growatt Inverter which Accept only 9600 baud.

GKXTRONIC, 2023-01-20:
Then I don't understand it either, since you can't do anything wrong when setting the baud rate.

Of course it would be nice if you could find a solution!

In any case, thanks for your support!

tobiasfaust, 2023-01-20:
Just an idea: did you try to Change the Pins? Please try the the Original serial1 Pins. 
Another idea, you can try to change the serial1 to serial2 in Modbus.cpp at line 47 and Setup the Original serial2 Pins of your ESP32

GKXTRONIC, 2023-01-21:
That is the solution! I had used the alternative UART!
Now it works, thanks for your help...!!!

tobiasfaust, 2023-01-21:
Hi
What was your exactly Solution? What did you changed? Serial1 to Serial2 and nothing Else?

GKXTRONIC, 2023-01-21:
I am now using D16/RX2 and D17/TX2 and the communication works fine, even at 19200 baud.

I'm excited about the project! Unfortunately, the register positions for SOLAX X3-G4 are not all correct (see pictures).

Is it possible to create a local firmware with the corrected file "register.h" with "GITPOD"...?

[image]

[image]

tobiasfaust, 2023-01-21:
Did you also changed the serial1 to serial2 in Code?

a local Firmware is possible with gitpod. You Need it for Testing purposes.
After you finished your corrections of Register, i want your Register.h please to add all correct items of your Solax x3 in my repo please :)

GKXTRONIC, 2023-01-21:
I haven't changed the code! I had previously selected UART 1!

It will take a while and when I have corrected the registers I will of course make them available to you! ;-)

GKXTRONIC, 2023-01-21:
Attached is the modified file "register.h"
https://github.com/tobiasfaust/SolaxModbusGateway/blob/6b97af58fef73bd2062c7f2c0b8a9397789e04fe/src/register.h#L403

Unfortunately, the inverter serial number is not displayed either...!?

[image]

tobiasfaust, 2023-01-21:
Oh, the Link doesn‘t work

GKXTRONIC, 2023-01-22:
Hi,

 

I hope you get the file this way...!?

tobiasfaust, 2023-01-22:
Hi,
its sufficient to post the solax-x3 json here directly. Or, a little bit more efforts,  to create a pullrequest against my repo

GKXTRONIC, 2023-01-22:
Sorry, I'm not very familiar with Github!
Here is the content of "register.h" regarding SOLAX-X3.
I tested the data using the raw-data tool, but not all of them because I'm using an off-grid system...

"Solax-X3": {
		"config": {
			"RequestLiveData": [
				["#ClientID", "0x04", "0x00", "0x00", "0x00", "0x77"],
				["#ClientID", "0x04", "0x00", "0x78", "0x00", "0x77"]
			],
			"RequestIdData": ["#ClientID", "0x03", "0x00", "0x00", "0x00", "0x14"],
			"ClientIdPos": 0,
			"LiveDataFunctionCodePos": 1,
			"LiveDataFunctionCode": "0x04",
			"IDDataFunctionCodePos": 1,
			"IdDataFunctionCode": "0x03",
			"LiveDataStartsAtPos": 3,
			"IdDataStartsAtPos": 3,
			"LiveDataErrorPos": 1,
			"LiveDataErrorCode": "0x84",
			"IdDataErrorPos": 1,
			"IdDataErrorCode": "0x83",
			"LiveDataSuccessPos": 1,
			"LiveDataSuccessCode": "0x04",
			"IdDataSuccessPos": 1,
			"IdDataSuccessCode": "0x03"
		},
		"data": {
			"livedata": [{
					"position": [215, 216],
					"name": "GridVoltage",
					"realname": "Grid Voltage",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [217, 218],
					"name": "GridCurrent",
					"realname": "Grid Current",
					"datatype": "float",
					"factor": 0.1,
					"unit": "A"
				},
				{
					"position": [219, 220],
					"name": "GridPower",
					"realname": "Grid Power",
					"datatype": "integer",
					"unit": "W"
				},
				{
					"position": [9, 10],
					"name": "PvVoltage1",
					"realname": "Pv Voltage 1",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [11, 12],
					"name": "PvVoltage2",
					"realname": "Pv Voltage 2",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [13, 14],
					"name": "PvCurrent1",
[... 164 lines cut ...]
					"unit": "h"
				}
			],
			"id": [
					{
					"position": [3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16],
					"name": "InverterSN",
					"realname": "Inverter SerialNumber",
					"datatype": "string"
				},
				{
					"position": [17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30],
					"name": "FactoryName",
					"realname": "Factory Name",
					"datatype": "string"
				},
				{
					"position": [31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42],
					"name": "ModuleName",
					"realname": "Module Name",
					"datatype": "string"
				}
			]
		}
	},

tobiasfaust, 2023-01-22:
Thanks fur your support.
Do you know what is the difference between GridVoltage_S, GridVoltage_R, GridVoltage_T of Solax-X3?
Which did you take?

GKXTRONIC, 2023-01-22:
...is my pleasure to make a small contribution... :)

R/S/T is an old designation for L1/L2/L3 in the three-phase network!
SOLAX-X3 is a three-phase system and accordingly the measured values for each phase can be read out, which I don't use, however.

[image]

GKXTRONIC, 2023-01-22:
Of course it would be nice if you could select the "R/S/T data" for ON-GRID & OFF-GRID in a future version...!

tobiasfaust, 2023-01-22:
Thanks for that.
Maybe, it would help if you can post the complete rawdata of live and ID Data 

then i‘m able to Check your items and Maybe some of unused items.

tobiasfaust, 2023-01-22:
Are you sure "OnGridRunTime" and "OffGridRunTime" are correct.  I calculated the positions from doc and i got position from 280.
I added your configuration plus lots more items into my repo. 
Can you test the current repo Solax-X3 configuration? 
In case of mismatches, i need your rawdata String ;)

GKXTRONIC, 2023-01-22:
On-gridRunTime --> Adr-0x0088 --> first Data-pos --> 0x0088 --> Dec 136*2+3 = 275...!?

I'm sure I'll test the new configuration!

I can offer this log-data of my SOLAX-X3...

[image]
[image]
[image]
[image]

[image]
[image]
[image]

GKXTRONIC, 2023-01-22:
Looks good!

[image]

96 / 5.000
Übersetzungsergebnisse
Übersetzung
For some reason the battery capacity is "0"...!?
In the RAW-DATA the value is correct!
[image]

GKXTRONIC, 2023-01-22:
Address 280 is correct...!!!
Unfortunately I don't have much time to test... :-(

GKXTRONIC, 2023-01-22:
This is correkt!
[image]

GKXTRONIC, 2023-01-22:
My The formula for calculating the data positions is wrong...!

GKXTRONIC, 2023-01-22:
Now I'm confused!
According to the following log, the address 279-282 for the Off-GridRunTime is correct after all...!?!

[image]

GKXTRONIC, 2023-01-22:
So "offset" *2+1 = address...!
Other values can be verified correctly with it...

tobiasfaust, 2023-01-22:
> On-gridRunTime --> Adr-0x0088 --> first Data-pos --> 0x0088 --> Dec 136*2+3 = 275...!?

i think no. With the First request you request address 0-77, that are 242 positions
So you have to add After Position 242  2 Bytes of CRC and 3 Bytes of the Header of 2nd request, 

where is our mistake? I‘m Confused :(

tobiasfaust, 2023-01-24:
Solax X3 items added by 47ba83b

tobiasfaust, 2023-02-02:
Hi
A Last question to your int32 items: are you sure that the Order of These Registers are correct? In documentation Starts with LSB and After that MSB.

so instead of Register [275, 276, 277, 278] for OnGridRunTime it should be [277, 278, 275, 276]

can you Check this with your X3?

GKXTRONIC, 2023-02-03:
Hi,

I created a small STM32 project to read the data from the X3. I can read all registers correctly with it.

You are right, the sequence is first "LSB", then "MSB", for the ONGRID or OFFGRID info...

Here is an example from my C code for the "OFFGRID time" conversion...

[image]

[image]

tobiasfaust, 2023-02-06:
Thanks, that Shows me that [277, 278, 275, 276] is correct :)
I will correct them in X3 items :)

closed 2023-02-06 by tobiasfaust