tobiasfaust/SolaxModbusGateway #6: ESP Gateway loses ModBus connection to Growatt inverter

saschoctm, 2023-01-27:
The Gateway repeatedly loses connection to the Growatt SPH inverter. So far about every 7 days. The problem exists with 3 different installations, all with Growatt SPH inverters.

In case the connection is broken:
- the TX/RX Leds of the MAX485 Module TTL Switch continue to work as if sending and receiving
e.g. if the baud rate setting is incorrect, no RX LED lights up

- RS485 icon in the display is off, normally the growatt shows an icon in the display with an active RS485 connection,

- A hardware reset fixes the problem

tobiasfaust, 2023-01-27:
Hi,
Please use at First the real RX/TX Pins 16 and 17.
The past Shows us, that other pins could raise wierd   Problems.
If that to helps, please use another ESP32.

saschoctm, 2023-01-27:
Pins 16 and 17 are used.

tobiasfaust, 2023-01-27:
unfortunally, i cannot confirm your behavior. At other growatt inverters, no problems occurs. At my Solax, the ESP works since last update 23 days ago with no failures.

I case of failure, can you connect to WebUI? The ESP lives? GND is connected to your Inverter?
Maybe you can monitor your ESP at seriell console.
And change your ESP32 to another one.

saschoctm, 2023-01-27:
I have no GND connected, is this nesseccary? Why?
Next time when loosing connection I will try to monitor and check if WebUI is still alive.
I think so that ESP is still alive because Rx/Tx is still working, otherwise LED would be dead too or?
I ordered already another ESP and will check if that helps.

tobiasfaust, 2023-01-28:
like described in my wiki, a missing ground connection "can" (!) raise some weird failures.
Ok, if RX/TX are blinking so ESP isn´t dead ;)

You can try some options:
* another ESP
* very short cable connection Inverter <-> ESP for a testing purpose
* if you are using a 5V RS-485 TTL module, try a 3.3V module

closed 2023-02-28 by tobiasfaust