tobiasfaust/SolaxModbusGateway #7: SolaX Mini implemented?

AlexPoeltl, 2023-01-29:
Does anyone has implemented the SolaX Mini inverters?
I have an old one (without display, only LEDs) and a newer one with the status display.
And... What has to be configured in the menu to communicate with the ESP? Does the SolaXModbusGateway simulates a meter?
This is the only thing I could enter in the menu.
I would like to implement these inverters in my Venus OS (Victron Energy) as AC inverters.

tobiasfaust, 2023-01-29:
Hi,
it is not implemented yet but basically possible if you want it.
The first test is to use the testscript and adapt the request string according your X1 mini modbus protokoll.
https://github.com/tobiasfaust/SolaxModbusGateway/tree/master/examples/test_RS485_Solax_ESP32

Modbus protokol is here: https://raw.githubusercontent.com/syssi/esphome-modbus-solax-x1/main/docs/SolaxPower%20Single%20Phase%20External%20Communication%20Protocol%20-%20X1%20Series%20V1.8.pdf

After it works successfully (you get an valid answer to your request) its easy to add it.

Alternative solutions for you are there:
* https://tasmota.github.io/docs/SolaX-X1/
* https://github.com/syssi/esphome-modbus-solax-x1

tobiasfaust, 2023-02-18:
still interested? 
i would wish: yes, but i need your support.
If not, i will close this issue

closed 2023-02-28 by tobiasfaust

MagicSven81, 2023-07-30:
Hallo Tobias,
da ich jetzt einen X1 Mini habe würde ich dir gerne dabei helfen, diesen zu adaptieren.
Was genau muss ich mit dem Testscript machen? Auf dem ESP32 läuft Release 3.1.1
Das Testscript liegt im Format .ino vor. Damit kann ich so leider aktuell nichts anfangen bzw. weiß nicht was damit zu tun ist.
Wenn du mir etwas hilfst würde ich dir die Zuordnung erstellen.
Gruß Sven