tobiasfaust/SolaxModbusGateway #12: X3 Mix

gerdluthe, 2023-04-07:
Hallo Tobias,

vielen Dank für dieses geniale Tool.
Ich habe neben einem X1 noch einen X3-Mic am Start, leider lassen sich mit Deinem Tool dort keine schlüssigen Daten auslesen.
Siehst Du eine Chance das zu implementieren damit man die Daten auslesen kann?

Danke

tobiasfaust, 2023-04-08:
Hi,
Ich denke mal, grundsätzlich ja wenn er einen Modbus RTU, RS-485, Port hat. 
Ich brauche das Modbus Protokoll als PDF auch noch und dann musst du viel testen bis es klappt :)

passt eventuell das X1 Mini Modbus Protokoll zu deinem? Das habe ich schon im doc Ordner

gerdluthe, 2023-04-08:
Also einen RS485 Port hat er, da hängt das MOdul dran.
Ich habe beide Protokolle die Du in Deinem Programm anbietest ausprobiert, X1 und X3 - leider passen die ausgelesenen Werte nicht.
Ich kann Dir die RAW Daten zu Verfügung stellen, wie man diese interpretiert weiß ich leider nicht einzuschätzen.

Typ | Raw Data
-- | --
RawData of ID-Data |  
RawData of Live-Data | 0x01 0x04 0xA6 0x08 0xF4 0x00 0x0D 0x01 0x0D 0x0A0x26 0x00 0x00 0x00 0x0A 0x00 0x00 0x13 0x8B 0x000x1C 0x00 0x02 0x01 0x1B 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xE6 0x000x00 0x00 0x46 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x03 0x00 0x00 0x09 0x17 0x040x56 0x01 0x04 0xEE 0x00 0x06 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x00 0x20 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x82 0x78 0x00 0x00 0x00 0x000x0A 0x05 0x00 0x1B 0x00 0x00 0x00 0x00 0x05 0x3A0xFE 0x79 0xFC 0x49 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x00 0x09 0x43 0x02 0x90 0x00 0x440x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x82 0x780x00 0x00 0x04 0x09 0x04 0x17 0x03 0xFB 0x00 0x000x82 0x78 0x00 0x00 0x00 0xEF 0x00 0x06 0x1C 0x1C0x00 0xFA 0x00 0x00 0x00 0x26 0x00 0x00 0x06 0xC10x00 0x00 0x00 0x02 0x00 0x00 0x00 0x09 0x00 0x000x00 0x57 0x00 0x00 0x01 0x5D 0x00 0x00Position: 287Integer: 0String: 0x00 0x6E0x00 0x00 0x01 0xB9 0x00 0x00 0x00 0x6F 0x00 0x000x07 0x5F 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x09 0x2F 0x00 0x00 0x00 0x00 0x00 0x00 0x09 0x1A0x00 0x00 0x00 0x00 0x00 0x00 0x09 0x0B 0x00 0x000x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x82 0x00 0x00 0x00 0x00 0x00 0x1B 0x0A 0x050xFB 0x00 0x01 0x20 0x00 0xFA 0x04 0xE2 0x11 0xBC0x00 0x00 0x00 0x18 0x00 0x03 0x00 0x64 0x0B 0x180x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x000x00 0x00 0x00 0x02 0x00 0x00 0x00 0x00 0x00 0x000x0B 0x18 0x09 0x38 0x0C 0x95 0x0C 0x7D 0x04 0x2F0x05 0x49 0x00 0xEF 0x00 0xDA 0x01 0x04 0x05 0x110x00 0x50 0xA4 0x20

gerdluthe, 2023-04-08:
Die RAW Daten sind falsch habe ich gerade gesehen. Wenn ich einen ESP32 flashe ohne einen RS485 Baustein daran zu löten, dann sind die RAW Daten identisch zu dem anderen ESP32 den ich am Wechselrichter angeschlossen habe.

Da gibt es wohl noch ein anderes Problem.... !?

gerdluthe, 2023-04-08:
Habe es mit beiden Examples aus Deinem Github probiert, er sendet die Daten aber es kommt keine Antwort

Genaue Bezeichnung des WR ist Solax X3-MIC/PRO-G2

tobiasfaust, 2023-04-08:
Die examples kann man nicht so 1:1 verwenden. Die sind zum Testen gedacht indem man den sende String im Code immer anpasst, neu kompiliert, auf den ESP brennt und schaut was raus dann als Antwort vom WR kommt

wichtig ist aber zuerst, das die eingestellten Daten im Code mit den eingestellten Daten im WR übereinstimmen:
Modbus-ID und Baudrate

gerdluthe, 2023-04-08:
Nein, das können keine plausiblen Daten gewesen sein, wie gesagt ich flashe deine 3 Dateien auf einen ESP32, verbinde den mit meinem Wlan und sehe sofort die Daten die angeblich vom WR kommen, ohne das eine RS485 Schnittstelle angeschlossen ist.

Aber, das habe ich hier gefunden zum X3 Mic (kein Hybrid!!!) 
Das Protokoll ist wohl komplett anders, jedenfalls wird es dort so beschrieben...
vielleicht kannst Du damit etwas anfangen und das in Dein Programm mit implementieren !?

[
](https://github.com/InfernoEmbedded/PowerScraper/pull/16/commits/0b711bf74d4047fbf29464858aed2f03c74a0797)

gerdluthe, 2023-04-08:
https://github.com/InfernoEmbedded/PowerScraper/pull/16/commits/0b711bf74d4047fbf29464858aed2f03c74a0797

tobiasfaust, 2023-04-09:
Welche Dateien hast du denn zum flashen genommen? Aus dem Release oder frisch gebaut mit gitpod?
Es gibt einen debugmode den ich benutze zum Testen. Den kann man in der Modbus.h anschalten.

im repo Code ist er deaktiviert, wenn du über gitpod gebaut hast sollte also nichts kommen.
Bei den files aus dem Release kann es sein das ich es vergessen hatte abzuschalten. 

Also keine Magie sondern einfach nur Bequemlichkeit von meiner Seite damit ich ohne WR testen kann und nicht in der Garage stehen muss die ganze Zeit ;) ;)

Ich schau es mir mal an, grundsätzlich kann man alles was RS-485 kann anbinden

gerdluthe, 2023-04-09:
Ich habe Deine Release Dateien genommen, mit GitPod compilierte Dateien laufen überhaupt nicht, da meldet sich das Modul nicht als Solax Accesspoint

tobiasfaust, 2023-04-09:
Achso, wichtig ist das auf der RAW Seite irgendetwas steht. Erst wenn man eine Antwort vom WR hat kann man sich ans auswerten und Register zuordnen machen.

deswegen ist es wichtig zuerst den Request-String korrekt zusammen zu bauen um eine valide Anwort zu bekommen. Da kann man gut das example Programm benutzen, zusammen mit der Modbus Doku des WR

tobiasfaust, 2023-04-10:
Ich habe mal solax angeschrieben und das Modbus protokoll bekommen. 
[https://github.com/tobiasfaust/SolaxModbusGateway/tree/master/docs](https://github.com/tobiasfaust/SolaxModbusGateway/blob/master/docs/SolaX%20Power%20MIC-G2%20three%20phase%20inverter%20Modbus-RTU%20comms%20protocol%20V1.9.pdf)
Dort stehen die Modbus parameter und die Register. Der Solax Mic arbeitet wohl nur mit 9600 Baud. Die müssen erstmal eingestellt sein. 
Auf Seite 14 steht ein BEispiel für die Read Input Register (Live Daten).  Dieser ist identisch aufgebaut zu den anderen Solax Protokollen die ich bisher habe. 
Das bedeutet, mit der Einstellung "Solax X3" oder "Solax-X1" musst du vom MIC RAW Daten erhalten wenn du die Modbus parameter korrekt eingestellt und alles korrekt verkabelt hast ;)

happy testing

gerdluthe, 2023-04-10:
Das ist ja schon mal eine wichtige INfo, werde die Baudrate anpassen und es nochmal versuchen. Vielen Dank für die Mühe, melde mich ob ich erfolgreich war oder nicht

gerdluthe, 2023-04-10:
Will nicht gelingen, aber auf dem RS485 Modul blinkt eigentlich auch nur die TX LED, die RX ist off. Von meinen anderen Modulen kenne ich das, das dort ein reger Austausch stattfindet. Hattest Du ja auch korrekterweise in Deinem ReadMe beschrieben. 
Hab jetzt einen ESP32 Wroom und einen Wemos ESP32 mini mit jeweils einem HW-0519 zusammengebaut, das ist eigentlich auch eine super Kombi die mit meinem Deye Wechselrichter oder auch mit meinem Seplos BMS wunderbar zusammenarbeitet.

Ich schaue morgen mal in der Firma ob ich noch einen Node MCU 32 rumliegen habe, nicht das es daran liegt...
Sonst muss ich die Daten mit Curl am Stick abgreifen und mir einen Script basteln der die Daten weiterverarbeitet, in der Trockenübung via Terminal bekomme ich plausible Daten und mit NodeRed habe ich die auch schon an meinen ioBroker weiterleiten können. 
Schöner wäre es allerdings mit einem ESP32

tobiasfaust, 2023-04-10:
Wenn nur eine LED blinkt passen die Modbus Einstellungen nicht.

gerdluthe, 2023-04-10:
Ich hab keine Ideen mehr, bereits alle möglichen Einstellungen am Wechselrichter und am Modul probiert. Kein Erfolg, das einzige was halt noch sein könnte, das es kein ModbusRTU ist sondern irgendwas eigenes von solax nur für die Kommunikation mit einem Smartmeter. Im Wechselrichter kann ich zwar Comm485 auswählen, aber da es partout nicht klappen will kommen mir Zweifel am modbus des WR an sich auf.

gerdluthe, 2023-04-10:
A = blau
B = blau / weiß
Modbusadresse im WR 1 und im Modul 1
Baud = 9600, im WR nicht einstellbar !

TX und RX blinken anfangs und nach einer Weile <30s blinkt nur noch TX

Das war's, mehr passiert nicht

tobiasfaust, 2023-04-10:
Wie gesagt, teste es am anderen WR der ja wunderbar funktioniert:)

gerdluthe, 2023-04-12:
Kurze Rückfrage, Dein Programm fragt die Read input register 0x400 und folgende ab?
Darüber kann ich auf jeden Fall mit mbpoll oder modbus über die cli Werte abfragen, mit einem entsprechenden USB2RS485 Adapter

tobiasfaust, 2023-04-12:
Nein,
Es fragt das ab was schlussendlich in der Register.h eingestellt ist. Und das wiederum hängt vom Modbus Protokoll ab, was im PDF steht

gerdluthe, 2023-04-13:
Kannst Du schliessen, ich hab die syssi Variante für den ESP32 benutzt, damit klappt die Kommunikation ohne Probleme. Warum ich das mit Deinem nicht realisieren konnte.... keine Ahnung.

Aber herzlichen Dank für den Support !!!

tobiasfaust, 2023-04-14:
Aber syssi unterstützt doch den Solax Mini und nicht den Solax MIC??
Der Mini hat ein ganz anderes Modbus Protokoll als der Mic….

ich bin verwirrt, welchen Wechselrichter hast du nun?

gerdluthe, 2023-04-14:
Ich habe den Solax X3-MIC G2 und der wird mit folgendem Script unterstützt:
https://github.com/syssi/esphome-solax-x1-mini/blob/main/modbus-examples/esp32-solax-x1-boost.yaml

Da werden auch die Register 0x400 folgende abgefragt, das sind ja die die laut Anleitung von Solax die für mich wichtigen Werte rausschmeissen, nach denen hatte ich Dich ja ein paar Posts zuvor gefragt.

Das Script ist für den X1 Boost, geht aber halt auch für meinen X3 
:-)

tobiasfaust, 2023-04-14:
Find ich merkwürdig… syssi benutzt das proprietäre Mini Protokoll welches mit 0x0A 0x55 startet. Dann muss dein boost das ebenfalls benutzen.

ich hatte mir extra das Protokoll für den MIC G2 von Solax schicken lassen, da steht das das normale Protokoll verwendet wird…. Sehr merkwürdig 🤨 

egal, bei mir müsste die Konfiguration für das Mini Protokoll erst noch konfiguriert werden, dann würde es auch laufen.

wenn es bei dir aber jetzt so läuft dann ist es ja auch super :)

gerdluthe, 2023-04-14:
Nun, ich bin kein Modbus Experte, aber mit dem ioBroker Modbus Adapter konnte ich den Inverter auch einfach über die Input Register auslesen. Ohne das ich dem Inverter vorher eine Zeichenkette senden musste damit er die Daten ausspuckt.

Dazu habe ich einfach einen USB>RS485 Adapter verwendet.

tobiasfaust, 2023-04-14:
Ah, ok.
Man hätte im example auch mal folgendes probieren können:

byte request[] = {0x01,
                               0x04,
                               0x04,
                               0x00,
                               0x00,
                               0x07, 
                               0x00,
                               0x00
           };

bedeutet, er liest von Client Modbus id 1 das ReadInputRegister 0x04 von Stelle 0x0400 die nächsten 7 Register

tobiasfaust, 2023-04-14:
Also wenn du noch Lust hast wäre ich interessiert. 
wurmt mich selbst warum es bei dir nicht klappt….

gerdluthe, 2023-04-14:
Alles klar, melde mich dann am Wochenende mit Ergebnissen zurück ;-)

gerdluthe, 2023-04-14:
Sieht schon besser aus :-)

Lese Daten ....
1 4 E 9 62 6 82 0 36 0 4E 9 5E 9 5D 9 5C 64 68 
Schreibe Daten ....
0x01 0x04 0x04 0x00 0x00 0x07 0xB0 0xF8 
Lese Daten ....
1 4 E 9 5E 6 81 0 36 0 4D 9 5D 9 5B 9 5B ED D3 
Schreibe Daten ....
0x01 0x04 0x04 0x00 0x00 0x07 0xB0 0xF8 
Lese Daten ....
1 4 E 9 61 6 77 0 36 0 4E 9 5E 9 5C 9 5B 63 66 
Schreibe Daten ....
0x01 0x04 0x04 0x00 0x00 0x07 0xB0 0xF8

tobiasfaust, 2023-04-14:
Haaa, super!!!
Damit wirds klarer.... Kannst du mal bitte die register.h ersetzen? Ich habe eine neue sektion für den MIC eingebaut. 
Im json an Position RequestLiveData frage ich von 0x400 genau 40 Register ab. Habe erstmal nur die ersten 2 Items eingebaut. Wichtig wäre das in RAW Data die 80 Ergebniss-Bytes der abgefragten 40 Register auftauchen.

[register.zip](https://github.com/tobiasfaust/SolaxModbusGateway/files/11235730/register.zip)

gerdluthe, 2023-04-14:
Gute Arbeit, jetzt kommen Daten auf der Weboberfläche an, kannst also alle anderen Datenfelder auch noch implementieren.

tobiasfaust, 2023-04-15:
Perfekt…. Mach doch mal bitte von der RAW Seite ein Screenshot und poste die beiden (id Data und live Data) langen hex Strangs hier

dann habe ich echte Daten anhand derer ich die restlichen Items anlegen kann.

btw: du hast jetzt alles auf esphome umgestellt damit der syssi Code laufen kann?

gerdluthe, 2023-04-15:
Hi hier die Daten:

ID Data:
0x01 0x03 0x28 0x4D 0x43 0x32 0x31 0x30 0x54 0x49
0x41 0x30 0x38 0x39 0x30 0x34 0x36 0x73 0x6F 0x6C
0x61 0x78 0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x20
0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x20
0x20 0x20 0x20 0x44 0xCE

LiveData:
0x01 0x04 0x80 0x09 0x4B 0x06 0x63 0x00 0x02 0x00
0x04 0x09 0x35 0x09 0x42 0x09 0x39 0x13 0x8A 0x13
0x8F 0x13 0x8B 0x00 0x04 0x00 0x05 0x00 0x08 0x00
0x19 0x00 0x54 0x00 0x02 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x41 0x00 0x44 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x10 0xD5 0x00 0x00 0x00 0x03 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x19 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x02 0x0E

Umgestellt habe ich nichts auf esphome, ich nutze den ioBroker und brauche die Daten via MQTT von daher ist es egal ob das ein esphome Node macht oder z.B. Dein Programm. Hauptsache eine MQTT Anbindung
Ich habe für meinen Deye Inverter z.B. auch einen esphome Node

tobiasfaust, 2023-04-15:
ich habe eine neue Version der register.h eingecheckt. Kannst du mal prüfen ob jetzt alle Daten für den MIC da sind?

Also repo neu auschecken, kompilieren und drauf flashen

gerdluthe, 2023-04-15:
[image]

gerdluthe, 2023-04-15:
Sollte passen; Status ist noch "NULL" aber ansonsten sieht es plausibel aus. ConsumeIn Energy ist sicherlich nicht korrekt aber auch glaube ich uninteressant

tobiasfaust, 2023-04-16:
solved by fcea412

closed 2023-04-16 by tobiasfaust