tobiasfaust/SolaxModbusGateway #15: Support for Solax X3-MIC-G2 ?

MagicSven81, 2023-05-14:
Hallo Tobias,

erst einmal Glückwunsch zu einem super Projekt.
Gibt es eine Möglichkeit den X3-MIC-G2 ebenfalls auszulesen oder scheitert dies an der RS485 Schnittstelle des Inverters?

tobiasfaust, 2023-05-14:
Hi, 
Danke
Die Solax MIC Reihe habe ich mit #12 integriert.
Falls ein paar items noch fehlen , einfach hier kurz melden

MagicSven81, 2023-05-14:
Vielen Dank, leider bekomme ich keine Verbindung. Esp32 S über RX2, TX2 am HW-0519GPIO 16&17 und im Webinterface auf X3 und auch auf Mic gestellt. Leider je keine Rückmeldung. --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

tobiasfaust, 2023-05-14:
Hi,
Wenn die Kommunikation nicht klappt liegt es an den Modbus Einstellungen. 

Wenn RX aufblinkt sind zumindest die RX/TX Pins korrekt.
Wenn TX nicht blinken will muss Du die Baudrate und die clientID des Inverters  prüfen.
In #12 gibt es dazu auch wertvolle Hinweise.

MagicSven81, 2023-05-14:
An der Baudrate lag es. Mit 9600 klappt es nun. Aber ein paar Werte stimmen wohl nicht. Werde mich morgen mal genauer damit befassen. 

Ansonsten bin ich schon mal sehr glücklich. 
Wobei ich das auslesen über den USB Stick direkt in iobroker jetzt auch implementiert habe.

MagicSven81, 2023-05-15:
Hallo Daniel,Ich habe gestern etwas gebaut, was man nur per Rj45 am Wechselrichter einstecken muss. Auch der Strom kommt direkt von diesem Anschluss. Also komplett Steckerfertig mit eigens gedrucktem Gehäuse. Schreib mir einfach mal per PN.Bilder kann ich heute Abend hier posten. Gruß Sven

tobiasfaust, 2023-05-15:
> Hallo, würde gerne ebenfalls meinen X3 Mic in OPEN WB als PV 2 einbinden.
> Leider fehlen mir offensichtlich die Skills dieses Projekt hier umzusetzen. Gibt es hier jemand der mich bei der Umsetzung unterstützen könnte bzw. jemand dem ich eine nahezu Plug&Play Lösung abkaufen könnte ? :) Gruß Daniel

Ok, ich dachte die Anleitung zum zusammenstecken ist mehr als simpel und eindeutig? Sind ja nur eine paar fertige Kabel….

DaniElTsla, 2023-05-15:
Hi Sven,

das klingt hervorragend.
Was möchtest du denn dafür haben? :)

Gruß Daniel

MagicSven81, 2023-05-15:
https://www.kleinanzeigen.de/s-anzeige/solax-x3-mic-g2-modbus-adapter-realtime-auslesen-via-rs485-ttl/2442788497-168-8090

hier mal bitte drauf gehen und mich anschreiben. Dann können wir alles weitere besprechen.

MagicSven81, 2023-05-15:
Folgende Werte scheinen beim MIC-G2 noch nicht korrekt zu sein:

[image]

tobiasfaust, 2023-05-15:
Ich würde die Konversation lieber hier belassen, dann kann ich später auch alles nachvollziehen.
Ich bin gerade im Urlaub bis Sonntag. Um die fehlenden Werte zu ermitteln, kannst du bitte mal nach der Anleitung im Wiki vorgehen?
Beim Inverterstatus muss ich sicherlich noch eine Anpassung im Code einbauen, die andern werte kannst du aber erstmal selbst ermitteln, eintragen und dann hier posten :)

tobiasfaust, 2023-05-19:
Hast du schon die korrekte Item Konfiguration finden können? Ansonsten kann ich es mir erst nächste Woche anschauen. Dazu brauche ich aber von der RAW Data Seite einen Screenshot und den kopierten Hex String der Live Daten.

MagicSven81, 2023-05-21:
Hallo Tobias,
irgendwie tu ich mir schwer mit dem Umrechnen der HEX Werte.
Daher hier mal der Screenshot.

[image]

und hier der Live Daten.

[image]

Parallel dazu die "echten" Werte ermittelt mit einem SD72M MID:

[image]

tobiasfaust, 2023-05-21:
Ich bräuchte noch den kopierten hex String hier. Zusätzlich zum Screenshot ;) dann kann ich es automatisiert ermitteln

den Link zum hex umrechnen habe ich auch im Wiki hinterlegt ;)

MagicSven81, 2023-05-21:
ID-Data:

0x01Position: 0
Integer: 1
String: 0x03 0x28 0x4D 0x43 0x32 0x30 0x36 0x54 0x49
0x42 0x31 0x38 0x30 0x31 0x31 0x31 0x73 0x6F 0x6C
0x61 0x78 0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x20
0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x20
0x20 0x20 0x20 0x54 0xA3

Live-Data:

0x01Position: 0
Integer: 1
String: 0x04 0x80 0x12 0x34 0x00 0x00 0x00 0x42 0x00
0x00 0x09 0x64 0x09 0x67 0x09 0x6B 0x13 0x8C 0x13
0x8D 0x13 0x8B 0x00 0x27 0x00 0x28 0x00 0x28 0x00
0x2C 0x0B 0x54 0x00 0x02 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x0C 0x0D 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x07 0x70 0x00 0x00 0x00 0x80 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x24 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x5B 0xC3

tobiasfaust, 2023-05-21:
ich habe noch ein paar mini anpassungen gemacht. Problem ist aber hauptsächlich, das dein Inverter einige Daten nicht liefert die er abert laut Protokoll liefern müsste, zb.  "Grid Power L1" steht an stelle 35,36, direkt hinter dem Inverter Status. In den Daten sind aber nur 00 zu finden.... warum?? Eventuell musst du ein Firmwareupdate des Inverters machen?

Das Modbus Protokoll deines Invertes ist hier: https://github.com/tobiasfaust/SolaxModbusGateway/blob/master/docs/SolaX%20Power%20MIC-G2%20three%20phase%20inverter%20Modbus-RTU%20comms%20protocol%20V1.9.pdf

MagicSven81, 2023-05-21:
ok, ich suche schon seit 2 Stunden verzweifelt online nach einem Firmware Update. Aber ich werde nicht fündig.
Hast du hier eine Idee?

tobiasfaust, 2023-05-22:
Der Solax Support antwortet sehr schnell, frag mal warum die Modbus Antwort nicht zum PDF Dokument passt.
Musst aber den Modbus Anfragestring (Register.h) und die Antwort sowie das Dokument mitsenden.

MagicSven81, 2023-05-23:
Habe heute die neue Firmware erhalten. Werde heute abend nochmal auslesen und testen. Ich melde mich dann. Danke

MagicSven81, 2023-05-31:
Mhh komisch. 
Ich hab jetzt eine Möglichkeit gefunden über den WLAN Stick alle Daten in iobroker zu implementieren.  Daher schließe ich diesen thread, da für mich erledigt. 
Dennoch vielen Dank für den Support

closed 2023-05-31 by MagicSven81