tobiasfaust/SolaxModbusGateway #16: X1 mini G3

neb382, 2023-05-22:
Hallo Tobias, 

zunächst vielen Dank für das tolle Projekt. 
Ich habe ein Problem das mir immer nur die gleichen Werte geliefert werden und bin mir nicht ganz sicher ob im Modbus Protokoll zwischen dem X1 und dem X1 mini G3 unterschiede sind. 
Ich habe ein HW97 HalfDuplex Board genommen und den GPIO5 für DE/RE aber die LEDs blinken auch nicht. 
Ich habe die Default Baudrate belassen, könnte das schon ein Problem sein? Ansonsten wüsste ich nicht was beim wiring noch falsch sein könnte. 

Vielen Dank
Gruß
Ben

tobiasfaust, 2023-05-22:
Hi,
Das ist logisch ;) der Mini hat ein etwas spezielles Protokoll und ist noch nicht implementiert bei mir. Ich würde es aber gerne, mir fehlen aber die Testpersonen mit einem Mini die auch Lust drauf haben ;)
Schau mal in #7 

ich habe ein spezielles example für einen Mini bereitgestellt, mit dem muss man anfangen zu testen…. Anhand der Ergebnisse kann ich dann Stück für Stück den Mini hier einbauen.

neb382, 2023-05-22:
Hey achsooo  okay ja gerne hab auf jeden Fall Lust! 
Das ding steht nur bei Schwiegervater in der Garage, am besten hau ich mal einen miniPC davor dann kommen wir remote da drauf und können testen. 
Vielen Dank für die schnelle Rückmeldung!!

neb382, 2023-06-06:
Hallo Tobias, ich hab einen kleinen iNUC am ESP der mit dem MiniG3 verbunden ist, leider habe ich nicht bedacht das ich das example jetzt nicht aufspielen kann, da der ESP nicht im boot modus ist :-/ 
Kann man mittels dem Reiter RAW data noch was testen?  Was mich verwundert ist, das initiale Werte wie Grid Voltage,, Grid Frequency scheinbar richtig sind, also irgendwas kam ja mal an. Gibt es noch irgendwas das man hier tun kann oder das Example zunächst ausgeführt werden? 

Danke

neb382, 2023-06-06:
ja stimmt, hab die fertigen binarys. Okay das erklärt die werte. 
Dann werd ich da nochmal hinmüssen und das example aufspielen. 

Danke für die Erläuterung

neb382, 2023-07-02:
Hi, 
hat was länger gedauert, hab das example 
drauf aber bekomme keine response zurück :-/ 
sind die Adressen A0 so korrekt oder müssen noch angepasst werden?

tobiasfaust, 2023-07-02:
Du musst die Einstellungen mit deinem Wechselrichter vergleichen und anpassen.

du musst mit der Doku zum Modbus Protokoll selbst austesten, ich habe ja keinen Mini

neb382, 2023-07-02:
achso d.h. ich finde die daten in den Einstellungen des WRs? 
Die Daten aus der Doku hab ich bereits probiert aber ohne Erfolg, ich dachte die Antwort auf z.B. die Seriennummer würde man immer bekommen, aber dann muss ich mal schauen was es da so gibt

tobiasfaust, 2023-07-02:
Ja genau, die Modbus Einstellungen des WR müssen mit den Einstellungen im Sketch übereinstimmen ;)

kommando828, 2023-07-02:
This is the X1-Boost 485 data pdf, as the list of models includes the Mini in the 'Model' register then the Mini could use this data or be close to it.

0x34E PowerType R Machine Type See Machine Type Uint8_Lo

Machine Type:
X1-BOOST
#define 3.0Kw 0
#define 3.3Kw 1
#define 3.6Kw 2
#define 4.2Kw 3
#define 4.6Kw 4
X1-MINI
#define 0.7Kw 5
#define 1.1Kw 6
#define 1.5Kw 7
#define 2.0Kw 8
#define 0.6Kw 9
X1-AIR
#define 2.5Kw 10
#define 2.8Kw 11
#define 3.3Kw 12
X1-BOOST
#define 5.0Kw 13
#define 5.3Kw 14

[Solax Power X1-BOOST Single phase inverter Modbus-RTU comms protocolV1.0 - Public version.pdf](https://github.com/tobiasfaust/SolaxModbusGateway/files/11931194/Solax.Power.X1-BOOST.Single.phase.inverter.Modbus-RTU.comms.protocolV1.0.-.Public.version.pdf)

neb382, 2023-07-07:
So erste Ergebnisse , kann man was damit tun? Oder muss ich jetzt mit Read 0x11 die einzelnen Register nach Funktion 0x04 abfragen?

[testing.txt](https://github.com/tobiasfaust/SolaxModbusGateway/files/11985917/testing.txt)

tobiasfaust, 2023-07-08:
Deine responses sind immer unterschiedlich. Da passt etwas nicht. 
Ich denke das deine Anfrage nicht stimmig ist 

ControlCode ist 11, danach functionCode 02

siehe Beispiel im doc Kapitel 6

wenn alles richtig läuft muss der Response immer mit AA 55 beginnen

neb382, 2023-07-09:
hmm hab das auch probiert, wobei source and destination address ja 2 byte sind, 
Also auch mit 
AA 55 00 00 00 01 11 02 10 00 00 bekomme ich nur unterschiedliche responses
In dem Menü Einstellungen ModBusAddress stand 1 und ich habe es auf 2 geändert gehabt. 
Daher dachte ich ESP ist 01 und WR 02, aber auch mit ESP 00 und WR 02 bekomme ich immer nur diese  
unterschiedlichen responses. 
In einem anderen Forum habe ich noch was von Broadcast gesehen, wo angeblich die Seriennummern aller im Bus hängenden WRs zurück kommen sollen ohne eine Adresse angeben zu müssen, das soll dann so aussehen:
# Send discovery to broadcast address (0x10 0x00)
[VV][solax_modbus:200]: TX -> AA.55.01.00.00.00.10.00.00.01.10 (11)
[VV][solax_modbus:084]: RX <- AA.55.00.FF.01.00.10.80.0E.31.32.33.34.35.36.37.37.36.35.34.33.32.31.05.75 (25)
                                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

Aber auch hier bekomme ich keine AA 55 response, also muss doch irgendetwas anderes nicht stimmen, oder? 
Bzgl. der Verkabelung habe ich auch keinen Fehler gefunden.. 
Noch eine Idee was ich testen kann? 
Wie gesagt da hängt ein kleiner PC vor auf den man remote drauf kann und wir können da gerne auch mal zusammen anschauen

Mit der Baudrate hab ich auch mal rumgespielt, aber in dem Solax Modbus Protokol Dokument steht 9600 daher denke ich das sollte so stimmen. 

Bin aktuell Ratlos was ich noch testen kann :-(

tobiasfaust, 2023-07-09:
Probier mal Tasmota ob es klappt, da gibt es ein Solax X1 Plugin. Funktioniert wohl nur mit dem Mini.

entweder reicht das dann schon oder du kannst damit testen.

muevol, 2023-07-17:
Hallo Tobias,

ich habe diesen Thread aufmerksam verfolgt, da ich seit 3 Wochen auch einen Solax X1-2.0-S-D Mini G3.1 in Betrieb habe und ich Dein Projekt sehr gelungen finde.
Da Modbus für mich zu komplex ist, habe ich auf ein fertiges Produkt gehofft, also ESP32 und HW-0519 besorgt un einfach mal Dein Beispielscript ausprobiert, die RX Diode blinkte aber die TX nicht und hatte damit auch keine Antwort bekommen.
Im Fhem-Forum hatte ich einen Thread gefunden, der den Anschluß eines X1 Mini per Modbus beschreibt, aber als Vorraussetzung die Freischaltung von Modbus RTU und die Einstellung einer ID verlangt, und das ganze in den erweiterten Einstellungen mit speziellen Passwort.
Nur finde ich keine Beschreibung, weder im Hanbuch noch im Internet, wie man da rein kommt.
Im Hanbuch sieht man über Power -> Setting -> 0000 -> Safety Grid kommt, aber dort ist kein Menüpunkt "erweiterte Einstellungen" aufgeführt.
Kannst Du mir auf die Sprünge helfen ?
Ich würde gerne meinen Solax WR testen, was der mir so erzählt, auch Tasmota wird doch nicht ohne Freischaltung funktionieren.

Gruß

Volker

kommando828, 2023-07-17:
Settings is the right place, change the 0000 to 6868 and then you can enter the full settings menu. Turn ModbusRTU to enabled and Modbusaddress to 1.

neb382, 2023-07-17:
hi kommando as i stuck on the same as Volker, the X1 mini has two options
RTUModbusEnable  -> enable -> disable
not "enabled" or "disabled" which i was thinking enable is correct, so l leave the display with "disable"
its maybe a transaltion issue (i used english instead of german as this is even more worse) 
but i am not sure what the correct settings should be to finally enabled it :O) 
I also used modbusaddress 2 and none of my test even with tasmota and the x1 plugin provides me with any response at all. 
Is 1 a must have for the address??

kommando828, 2023-07-17:
Set ModbusRTU to read enable.

Address can be 1 to 254 but Tobias's program defaults to 1 so I set mine to 1 to save having to check for 2. You really only need 2 when you have 2 inverters.

tobiasfaust, 2023-07-18:
Das Problem ist, das der Mini ein abgewandeltes Protokoll hat. Das kann ich mit meinem Projekt unterstützen, wer es nutzen will muss aber selbst testen, Doku lesen, serielles Log auswerten und weiter testen. Ich kann hier nicht wirklich viel unterstützen außer hinweise zu geben.
Bei dem Solax-Mini ist auf jeden Fall das Mini-Example und die Mini-Doku der Startpunkt.

MagicSven81, 2023-07-30:
Hallo Tobias,
wie genau muss ich mit dem Testscript (.ino) vorgehen? Ich habe einen X3 mini G3 und den ESP32 angeschlossen. Release 3.1.1 ist installiert.
Es kommen allerdings folgende Werte raus:
[image]

kommando828, 2023-07-30:
Read from

https://github.com/tobiasfaust/SolaxModbusGateway/issues/19#issuecomment-1598407475

I had exactly the same and fixed it by reducing the number of registers being polled, follow the link and use Tobias or my version.
[image]

MagicSven81, 2023-07-30:
Hi kommando828
i´m sorry but i did not know what to do.
How can i reduce the numbers?
I still create the .bin file with gitpod and uploaded it to the ESP32.

kommando828, 2023-07-30:
First follow this and edit Register.h file

_Please change the Parameter of RequestLiveData of MIC in Register.h

["#ClientID", "0x04", "0x04", "0x00", "0x00", "0x40"]

Maybe the 0x40 to 0x20 to request only 20 register instead of 40. please check your modbus doc._

Then use Gitpod to make a new bin file with the changed Register.h and upload to the ESP32

If that still does not work then the 0x40 to 0x16 instead of 0x20, this one worked for me on a X1 Boost G3.

MagicSven81, 2023-07-30:
ok, register.h was edited but how can i start the new compilation on gitpod?

kommando828, 2023-07-30:
I don't use Github, it does not work with my ESP32 So I use Virtual Studio on the downloaded files which can be edited before uploading.

MagicSven81, 2023-07-30:
can you send me your .bin file please?
bruce_@gmx.de

i have the same X3 mini G3

kommando828, 2023-07-30:
Virtual Studio does not make a bin file.

tobiasfaust, 2023-07-31:
Hi Sven,

wenn du wie in den Screenshots angegeben Daten irgendwelcher Form erhältst hast du keinen echten Mini sondern wohl einen Solax Boost. ;)
Mit einem echten Solax Mini würdest du erstmal keine Daten erhalten.

Ich glaube mit einem Firmwareupdate für deinen Boost solltest du auch die echten 64 Register auslesen können, so wie es aktuell in der Register.h steht.
Ohne Fw Update bekommst du weniger Register raus und musst in der Register.h die 0x40 wohl auf 0x20 oder 0x16 anpassen.

neb382, 2023-07-31:
Hey all, 
so could it be the same on my end? Saying the firmware may affect my X1 G3 to not respond correctly and therefore all my values are 0 also in Tasmota with the plugin? 
I never did a firmware update on the Inverter yet. 
I changed the Inverter Settings to 1 for the Address and "read enable for the RTU modbus" still nothing.

kommando828, 2023-07-31:
According to the Solax RS485 PDF for the X1 Boost G3 then it also has entries in the inverter identification register for

Machine Type:
X1-BOOST
#define 3.0Kw 0
#define 3.3Kw 1
#define 3.6Kw 2
#define 4.2Kw 3
#define 4.6Kw 4
X1-MINI
#define 0.7Kw 5
#define 1.1Kw 6
#define 1.5Kw 7
#define 2.0Kw 8
#define 0.6Kw 9
X1-AIR
#define 2.5Kw 10
#define 2.8Kw 11
#define 3.3Kw 12
X1-BOOST
#define 5.0Kw 13
#define 5.3Kw 14

So its also likely the same RS485 configuration is used. Now likely is not 100% certainty but worth trying out. So what worked for me ie 0x16 should also work for an X1 Mini G3.

The first RS485 enabled firmware in the X1 Boost G3 was ARM 1,37, so as long as your X1 Mini G3 is ARM 1.37 or later then its also likely to work. That you can change the RS485 address to 1 and enable 485 RTU on your inverter suggests your firmware is the right one. You can check it in the settings on the inverter.

MagicSven81, 2023-07-31:
@kommando828  -  it works - thx.
The only missing register is "Total Energy to Grid"

[image]

kommando828, 2023-07-31:
Well done.

That entry is zero if you do not have the Solax grid meter connected, you will note the grid power is 0W too as there is no info coming to the inverter from the meter. You only need the meter if you need to control your feed in power to the grid. I have an auto diverter that sends the power to a water heater by monitoring the export 50 times a sec and keeping it at zero,

tobiasfaust, 2023-08-01:
Hi Sven,
What is your final and last configuration of  Register.h? Which Firmware Version you are now using?
Did you updated your Firmware?

MagicSven81, 2023-08-01:
Hallo Tobias,Firmware ist der neuste Stand. Es handelt sich wirklich um einen X1 mini G3.Nach Anpassung auf 0x16 funktioniert es, bis auf die Gesamte Erzeugerleistung.Gruß Sven--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

tobiasfaust, 2023-08-01:
Kannst du mir die genaue FW Version nennen? Welchen invertertyp hast du ausgewählt bzw bei welchem hast du auf 0x16 geändert? Ich würde dann eine extra section dafür aufsetzen.

MagicSven81, 2023-08-01:
Hallo Tobias, Bild im Anhang. Unter Solax-Mic habe ich von 0x40 auf 0x16 gewechselt dann kamen Daten an. Werde heute Abend testen ob es mit 0x20 auch noch geht, da die gesamt Erzeugte Leistung noch mit 0kwh angezeigt wird.Gruß Sven--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

tobiasfaust, 2023-08-01:
> Hallo Tobias, Bild im Anhang. Unter Solax-Mic habe ich von 0x40 auf 0x16 gewechselt dann kamen Daten an. Werde heute Abend testen ob es mit 0x20 auch noch geht, da die gesamt Erzeugte Leistung noch mit 0kwh angezeigt wird.Gruß Sven>

logisch, die Daten fangen ab Stelle 75 an. Mit der Einstellung 0x20 (32 Register) kommst du bis Stelle 64. 
Geh mal im WebUI auf die RawData Seite. Dort musst du deinen Wert "suchen".
Siehe auch hier: [adding new inverters or items](https://github.com/tobiasfaust/SolaxModbusGateway/wiki/adding-new-inverters-or-items)

tobiasfaust, 2023-08-03:
Welche Firmware Version hast du? Hast du schon auf die neueste FW-Version geupdated? Wenn dann wirklich weniger als 64 Register (0x40) abfragbar sind würde ich eine eigene Section in der register.h dafür anlegen

MagicSven81, 2023-08-03:
Hallo Tobias,
die neuste Firmware ist aufgespielt, A:1.48 D:2.31
Master 1.08  // Manager 1.07

mit dem Register (0x40) kommen keine Daten.
Ändere ich es auf (0x16) kommen alle Daten, bis auf Total Energy und Today Energy.

Ich teste gleich noch was bei 0x20 passiert und ergänze den Post hier.

gerdluthe, 2023-08-05:
Ich schliesse mich dem an, mein X3 gibt auch keine Daten mehr raus.
Firmware Update habe ich keins installiert.

Hardware habe ich nochmal nachgebaut, spuckt auch nur noch undefined aus.

Hatte Kontakt mit dem Support meines Lieferanten, die haben auch auf den WR geschaut, weil ich dachte er produziert nicht mehr, eventuell haben die eine neue Firmware installiert, das entzieht sich aber meiner Kentniss.

tobiasfaust, 2023-08-06:
> Ich schliesse mich dem an, mein X3 gibt auch keine Daten mehr raus. Firmware Update habe ich keins installiert.
>
> Hardware habe ich nochmal nachgebaut, spuckt auch nur noch undefined aus.
>
> Hatte [...]

Dasselbe wie hier?
https://github.com/tobiasfaust/SolaxModbusGateway/issues/16#issuecomment-1657214013

gerdluthe, 2023-08-06:
Ja, so war es. Ich habe allerdings heute Nacht noch einen Node gebaut und damit funktioniert es wieder...warum auch immer. 
Also alles in Ordnung bei mir.

Sorry

tobiasfaust, 2023-08-06:
> Ja, so war es. Ich habe allerdings heute Nacht noch einen Node gebaut und damit funktioniert es wieder...warum auch immer. Also alles in Ordnung bei mir.
> 
> Sorry

Alles gut…. Manchmal ist man in der Nacht am produktivsten ;)

tobiasfaust, 2023-08-06:
@MagicSven81 

Poste mal bitte:

- Screenshot der RawData Seite
- diese Daten mal als Text herauskopiert damit ich mit arbeiten kann
- Die vermissten Werte die am WR auf dem Display angezeigt werden, exakt zur selben Zeit wie der Screenshot 
- 
damit kann man dann die Werte suche wie im Wiki beschrieben

MagicSven81, 2023-08-06:
[image]

Vermisster Wert auf dem Display "Gesamt: 512 kwh"

tobiasfaust, 2023-08-06:
Bitte mal exakt diese Daten:
- diese Daten mal als Text herauskopiert damit ich mit arbeiten kann
- 
Hast du leider vergessen

MagicSven81, 2023-08-08:
Hallo Tobias,
sry aber ich bin ehrlich gesagt totaler Laie und Endanwender. Was genau muss ich bei den Hex-Werten schauen?
Bzw. was für Daten kann ich dir als Support noch liefern damit der X1-mini implementiert werden kann?

tobiasfaust, 2024-02-07:
closed due inactivity

closed 2024-02-07 by tobiasfaust