tobiasfaust/SolaxModbusGateway #17: Issues trying to Add Solarmax SGA /Sofar Solar Inverter

Dobie007, 2023-05-29:
First of all : Thanks for the great stuff here!
I am trying to add my Inverter and am running in to some issues that i don't know how to solve:
This is what i have:

- ESP and RS232 interface are working and i was able to get answers to the LiveData as well as the IdData over the test files using arduino IDE. 
- I edited the register.h file using the gathered answers and the Modbus document I found with this search string: "SofarSolar Modbus Registry Details.pdf" on OpenEnergyMonitor. 
- I checked the Json to be correct.
- I compiled everything using PlatformIO and set up the ESP using the Web Interface. 

In parts things work  but:
1) The IdData is seldomly being filled - it actually worked at one instance I currently cannot reproduce. The log file looks strange in that point - no clear request string...
2) I cannot enable any commands although the "enable set commands over MQTT2" is active: The only item that is showing up in MQTT is the inverter status. (Yesterday I had a few to select but not all of my listed objects).

[image]
[image]
[image]

I nearly made it...
Here the latest log file:
[Log2.txt](https://github.com/tobiasfaust/SolaxModbusGateway/files/11591406/Log2.txt)
And the Response i got with the test program.
[TestResponse.txt](https://github.com/tobiasfaust/SolaxModbusGateway/files/11591471/TestResponse.txt)

I would appreciate your help..

Dobie007, 2023-05-29:
I posted the register.h as text file above..   
Github tells me "We dont support that file type" if i drag and drop it here...

May that be caused by the name string I chose?: "Sofar-KTL Solarmax-SGA"

tobiasfaust, 2023-05-29:
at first, your definition of "RequestIdData" is wrong. please use:

"RequestIdData": ["#ClientID", "0x04", "0x20", "0x00", "0x00", "0x0B"],

and try again

Dobie007, 2023-05-29:
Hi, thanks for your response.. (its not urgent, you can take your time :-))
I changed it, but it seems to make no difference.  This time:

Inverter found: Sofar-KTL Solarmax-SGA
Read config data for inverter ���?␟
{
That is directly after reset before communication - so which config data does that refer to..? 
I assume its from the ESP flash...

Further down in the log the IdData request comes ( I set the live data to one every 10s, and the IdData to one every 60s)
but I get "no response from client"

This is after I changed  the Register file as you advised to: 

	"Sofar-KTL Solarmax-SGA": {
		"config": {
			"RequestLiveData": [
				["#ClientID", "0x03", "0x00", "0x00", "0x00", "0x30"]
			],
			"RequestIdData": ["#ClientID", "0x04", "0x20", "0x00", "0x00", "0x0B"],
			"ClientIdPos": 0,
			"LiveDataFunctionCodePos": 1,

[register_h.txt](https://github.com/tobiasfaust/SolaxModbusGateway/files/11592245/register_h.txt)
[Log3.txt](https://github.com/tobiasfaust/SolaxModbusGateway/files/11592301/Log3.txt)

Could the 9600 Baud be a problem - not sure if I can  change that.

tobiasfaust, 2023-05-29:
Hi,
i just committed an update. There was an memory leak in weboutput of "itemconfig" (-> null after Inverterstatus)
Another mistake by you was the definition of "id", you forgot a space ;)

So, just clone the repo and test again

Dobie007, 2023-05-29:
Good One!: I can select the items now and they are being passed to MQTT. In other words I can actively use it now.

Only little drawback: The serial number is not decoded in the first attempts to read the "id": The log shows:

_Request queue data to inverter: 0x01 0x04 0x20 0x00 0x00 0x0B 0xBA 0x0D 
Read Data from Queue: 
no response from client_

After a few minutes it gets the id and I have my Serial Number display as well.

Attached is the register.h file (for further reference):
[register_h.txt](https://github.com/tobiasfaust/SolaxModbusGateway/files/11592966/register_h.txt)

Thanks, I can dismiss the Chinese logger now. :-)

Dobie007, 2023-05-30:
Big THANKS! to Tobias - Everything solved in no time!

closed 2023-05-30 by Dobie007

tobiasfaust, 2023-05-31:
finally added by 9ef955d