tobiasfaust/SolaxModbusGateway #18: X1 Boost Gen 3

kommando828, 2023-06-05:
I have attached the Modbus register PDF I obtained from Solax for the X1 Boost G3.0/G3.1/G3.2, my inverter has a serial number starting XB32 which may indicate a Gen3.2. I have access to the Cloud but would prefer to access the data locally. I have connected to the inverter's RS485 wires via an EW11 which converts to Modbus TCP but while I can communicate the signals back show as invalid in Modbus poll but at least communication is there. The HEX register list in the PDF looks to match the ones posted in Benjamin's ESPHome from this forum thread which you started. https://community.home-assistant.io/t/universal-solar-inverter-over-modbus-rs485-to-mqtt-formally-solax-no-pocket-wifi/483606/4 

[Solax Power X1-BOOST Single phase inverter Modbus-RTU comms protocolV1.0 - Public version.pdf](https://github.com/tobiasfaust/SolaxModbusGateway/files/11653824/Solax.Power.X1-BOOST.Single.phase.inverter.Modbus-RTU.comms.protocolV1.0.-.Public.version.pdf)

tobiasfaust, 2023-06-05:
Hi,
I dont Support an EW11.
Please Look into the Wiki to figure out which hardware you need ;)

kommando828, 2023-06-05:
The EW11 was to hand and showed me the comms is working, will not be using it in place of your solution. Sorry for confusion.

tobiasfaust, 2023-06-05:
I think with the settings of SOLAX-MIC it will work out-of-the box. We will see ……

kommando828, 2023-06-05:
Yes, it looks like the X1 Boost uses a truncated version of the Solax-MIc register. The machine type returned digit maps to a different description is the one difference I can see. For example a return of a 2 from 0x34E refers to a #define 3.3Kw  in the X1 Boost PDF but in the Mic PDF refers to a POWER_4KW_MIC. As they could be used together you would have expected Solax to extend the list and have no duplicates.

tobiasfaust, 2023-06-06:
It sounds good :)
Please inform us if your X1 boost fits to the MIC settings.  So i can insert the Boost in the compatibly list:)

tobiasfaust, 2023-06-09:
Try it out what fits for you ;)
Solax X1 and Solax MIC use different Register adresses

kommando828, 2023-06-09:
So the register.h on Firmware ver 3 does not include the MIC but the register.h in Github does. As programming is not one of my skills do I copy the Github register.h which included MIC into the Ver 3 folder overwriting the current version recompile and then send this firmware to the ESP32.

tobiasfaust, 2023-06-09:
Ok, but easier by using gitpod, See my wiki

kommando828, 2023-06-09:
Ok, heads bursting but got the 3 files downloaded ;) now to load the firmware and get the ESP32 up and running again. I assume when I test and if I find the MIC extra registers causes issues then I will need to create a new register.h with an extra X1-boost section with only the X1-boost registers. That will then add an extra option for the X1-boost.

tobiasfaust, 2023-06-09:
Not a new register.h, only a new section. You will understand if you take a look into it and understand the json Format ;)
The wiki explains it too ;)

kommando828, 2023-06-09:
No joy with the firmware upload, showed as Finish in B;lue but the ESP32 is still up and running after a reboot on Ver 3 with all the settings retained. I had expected it to go back to showing as a WiFi AP and needing to scan for WiFi to connect to and then need to change Baud rate etc, Must be this error in the terminal that is the cause

NO XMC flash  detected!

So need to find out exactly which ESP32 version chip I have

kommando828, 2023-06-09:
Tried Gitpod after a factory reset, still did not work. Went back to Visual Studio after a second factory reset, used the current Github files and followed same procedure with a build and load and it works plus MIC is now selectable. No idea what the issue is with Gotpod but my ESP32 does not like it.

kommando828, 2023-06-11:
Hardware is up and running, RX and TX are flashing and a serial monitor shows no data is being lost/dropped. However as with all the previous attempts with software on my Win11 laptop, the response back from the X1-boost under raw data shows as being 'undefined', similar to 'invalid' as declared by Modbus Poll and others. I have a suspicion Solax have not implemented Modbus on this inverter at all as the same 2 pins on the RJ45 are also defined as being the comms with the output limiting meter. Will have to try other methods even if it ends up just being a CT meter monitoring the output KwH. This is 100% a Solax issue IMHO.

closed 2023-06-11 by kommando828