tobiasfaust/SolaxModbusGateway #19: Just read total energy from inverter

cob555, 2023-06-15:
Hello.
This is a nice project, but waaaaay to comprehensive for my needs. It would be enough, to read the total produced energy out and have this value displayed on a little LCD or 'even' on a HTML-page via asyncwebserver.

I have run the test-script on a ESP32 and got at least some answer from the inverter: "07"
Not via the serial port (can't get the PC to the inverter), but on the LCD. The code was modified a bit and I don't even know if it can display the whole serial number of the inverter this way.

`#include <LiquidCrystal_I2C.h>
#include <Arduino.h>

#define RX         16   // UART2 RX
#define TX         17   // UART2 TX
#define RTS_pin    5    // RS485 Direction control
#define RS485Transmit    HIGH
#define RS485Receive     LOW

// set the LCD number of columns and rows
int lcdColumns = 16;
int lcdRows = 2;

//LCD SCL: Pin 22; SDA: Pin 21
// set LCD address, number of columns and rows
// if you don't know your display address, run an I2C scanner sketch
LiquidCrystal_I2C lcd(0x27, lcdColumns, lcdRows);  

void setup(){
  // initialize LCD
  lcd.init();
  // turn on LCD backlight                      
  lcd.backlight();

  pinMode(RTS_pin, OUTPUT);
  digitalWrite(RTS_pin, RS485Receive);

  // Start the built-in serial port, for Serial Monitor
  Serial.begin(115200);
  Serial.println("Test RS485 Connection");
  
  lcd.setCursor(0, 0);
  lcd.print("Test RS485");

  // Start the Modbus serial Port
  Serial2.begin(19200, SERIAL_8N1, RX, TX);
  delay(1000);
  
  while(Serial2.available())
  {
    Serial.print(Serial2.read(), HEX);
    Serial.print(" ");
  }
  
  delay(1000);

}
uint16_t Calc_CRC(uint8_t* message, uint8_t len) {
   //Calc the raw_msg_data_byte CRC code

  uint16_t crc = 0xFFFF;
  for (int pos = 0; pos < len; pos++) {
    crc ^= (uint16_t)message[pos];          // XOR byte into least sig.byte of crc
    for (int i = 8; i != 0; i--) {    // Loop over each bit
      if ((crc & 0x0001) != 0) {      // If the LSB is set
        crc >>= 1;                    // Shift right and XOR 0xA001
        crc ^= 0xA001;
      }
      else                            // Else LSB is not set
        crc >>= 1;                    // Just shift right
    }
  }

  return crc;
}

String Hex2String(uint8_t num) {
  char hexCar[2];

  sprintf(hexCar, "0x%02X", num);
  return hexCar;
}

void loop(){
byte request[] = {0x01,
                               0x03,
                               0x00,
                               0x00,
                               0x00,
                               0x07, // 0x0D - 14 stellen, Inverter + Modulname
                               0x00,
                               0x00
           }; //

  uint16_t crc = Calc_CRC(request, sizeof(request)-2);
  request[sizeof(request)-1] = highByte(crc);
  request[sizeof(request)-2] = lowByte(crc);

  Serial.println("Schreibe Daten ....");
  for(uint8_t i=0; i<sizeof(request); i++){
    Serial.print(Hex2String(request[i]));Serial.print(' ');
  }
  Serial.println("");
  
  digitalWrite(RTS_pin, RS485Transmit);
  Serial2.write(request, sizeof(request));
  Serial2.flush();
  digitalWrite(RTS_pin,RS485Receive);
  
  delay(100);
  Serial.println("Lese Daten ....");
  
  while(Serial2.available())
  {
    Serial.print(Serial2.read(), HEX);
    Serial.print(" ");
  
  lcd.setCursor(0, 1);
  lcd.print(Serial2.read(), HEX);
  }
  Serial.println("");
  delay(2000);

}`

How would the request to the inverter has to be changed, to read the total energy? Is this even possible with this code?

Best regards, Mathias

tobiasfaust, 2023-06-16:
Hi
Which Inverter you have? You have to know which Register you need.

at First, Monitor all Output via your ESP32 serial console. All others would be „picking in the fog „

cob555, 2023-06-16:
Hello Tobias, 
I have a X1-3.0 Mini and a X1-3.6 Boost (dual PV-input) inverter.
Using the serial console would be rather complicated, since I would have to bring the PC to the inverter.

kommando828, 2023-06-16:
There are no shortcuts to connecting to the RS485 lines in the inverter. RS485 cables can be long but need to be twisted pair and have terminating 120 ohm resisters on the end connection and the shielding grounded. By doing it close to the inverter with the PC you can use 2 plain wires and forget the resistor and shield, once you know it works you can invest in the proper RS485 or use wireless RS485 adaptors. I have a laptop set up for this which works well with all my SMA inverters, after confirming they can communicate I fit wireless RS485 adaptors. However the X1 Boost Gen 3 will not connect properly with the laptop or an ESP32 with Tobias software, either Solax have not properly implemented RS485 on the boost or my inverter has a RS485 circuit fault.

cob555, 2023-06-16:
> There are no shortcuts to connecting to the RS485 lines in the inverter
Sure, the ESP32 is connected via a short patch-cable and RS485-TTL-converter to the inverter.
Only the connection to the PC for a serial interface from the ESP32 is not possible.
If there is a communication problem, I can't do anything about it anyway, so the test via the serial interface would be nice but not strictly mandatory.

tobiasfaust, 2023-06-17:
Sorry, i cannot Support without directly access to log output.
Maybe, please rent a laptop from your Friends or someone like that.

cob555, 2023-06-17:
O.k. - you won :-)
I have torn down my PC and brought it to the inverter.
Here is the result for the X1-Boost (the other unit did not reply. but that's a problem for another day):

```
0x01 0x03 0x00 0x00 0x00 0x07 0x04 0x08 
Lese Daten ....
7F FF F1 EF 
```

At least I know, the LCD was not displaying what it should have.

tobiasfaust, 2023-06-17:
Something goes wrong. Your Wiring or modbus config.
The correct answer has to Start with: 0x01 0x03 ….

kommando828, 2023-06-18:
Looks like you have the same issue I have, according to Solax support my wiring is wrong but a multimeter check shows I am connected correctly to pins 4 and 5 on the RJ45 when I test the pigtail. Just in case the pigtail I used, which is a short Cat6 cable with thin solid copper wires that break easily, is the problem I have ordered some CAT6 network cables with braided 24AWG wires and will test with these. Other than this short 6" pigtail the setup is exactly the same as I use on my other SMA inverters, these have been working with no comms issues for 2 years. I am using CAS Modbus Scanner which allows you to set a scan for the known Modbus registers and give you the return, on my X1 Boost Gen 3 with serial number starting with XB32 I get the following

Read holding register starting at 40768 for 7 registers (calling for Inverter SN) 
=> Poll: 01 03 02 FF 00 07
<= Response: F9 FB 01 F1 

So same type of nonsense reply despite using Solax's X1-Boost register for serial number.

tobiasfaust, 2023-06-18:
Its also important to match port A and B correctly. If they are exchanged, communication fails.

to find out the correct wiring, its recommend to use the example program

what is Register 40768? You mean request string 
0x01 0x03 0x9F 0x40 0x00 0x07. ???

kommando828, 2023-06-18:
If I swap A and B then all comms stop.

Register 40768 is the decimal version of the hex naming of the register.

So 0x03 Read holding register is the decimal range being 40000 onwards

within 0x03 the registers 0x300 to 0x306 is the range 40768 to 407774 which is the serial number of 14 digits as the length should be 2 for each register.

I may have some of it wrong as it could have an offset of 1 but I have tried various combination of registers and get garbage back every time and with several different programs. Your own programs raw data query returns 'undefined' for all register queries I have tried.

tobiasfaust, 2023-06-18:
Ok, got it
In this case the best strategy is to test with the example Sketch as long as your need to find out the right wiring or request string without garbage return

 0x01 0x03 0x03 0x00 0x00 0x07
to request register 0x300 - 0x306

please don‘t Write as decimal, its confusing us ;) ;)

cob555, 2023-06-18:
Quite an aggravating interface...I'm pretty sure, the connection is correct. If I swap A and B I don't get any response on the LCD. I have not swapped the connection via the PC-test.
Good that you say the response is garbage, to me it just looked like some HEX-numbers, without the leading "0x".

I just noticed, that I most likely used the wrong example code, not the one specifically named "test_RS485_SolaxMini_ESP32" ...doh!!!

New test-setup is in the making......

kommando828, 2023-06-18:
I have dismantled everything in prep for the new cables so no more testing possible until late this week from me, the rewire will include using the Reg GND and a 120ohm resistor but it really should be working now. I am still of the opinion that Solax have not implemented RS485 on certain versions of the X1 non hybrid single phase inverters, that is despite their support confirming my inverter settings including firmware version are correct and it was Solax support that provided the register map PDF which matches your MIC pdf. But after the new cable arrives then I will test again but I have my doubts it will work and I am seeing cobb's posts as confirming that suspicion. It may be something such as corrupted firmware but then why are Cobb's issues so similar unless the original firmware file is corrupt.

My excuse for Decimal is the readout of CA query field is decimal even if the comms box is hex but its really as all previous register maps such as for SMA and Janitza are all in decimal ;) .

tobiasfaust, 2023-06-18:
The example program „test_RS485_SolaxMini_ESP32“ is only for a Solax **mini**. It has a very different protocol and its a pre-alpha status 

the other example it the better one

cob555, 2023-06-18:
All right. But in this case I don't know what to do.
If the inverter from kommando828 show's the same (miss)behavior, there maybe is no solution (MODBUS-wise) at all...

The next best thing would be a WiFi-dongle, but without using the Solax-cloud.

kommando828, 2023-06-18:
I have the Wifi dongle and its currently using the cloud which I do not like. There are solutions which use the dongle locally but you have to first buy the right dongle (X1 Boost up to Serial number XB32 use Ver 2, XB33 onwards use Ver 3) and the right firmware loaded to be able to collect the data locally and drop the cloud. 

Currently I am happy to carry on testing once the cable arrives as with 2 examples of faulty output I know there is a more likelyhood of a general issue than a solitary issue of my own making. 

If the new cable does not work I will be asking for new inverter firmware to load from Solax.

I only got this inverter to be UK G99 compliant, my SMA's are only UK G83 which is out of date but they are looking more and more tempting to reinstall by the day.

tobiasfaust, 2023-06-18:
hmm, i defend myself against to believe, the Boost Inverter dont work with modbus RTU if the solax documentation tells us that it should works.....
If wiring is correct (RX and TX led´s are flashing both) i suggest the following investigations (with example program):
* "play with it" and change the Request String
* maybe the CRC computation is different?
* playing with the delays between request and reading answer

Any google findings of other users regarding X1 Boost and Modbus RTU?

cob555, 2023-06-18:
How would have the request string be "played with"?

kommando828, 2023-06-18:
I get the TX and RX flashing so if that confirms the wiring works the new cable will not be helping then.

Google gets you to

https://github.com/wills106/homeassistant-solax-modbus/wiki/Compatible-Inverters

and the entry for X1 Boost is

PV Only
Single Phase
X1-Air, X1-Boost & X1-Mini

Only Firmware Arm 1.37 and above is supported.

Modbus needs to be enabled in your Admin Menu on the Inverter.

Firmware version below use a non standard protocol over RS485, they do no speak Modbus.

Known Serial Numbers

XB3
XMA
XM3

So my firmware is later than 1.37 and Modbus is enabled in menu and address is set to 01. The note on pre 1.37 firmware seems to match what I am getting but Solax say my firmware should work as it is 1.42,

kommando828, 2023-06-19:
New cables came today and given a partial success, read elsewhere yesterday that some cheap network cables use copper plated aluminium cored wire that causes RS485. So full copper stranded 24AWG works but a randomly selected cable may not. So after trying the ESP32 and not really getting any further I went to CAS Modbus Scanner, scans starting at address 40768 resulted in the inverter sending an error message.

0x02 - ILLEGAL DATA ADDRESS

The data address sent in the query is not an allowable address for the server (or slave). More specifically, the combination of reference number and transfer length is invalid. For a controller with 100 registers, a request with offset 96 and length 4 would succeed, a request with offset 96 and length 5 will generate exception 02. A request for offset 200 will also produce this exception.

But starting at 40769 and asking for 125 registers gets this 

[image]

There are regognisable parts of my Inverter serial number being returned but only the second of 2 digit pairs.

So it would appear there is an offset of 1 on all addresses which is undocumented or my hex to decimal calcs were wrong by 1.  

Will do some more testing tomorrow.

I will need to connect the ESP32 to a different WIFI AP, if I click on Wifi Reset can I then use a mobile phone to access the ESP32  AP to then scan and connect to a new Wifi AP.

tobiasfaust, 2023-06-19:
Your poll request looks quite strange.
What do you get with:

0x01 0x03 0x03 0x00 0x00 0x07
to request register 0x300 - 0x306

kommando828, 2023-06-19:
Sorry, but no idea how to poll request 0x01 0x03 0x03 0x00 0x00 0x07 in CAS.

The CAS poll request format matches in format the poll request examples in the Solax X1 Boost pdf page 10

01 is the Inverter address, 03 is function, etc

kommando828, 2023-06-19:
When I use CAS to poll the input registers then if I allow the poll end register to stray 9 registers beyond  0x434 (the final X1 Boost input register) then the error gets triggered as when I went 1 register too early on the holding registers. This is well before the MIC input registers finish so maybe the MIC setting in SolaxModbusGateway is polling for registers that the X1 Boost replies with an error message and no registers are polled. The fix for that would be a X1 Boost setting with the truncated registers.

cob555, 2023-06-19:
Who knew, it's going to be so complicated :~)

> Modbus needs to be enabled in your Admin Menu on the Inverter.

Is it? Holy crap....
I have not messed around with the inverter-settings, since they are not user-available without a password.
In the manual, I did not fund a paragraph regarding this topic.

tobiasfaust, 2023-06-20:
> Poll: [01 03 03 00 00 07](tel:01 03 03 00 00 07) which could be 0x01 0x03 0x03 0x00 0x00 0x07
> 
> [image]

Those returned values looks perfect. Just the same like in my solaxmodbusgateway with MIC settings

kommando828, 2023-06-20:
Could you clarify this I asked earlier, need it for todays testing if the rain holds off.

I will need to connect the ESP32 to a different WIFI AP, if I click on Wifi Reset can I then use a mobile phone to access the ESP32 AP to then scan and connect to a new Wifi AP.

tobiasfaust, 2023-06-20:
Oh, not recognized as a question ;)
Yes, its right

kommando828, 2023-06-20:
Ok then ;)

Test 1 will be as its is with setting set to MIC. 

Test 2 If Test 1 does not work then I will edit the MIC portion of register.h and remove the later input registers, build and load in VCS and retry.

cob555, 2023-06-20:
I'm curious, what benefit is earned from this testing (regarding the initial question)?

If the total energy-register is requested (as it is known to work) the inverter either gives a correct response, or not.

kommando828, 2023-06-20:
As long as a new copper stranded cable has fixed your issue then you are sorted. Tobias has his program listed for the MIC but not the X1-boost, however they use the same registers with the MIC having extended registers so the testing hopefully will allow him to add that it supports X1-boost.

kommando828, 2023-06-20:
What CAT number the network cable is does not really matter, just that the wire is 100% copper, stranded not solid and as thick as you can get it, 24 AWG worked for me. The first one I used is in the bin but was solid and fatigue fractured easily so was not good enough for RS485 and that was 300mm long. New one was £2.75 delivered and cuts into 2 pigtails.

Test 1 results

ID data is being received

So holding registers are being communicated.

The live data ie input registers  is not coming through, which is where the MIC map extends beyond the point at which CAS got the error message. So will edit the register.h and do test 2.
[image]

tobiasfaust, 2023-06-20:
Well done ;)
Please change the Parameter of RequestLiveData of MIC in Register.h

["#ClientID", "0x04", "0x04", "0x00", "0x00", "0x40"]

Maybe the 0x40 to 0x20 to request only 20 register instead of 40. please check your modbus doc.

the request above starts at register 0x0400

kommando828, 2023-06-20:
CAS failed at 65 registers and worked at 64, so I have changed 0x40 to 0x16 so it should work until it gets to the 5th batch of 16 registers request. If that does not work then I will also remove the input registers over position 64 so it never gets past 64.

IE

{
					"position": [75, 76, 73, 74],
					"name": "EnergyTotalToGridWh",
					"realname": "Total Energy to Grid in Wh",
					"openwbtopic": "openWB/set/pv/WhCounter",
					"datatype": "integer", 
					"factor": 100,
					"unit": "Wh"

onwards.

kommando828, 2023-06-20:
The 0x40 to 0x16 change on its own worked !!!!

Last live register is #48

[image]

kommando828, 2023-06-20:
Cobb the resister you are looking for is

0x40E Pac R AC Output power 1W Uint16

From page 7 , note that on page two the RS485 detail is

RS485 Parameter:
Parameter Value
Baud rate 9600bps
Data bit 8
Parity None
Stop bit 1
RS485 bus A(Data+); B(Data-)

[Solax Power X1-BOOST Single phase inverter Modbus-RTU comms protocolV1.0 - Public version.pdf](https://github.com/tobiasfaust/SolaxModbusGateway/files/11799282/Solax.Power.X1-BOOST.Single.phase.inverter.Modbus-RTU.comms.protocolV1.0.-.Public.version.pdf)

tobiasfaust, 2023-06-20:
> 
> ```
> void loop(){
> byte request[] = {0x423,0x424}
> }
> ```
> 

Better 

0x01,0x04,0x04,0x23,0x00,0x02

kommando828, 2023-06-20:
I am not a programmer, just someone that takes existing work and I try to mess it up without understanding what its doing ;)

kommando828, 2023-06-20:
This is how it looks, I only made the change you suggested but made the 20 registers at a time you suggested 16 just due to it dividing into 64 exactly. Not sure if that made it work any better than 20.

"Solax-Mic": {
		"config": {
			"RequestLiveData": [
				["#ClientID", "0x04", "0x04", "0x00", "0x00", "0x16"]
			],
			"RequestIdData": ["#ClientID", "0x03", "0x03", "0x00", "0x00", "0x14"],
			"ClientIdPos": 0,
			"LiveDataFunctionCodePos": 1,
			"LiveDataFunctionCode": "0x04",
			"IdDataFunctionCodePos": 1,
			"IdDataFunctionCode": "0x03",
			"LiveDataStartsAtPos": 3,
			"IdDataStartsAtPos": 3,
			"LiveDataErrorPos": 1,
			"LiveDataErrorCode": "0x84",
			"IdDataErrorPos": 1,
			"IdDataErrorCode": "0x83",
			"LiveDataSuccessPos": 1,
			"LiveDataSuccessCode": "0x04",
			"IdDataSuccessPos": 1,
			"IdDataSuccessCode": "0x03"
		},
		"data": {
			"livedata": [
				{
					"position": [3, 4],
					"name": "PvVoltage1",
					"realname": "Pv Voltage 1",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [5, 6],
					"name": "PvVoltage2",
					"realname": "Pv Voltage 2",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [7, 8],
					"name": "PvCurrent1",
					"realname": "Pv Current 1",
					"datatype": "float",
					"factor": 0.1,
					"unit": "A"
				},
				{
					"position": [9, 10],
					"name": "PvCurrent2",
					"realname": "Pv Current 2",
					"datatype": "float",
					"factor": 0.1,
					"unit": "A"
				},
				{
					"position": [11, 12],
					"name": "GridVoltage_R",
					"realname": "Grid Voltage L1",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [13, 14],
					"name": "GridVoltage_S",
					"realname": "Grid Voltage L2",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [15, 16],
					"name": "GridVoltage_T",
					"realname": "Grid Voltage L3",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [17, 18],
					"name": "GridFrequency_R",
					"realname": "Grid Frequency L1",
					"datatype": "float",
					"factor": 0.01,
					"unit": "Hz"
				},
				{
					"position": [19, 20],
					"name": "GridFrequency_S",
					"realname": "Grid Frequency L2",
					"datatype": "float",
					"factor": 0.01,
					"unit": "Hz"
				},
				{
					"position": [21, 22],
					"name": "GridFrequency_T",
					"realname": "Grid Frequency L3",
					"datatype": "float",
					"factor": 0.01,
					"unit": "Hz"
				},
				{
					"position": [23, 24],
					"name": "GridCurrent_R",
					"realname": "Grid Current L1",
					"datatype": "float",
					"factor": 0.1,
					"unit": "A"
				},
				{
					"position": [25, 26],
					"name": "GridCurrent_S",
					"realname": "Grid Current L2",
					"datatype": "float",
					"factor": 0.1,
					"unit": "A"
				},
				{
					"position": [27, 28],
					"name": "GridCurrent_T",
					"realname": "Grid Current L3",
					"datatype": "float",
					"factor": 0.1,
					"unit": "A"
				},
				{
					"position": [29, 30],
					"name": "Temperature",
					"realname": "Temperature",
					"datatype": "integer",
					"unit": "&deg;C"
				},
				{
					"position": [31, 32],
					"name": "OutputPower",
					"realname": "Output Power",
					"datatype": "integer",
					"unit": "W"
				},
				{
					"position": [33, 34],
					"name": "InverterStatus",
					"realname": "Inverter Status",
					"datatype": "integer",
					"mapping": [[0,"WaitMode"],[1,"CheckMode"],[2,"NormalMode"],[3,"FaultMode"],[4,"PermanentFaultMode"]]
				},
				{
					"position": [35, 36],
					"name": "GridPower_R",
					"realname": "Grid Power L1",
					"datatype": "integer",
					"unit": "W"
				},
				{
					"position": [37, 38],
					"name": "GridPower_S",
					"realname": "Grid Power L2",
					"datatype": "integer",
					"unit": "W"
				},
				{
					"position": [39, 40],
					"name": "GridPower_T",
					"realname": "Grid Power L3",
					"datatype": "integer",
					"unit": "W"
				},
				{
					"position": [41, 42],
					"name": "PowerPvTotal",
					"realname": "Erzeugungsleistung PV Gesamt",
					"datatype": "float",
					"unit": "W"
				},
				{
					"position": [43, 44],
					"name": "PowerPv1",
					"realname": "Erzeugungsleistung Pv1",
					"openwbtopic": "openWB/set/pv/1/W",
					"datatype": "float",
					"unit": "W"
				},
				{
					"position": [45, 46],
					"name": "PowerPv2",
					"realname": "Erzeugungsleistung Pv2",
					"openwbtopic": "openWB/set/pv/2/W",
					"datatype": "float",
					"unit": "W"
				},
				{
					"position": [75, 76, 73, 74],
					"name": "EnergyTotalToGridWh",
					"realname": "Total Energy to Grid in Wh",
					"openwbtopic": "openWB/set/pv/WhCounter",
					"datatype": "integer", 
					"factor": 100,
					"unit": "Wh"
				},
				{
					"position": [75, 76, 73, 74],
					"name": "EnergyTotalToGridKwh",
					"realname": "Total Energy to Grid in KWh",
					"datatype": "float", 
					"factor": 0.1,
					"unit": "kWh"
				},
				{
					"position": [79, 80, 77, 78],
					"name": "EtodayToGrid",
					"realname": "Today Energy to Grid",
					"datatype": "float",
					"factor": 0.1,
					"unit": "kWh"
				},
				{
					"position": [121, 122, 119, 120],
					"name": "GridPower",
					"realname": "Grid Power",
					"openwbtopic": "openWB/set/pv/W",
					"datatype": "integer",
					"unit": "W"
				},
				{
					"position": [125, 126, 123, 124],
					"name": "feedinEnergy",
					"realname": "FeedIn Energy",
					"datatype": "float",
					"factor": 0.01,
					"unit": "kWh"
				},
				{
					"position": [129, 130, 127, 128],
					"name": "ConsumeInEnergy",
					"realname": "ConsumeIn Energy",
					"datatype": "float",
					"factor": 0.01,
					"unit": "kWh"
				}
			],
			"id": [
				{
					"position": [3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14],
					"name": "InverterSN",
					"realname": "Inverter SerialNumber",
					"datatype": "string"
					}
				]
		}
	},

tobiasfaust, 2023-06-20:
hmm, requesting 16 register means 32 returned values.... How it will be 48 returned values??

However, by returning 48 values, all records after position 48 makes no sense, isn´t it?

the last one: 

"position": [45, 46],
"name": "PowerPv2",

kommando828, 2023-06-20:
Ver 3.1 has 40 instead of the 16, but the last position for the MIC is 128 which is after the 40, so after the first 40 are polled I assumed then the next 40 would be polled and so on until 128 was reached.

kommando828, 2023-06-20:
I entered a request for higher live data, comes back as undefined.
[image]

cob555, 2023-06-20:
> 0x01 -> ClientID
0x04 -> FunctionCode
0x04, 0x23 -> you want to start reading from register 0x423
0x00, 0x02 -> you want read only 2 register (0x423 and 0x424)

Thank you, that's somewhat relatable, apart from "0x00, 0x02"
According to the documentation, these bits mean "number of points high" and "number of points low".

I still get no meaningful result: 7F 5F FF (stays the same, even if the counter is incrementing a full kWh)

By the way, is it correct, that Rx and Tx LEDs of the RS485-TTL-converter (this type [https://imgaz.staticbg.com/images/oaupload/ser1/banggood/images/17/9E/29cbe5b1-42e2-4044-9d05-9c3e4fd95ff8.JPG](url) ) are blinking simultaneously (at least to the bare eye)?
It's a brand new unit, but it will not hurt, if I test it on a SDM120 meter, which I know has a working MODBUS-output.

tobiasfaust, 2023-06-21:
Both LEDs have to blink together, then its correct wired.
If you request above hex string in my arduino example, then you should get back a response from your inverter started with this: 0x01 0x04

cob555, 2023-06-21:
Today I've had a 'proper' laptop for testing-purposes just to find out, that I don't get any reply (no flashing Rx-LED) of the inverter anymore at all...what the...
Toggling in the menu between "MODBUS enable" and "MODBUS disable" makes also no difference.

The test of the RS485-converter at a SDM120 went positive though.

cob555, 2023-06-21:
> Firmware version below use a non standard protocol over RS485, they do no speak Modbus.
Known Serial Numbers
XB3
XMA
XM3

Have checked my unit, serial number XB323....

In the menu I found "Remote control-->Modbus" and "ModbusRTU" (or vice versa). Can only one of them be enabled?

tobiasfaust, 2023-06-22:
ModbusRTU  seems as the right one.
Or another way, Its possible to upgrade the firmware? Maybe with a newer firmware modbus support is enabled by default?

tobiasfaust, 2023-06-22:
> Ver 3.1 has 40 instead of the 16, but the last position for the MIC is 128 which is after the 40, so after the first 40 are polled I assumed then the next 40 would be polled and so on until 128 was reached.
> 
> "position": [129, 130, 127, 128], "name": "ConsumeInEnergy", "realname": "ConsumeIn Energy", "datatype": "float", "factor": 0.01, "unit": "kWh"

Hi,
To get all 128 values, i have to request 40 Register? Its Right? Precondition is Version 3.1 ?

`"RequestLiveData": [
["#ClientID", "0x04", "0x04", "0x00", "0x00", "0x16"]
], `

I‘m a little bit surprised anyway because by requesting 1 register you will get 2 values (2x 8bit -> a 16bit value)
And i dont know what has happened to get 128 values by requesting only 16 (should be 32 values get back) or 40 register (should be 80 register get back)

kommando828, 2023-06-22:
_how is your current working register.h section? Its identical with #19 (comment) ??_

Its taken directly from the folder the last build and load to ESP32 which is running and the screenshots are of this so yes identical.

 I‘m a little bit surprised anyway because by requesting 1 register you will get 2 value

This RS485 is new to me but from looking at the Solax pdf then the ID is held in 7 registers but each register holds 2 characters to give the 14 digit SN.

kommando828, 2023-06-22:
_Its possible to upgrade the firmware?_

Yes vis Solax support. I am on 1.42 but the PDF says 1.37 which matches the info ref the HA Solax integration Wiki. In the settings menu under info it gives you the firmware version 

_Maybe with a newer firmware modbus support is enabled by default?_

I can't remember if my setting was enabled by default but it was a setting I looked at as soon as I got the inverter hooked up to AC and 100v DC.

When my wire was the old network cable I got the flashing RX/TX lights to say the wiring was correct and working, but it wasn't working.

tobiasfaust, 2023-06-22:
> _I‘m a little bit surprised anyway because by requesting 1 register you will get 2 value_
> 
> This RS485 is new to me but from looking at the Solax pdf then the ID is held in 7 registers but each register holds 2 characters to give the 14 digit SN.

yes, 1 register holds 2 values. 1 character needs 8bit -> 1 value. So 1 register is able to hold 2 characters

kommando828, 2023-06-22:
Of course there is always the possibility Solax have not followed RS485 standards and the response is incorrect ie they are returning more registers than requested. E.g. would you expect an error message to be thrown back once a register poll goes past 64 registers. On your Solax if set the polling to outside the expected live data range do you get the same error message.

cob555, 2023-06-22:
I still don't get any response from the inverter. The Modbus-signal looks fine, 'crisp' edges and about +/-3V amplitude.
No idea, why it was blinking simultaneously the other day, and that still seem a bit suspicious to me. Tx blinking first, and with a little delay Rx for the response makes more sense (the SDM120 behaves this way).

tobiasfaust, 2023-06-23:
With the example sketch you can do anything by changing request string or whatever, standard or non-standard modbus protocols.
But the most important fact is, your inverter must reply a valid and successfully answer ;)

cob555, 2023-06-23:
And your code also works with a RS485-converter without RTS-pin?

I tried to get some data from a SDM120 with this code, also with no success. However I'm not sure how to deal with the checksum-part of the code.
[image]
Neither a request with or without error check "0x71,0xCB" gives a response.

kommando828, 2023-06-23:
I had a RS485 Janitza meter arrive yesterday, hooked up my USB RS485 to PC and meter and got nothing. Tested voltage at the RS485 and got 0.73 volts, not good. Then the only tool left in the box, until replacements arrive, which is a RS485 to TCP/IP converter and hooked that up, PC could connect over LAN. Tested voltage at RS485 at converter and got 3V so your 3V should work.

Has your program connect successfully to any RS485 equipment, if not then I would use Tobias code.

cob555, 2023-06-23:
This is the code which works fine for the SDM120: [https://github.com/masteripul/sdm120/blob/main/sdm120.ino](url)
If I change one of the read-registers to
`getRTU(0x0423);`
gives no response.

As mentioned, with Tobias code I don't get a response from the SDM120 either.

What hardware do you use for your testing?

kommando828, 2023-06-23:
I would not do it as it adds another variable and you want to reduce not increase the variables but what I used, as my inverter is outside on a wall, is a Hi-flying EW11.  This connects to the RS485 and set a server will broadcast Modbus on a LAN and then I used CAS to poll registers as per previous posts. Alternatively you can have a second EW11 on an RS485 master with the EW11 set as a client to convert the LAN signals back to RS485. But I have used them for years to connect outside inverters and they have their own foibles such as the WAN settings dialog box actually where you put the LAN settings. Once CAS showed the RS485 was working I then put the ESP32 connected directly to the Solax Boost RS485 wires.

kommando828, 2023-06-24:
Read post 1 in this closed issue.

https://github.com/tobiasfaust/SolaxModbusGateway/issues/1

Add a wire from the ref GND pin on the X1-Boost RJ45 to a GND pin on the RS485/TTL board and if the RS485 board supports it add a direction wire.

The ref GND will reduce noise on the RS485 signal.

cob555, 2023-06-24:
It's ALIVE!
Finally a success, not with Tobias script, but with a "USB-Modbus-sniffer" and a proper Modbus-decoding-software mind you. And without a additional GND-connection, but since it doesn't hurt, and comes for free in a RJ45-cable, I will connect it later.

That's the response for 50 registers, starting at 0x400.
[image]

And here only the request for 'total yield' at register 423 (accidentally started reading at register 422, since the software is adding 1 to the decimal value of the hex-number, which got me a bit confused).
[image]
It show's the correct value of "31614" which is 3161.4kWh

Now I know, which request gives a correct response. The request generated by the ESP32 was way shorter, if memory serves me correct.

So far, so good. _Only_ thing left, _convincing_ the ESP32 to generate the correct request-string (easier said than done, I'm afraid)

tobiasfaust, 2023-06-25:
Your first schreenshot shows your polling is the Same like the „MIC“ settings. That live registers start at 0x0400 too.

Please exchange the Same request string in example sketch, it should answer the same 

0x01 0x04 0x04 0x00 0x00 0x032

If that works you can continue with your specific 0x0423 Register.

[8 more replies]

closed 2024-02-07 by tobiasfaust