tobiasfaust/SolaxModbusGateway #20: [222556][E][WiFiSTA.cpp:317] begin(): connect failed! 0x300a

manatwork81, 2023-06-18:
Hi!
I am a Newbie with ESP32 and shopped any from AZ Del.... following Link
https://www.az-delivery.de/en/products/esp32-developmentboard

[image]
[image]

Is there anything incompatible?
Thanks for Help

kommando828, 2023-06-18:
I bought my ESP32 form AZ, it is an esp32-wroom-32 which has a 4MB memory and I had issues with the flash tool and the ESP32 would not boot properly. I used Visual Studio Code instead and this worked. Looking a bit further into this but only for curiosity I believe that the requirement for the memory registers to load the Bin files too means you have to have the exact same ESP32 as Tobias used or an exact copy. If like me you go for a different type of ESP32 then you either need the revised memory registers for your different board or use VSC as this loads the program without having to specify the memory register locations. I did a factory reset on my ESP32 and the restore file was a different size from the std ESP32, hence my belief the memory registers are different for different boards. How you find these out is unknown to me.

tobiasfaust, 2023-06-18:
Yea, its a little bit annoyed due different ESP32 Types

https://gist.github.com/sekcompsci/2bf39e715d5fe47579fa184fa819f421

i use the „Real“ ESP32, @kommando828 has Right, in Case of faults its safer to use Visual Studio Code

tobiasfaust, 2023-06-23:
fixed with dc3243a

tobiasfaust, 2023-06-23:
fixed with dc3243a

manatwork81, 2023-07-01:
Sorry for late answer, but there was no time to try.
VC --> same error, with latest Firmware
[image]

i have exactly this ESP.

Thanks for HelP ;)

kommando828, 2023-07-01:
Same as mine except the Serial chip is different which only affects the Driver needed to connect. the link is to a different fix and will not allow use of Flash download tool, you still need to use Visual Studio Code on a download of the code, load the folder, build and then upload to your ESP32.

manatwork81, 2023-07-01:
Now i have send it directly from VSC
1) i downloaded the latest code from Git
2) i erased the ESP32
3) i load the Filesystem via VSC
4) i upload the Code via VSC
same result.

you mean,

i should compile the files and upload with the Flash Tool?

kommando828, 2023-07-01:
No, you connect the ESP to a USB port on the PC running VSC, on mine VSC auto recognises it as an ESP and on selecting load sends the code to the ESP32. That way no memory points are needed as VSC sorts it out itself.

manatwork81, 2023-07-01:
Oh thanks, did you have an link to the factory Firmware?

kommando828, 2023-07-01:
Second link is the one I used but first I had to wipe the flash clear using this youtube

https://www.youtube.com/watch?v=3oEvXhgHZHo

using the link to erase and then used the bin file in below zip

https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Binary_Lists/ESP32_AT_binaries.html

https://dl.espressif.com/esp-at/firmwares/esp32/ESP32-WROOM-32/ESP32-WROOM-32-AT-V2.4.0.0.zip

manatwork81, 2023-07-02:
I have erased all &  installed factory Firmware -> worked
I have installed from VSC -> first was a spiffs error, after reboot there was the same error.

I have erased all &  installed factory Firmware -> worked
i have tried esptool 0x1000 bootloader / 0x8000 Partitions / 0x100000 Firmware
now there is a bootloop

My Device = ESP32-D0WD-V3 (revision 3)

Output Firmware:

[image]

i have no idea whats going wrong

kommando828, 2023-07-02:
I can only suggest you look for ESP32-D0WD-V3 (revision 3) Factory Firmware and try that.

Also read this

https://community.platformio.org/t/esp32-d0wd-v3-revision-3-how-to-properly-configure-platformio-ini/26676

Or just buy a bog standard ESP32 as AZ seem to mislabel what they sell.

manatwork81, 2023-07-02:
Have anyone a link to a working Product, that can delivered to Austria. Then i will buy an other Chip.
Thanks vor Help

tobiasfaust, 2023-07-02:
You Need a Real ESP32 Wroom32, 
Not a S2 or S3 light Edition ;)

manatwork81, 2023-07-02:
Is this a REAL one ? 

[Link](https://www.amazon.de/Espressif-Development-Bluetooth-WROOM32-NodeMCU/dp/B07K68RQTS/ref=asc_df_B07K68RQTS/?tag=googshopde-21&linkCode=df0&hvadid=407342190625&hvpos=&hvnetw=g&hvrand=1390348137999264596&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=20043&hvtargid=pla-862519322535&psc=1&th=1&psc=1&tag=&ref=&adgrpid=85377202485&hvpone=&hvptwo=&hvadid=407342190625&hvpos=&hvnetw=g&hvrand=1390348137999264596&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=20043&hvtargid=pla-862519322535)

kommando828, 2023-07-02:
Best to ask the seller. only they can know what they are supplying currently.

tobiasfaust, 2023-07-02:
> Is this a REAL one ? [...]

It Looks well

manatwork81, 2023-07-02:
Okay, i have ordered one of these.

The best way to flash with VSC?
Load zip from Github
open in VSC
and Upload?

sorry, but i have only worked with esp8266, this was much more easier.

Thanks

tobiasfaust, 2023-07-02:
Please read the Wiki ;)

https://github.com/tobiasfaust/SolaxModbusGateway/wiki/build-with-gitpod

kommando828, 2023-07-02:
Check the chip when it arrives and then if it is as described by seller use Gitpod, if not then send it back or use VCS. But make sure you know if it matches sellers description before using Gitpod or you will be back into wiping and then factory firmware loading.

manatwork81, 2023-07-10:
New esp tested, same result :(

I have took the first one for a Test, i loaded ESPEasy from Gibhub an tryed to install is MY ESP32,
Itook this config: normal_ESP32_4M316k   -> it worked. ;)
i have tried to install S2/S3/C3 from ESPEasy, but this is not possible.

I got a lot of helpful tips from you and have already tried a few things. Unfortunately, it is very difficult to find suitable information about the partition on the Internet.
Could it be that a certain part of the software is simply being flashed into the wrong department?

Excerpt from ESPEasy - File: esp32_partition_app4096k_spiffs1024k.csv

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x400000,
app1,     app,  ota_1,   0x410000,0x400000,
eeprom,   data, 0x99,    0x810000,0x1000,
spiffs,   data, spiffs,  0xf00000,0x100000,

Thank you all ;)

tobiasfaust, 2023-07-10:
A new esp, the Accesspoint was opened? Did you entered your wifi credentials? 

The error Message shows you, the the esp cannot connect to your wlan. It is opening an accesspoint in that Case.

tobiasfaust, 2023-07-11:
Sounds wonderful:)

closed 2023-07-11 by tobiasfaust