tobiasfaust/SolaxModbusGateway #22: Seit neuem Update keine OpenWB Daten mehr bei X3 Hybrid

MagicSven81, 2023-06-28:
Ich bekomme bei Auswahl des X3 Hybrid WR keine Möglichkeit mehr OpenWB Daten zu markieren.
Stelle ich auf X1 Hybrid oder MIC um, sind diese wieder zu sehen.
Bitte einmal prüfen - Danke.

[image]

tobiasfaust, 2023-06-28:
Ist hier schon berichtet worden #21 
Ist leider ein Speicherüberlauf der etwas dauert zu beseitigen.
Als workaround bis dahin hilft es in der Register.h die Anzahl der Items für den X3 zu reduzieren

tobiasfaust, 2023-06-29:
schneller Fix mit e8cdade um den Absturz des ESP wegen Speicherüberlaufes zu verhindern.
Aber Achtung: Die Infos ob ein Item OpenWB fähig ist oder nicht (spalte OpenWB) wird noch nicht angezeigt, kommt später

tobiasfaust, 2023-06-29:
> Ok danke, wie entferne ich unnötige Items aus der register.h ? Auf was muss ich da achten?

Mit dem letzten commit nicht mehr notwendig

DJFS, 2023-06-30:
Heute hab ich ein Update gemacht. Nun geht nach einmal MQTT Übertragung immer gar nichts mehr :(

tobiasfaust, 2023-07-01:
Ich Check das Montag, 
Kann ich mir nicht wirklich erklären da ich ausschließlich an der Web-Ausgabe gearbeitet hatte

Bitte noch mehr Details: gibt es Probleme nur beim X3? 
Wann hört MQTT auf?

tobiasfaust, 2023-07-03:
> Heute hab ich ein Update gemacht. Nun geht nach einmal MQTT Übertragung immer gar nichts mehr :(

getestet, kann ich nicht nachvollziehen, MQTT messages der aktiverten Items werden kontinuierlich übertragen

tobiasfaust, 2023-07-05:
finally fixed with 6adb6d2

closed 2023-07-05 by tobiasfaust