tobiasfaust/SolaxModbusGateway #24: Keine Daten mehr vom Wechselrichter

DJFS, 2023-07-12:
Hallo,  
nach einen kompletten neuflashen des ESP32 lief es ca. 10 Tage ohne Probleme, Jedoch bekomme ich jetzt seit 3 Tagen keine Werte mehr vom Wechselrichter per Modbus.
Weboberfläche und MQTT gehen. 

_

_```
<html>
<body>

Typ | Raw Data
-- | --
RawData of ID-Data |  
RawData of Live-Data | undefined undefined undefined undefined undefined undefined undefined undefined undefinedundefined undefined undefined undefined undefined undefined undefined undefined undefined

</body>
</html>

_
```_

tobiasfaust, 2023-07-12:
Restarten und beobachten….
Ein Log (Level 5) zum Zeitpunkt des Abrisses wäre hilfreich.
Konnte ich bisher auch nicht nachvollziehen

DJFS, 2023-07-13:
ein Reset hat nichts gebracht,  evtl neu flashen?
woher bekomme ich den log?

Die Daten von der letzten MQTT übertragen kann ich anbieten:

[image]

tobiasfaust, 2023-07-14:
Log bekommt man über die serielle Konsole. Also ESP32 an den PC anschließen und über den seriellen Port das Log (Level 4 oder 5) beobachten

DJFS, 2023-07-14:
[image]

Es ging mehrere Tage ohne Probleme und am 11.07. 07:57:08 war Schluss.

DJFS, 2023-07-14:
Hab nun den ESP neu geflasht,
der AP "AP_Solax"  hatte ein Passwort?  Kenne aber keins

tobiasfaust, 2023-07-14:
Sorry, Passwort im WIKI hinzugefügt. Das Passwort ist in einem der letzen changes hinzugekommen.

https://github.com/tobiasfaust/SolaxModbusGateway/wiki/start-and-integration-into-locale-wifi

tobiasfaust, 2023-07-14:
Laut log sieht es so aus als das du items aktiviert hattest, aber vergessen zu speichern. Deswegen lief es am Anfang.
Nach einem reboot sind keine items mehr aktiviert weil die Config nicht gespeichert wurde. Einfach mal auf der weboberfläche wieder aktivieren und speichern nicht vergessen.

DJFS, 2023-07-27:
Ich habe bisher nichts weiter gemacht. Ich bekomme es nicht zum laufen. Vllt hole ich mir einfach den original Stick. Oder ich schau mal nach Tasmota o.ä.

tobiasfaust, 2023-07-27:
Und du hast komplett von vorne nochmal angefangen? Mit dem aktuellen Repo stand aus dem Development Branch?

lexxmm, 2024-03-28:
klasse Ergebnisse hier...habe genau das gleiche Problem...von jetzt auf gleich keine Daten mehr...

tobiasfaust, 2024-03-28:
Etwas mehr Details wären sicher hilfreich.

lexxmm, 2024-03-28:
Genau wie bei DJFS...von heute auf morgen keine Daten mehr. Nur undefined undefined undefined undefined undefined undefined undefined undefined undefined
undefined undefined undefined undefined undefined undefined undefined undefined undefined

log daten: 

Ajax Json Antwort: 
Query Live Data into Queue:
0x01 0x04 0x04 0x00 0x00 0x40 
Request queue data to inverter: 0x01 0x04 0x04 0x00 0x00 0x40 0xF0 0xCA 
Read Data from Queue: 
no response from client
Parse 0 Bytes of data
Ajax Json Empfangen: {
  "action": "RefreshLiveData"
}

WR ist ein X3-MIC-4K-G2  

Es wurde NICHTS geändert. Habe stundenlang alles probiert und echt keine Lust mehr :(

lexxmm, 2024-03-28:
genau..kein Update. Ausschalten (gewisse Zeit) von beiden Geräten habe ich schon versucht...hat nix gemacht. Also ist der WR defekt?

tobiasfaust, 2024-03-28:
Ich kann mir nicht vorstellen das der WR kaputt gegangen ist.  Vielleicht haben sich die Modbus Einstellungen des WR verstellt?
Da kann ich nicht wirklich helfen ….

DJFS, 2024-03-28:
Hallo lexxmm,

ich habe den esp einmal komplett gelöscht und neu geflasht,  nach den ic dann alle Einstellungen wieder drin hatte lief er wieder

tobiasfaust, 2024-03-29:
Das wäre dann schon sehr „strange“ ;)
Meiner läuft seit genau „ 164 Days, 18 Hours, 51 Minutes“ durch ohne jegliche Probleme…..

lexxmm, 2024-03-29:
ok scheinbar wirklich WR defekt...jetzt arbeitet er nämlich gar nicht mehr. Nach ein paar Sekunden kommen komische Zeichen auf dem Display und er startet neu.