tobiasfaust/SolaxModbusGateway #26: Ethernet support for WT32-ETH01

kagehisa, 2023-07-27:
Is it planned to have an ethernet port of this gateway for the  WT32-ETH01 board? I know this is a niche request but some people have only ethernet access to their solar inverters. 
Is there something planned?

tobiasfaust, 2023-07-27:
Hi,
Not planned so far. But its interesting use Case. I ordered one to test it. Please be patient some weeks.

kagehisa, 2023-07-27:
This is already more than I expected. Thanks for the quick positive reply!

kagehisa, 2024-02-01:
Hi, has there been any progress on this?

tobiasfaust, 2024-02-07:
new release 3.2.0 is ready. Now I´m working on this

kagehisa, 2024-02-07:
Thank you! If there is anything I can help with let me know. If it is wihin my capabilities I'm happy to help.

tobiasfaust, 2024-02-09:
@kagehisa,  please test it

kagehisa, 2024-02-09:
I will do as soon as my rs485 adapter board arrives.

kagehisa, 2024-02-11:
@tobiasfaust, I checked out the dev branch and launched:
- a build via platformIO `firmware_ESP32/General/build`
- as well as a filesystem build via `firmware_ESP32/Platform/Build Filesystem Image`

This went without a hitch. But when I try to upload it with:
```
esptool -p /dev/ttyUSB0 -b 921600 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x8000 partitions.bin 0x1000 bootloader.bin 0x10000 firmware.bin 0x180000 littlefs.bin
```

I get the following error:

>esptool.py v2.8
>Serial port /dev/ttyUSB0
>Connecting.......
>Chip is ESP32D0WDQ5 (revision 1)
>Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
>Crystal is 40MHz
>MAC: c4:de:e2:b2:1f:f8
>Changing baud rate to 921600
>Changed.
>Enabling default SPI flash mode...
>Configuring flash size...
>
>A fatal error occurred: File littlefs.bin (length 1441792) at offset 1572864 will not fit in 2097152 bytes of flash. Use --flash->size argument, or change flashing address.

Am I using the wrong firmware target in PlatformIO? I guessed firmware ESP32 should work based on the docs.

tobiasfaust, 2024-02-11:
Der Parameter —flash_size muss für das Board angepasst werden
Check mal die Ausgabe von:
`esptool.py --port PORT flash_id `
Wenn dort sowas steht wie: 
`Detected flash size: 4MB `
Kannst du den Parameter entsprechend anpassen.

kagehisa, 2024-02-11:
Ok, thanks! 4MB flash size worked. I'll do the Modbus connection tomorrow.

kagehisa, 2024-02-13:
Ok, thanks! 4MB flash size worked. I'll do the Modbus connection tomorrow.

Edit:
Connected the device to my WIFI, but switching to Ethernet is not working. It is not taking over any settings at all. As soon as I save, the page gets reloaded and the option for WIFI is set again. Same for all the other settings.

@tobiasfaust : Where did you look up the required Pin numbers for connecting the RS485 module? The code states `RX=2`, `TX=4` and `RTS=5`. But this does not make much sense with the pin configuration in this [document](https://files.seeedstudio.com/products/102991455/WT32-ETH01_datasheet_V1.1-%20en.pdf). I have something in the back of my mind that the espressifIDE Pin convention was somehow different from what you would expect from the datasheets.

tobiasfaust, 2024-02-14:
You can change pin assignment via settings what you want.
Can you Check seriell console for log Output? At my side Ethernet works well.

kagehisa, 2024-02-14:
Nope that is my problem, I can't. Whenever I try to change a setting and hit save, the page gets a reload and the Settings are gone. Since debug level is initial at zero, there is nothing on the debug out terminal either. I'll try to rebuild and reflash, could be an issue with the littlels.bin, since the webpage is not shown correctly (style sheet) either...

tobiasfaust, 2024-02-14:
Can you please Open the Webpage in source Mode and try  to Open the /web/Javascript.js separately.  Its ok?

please Open the Page of files section, which files in each folder will show?
Please Post some screenshots of files,- baseconfig,- modbus section

kagehisa, 2024-02-14:
Hi, thanks for the hint with the Javascipt.js. Source mode tells me it is not finding anything:
[image]

That is what I get for the files section:
[image]

But it is weird, when I build the fs image I get the following output from platformIO:
>Building in release mode
Building FS image from 'data' directory to .pio/build/firmware_ESP32/littlefs.bin
/regs/Deye_SUN_SG04LP3.json
/regs/Solax-MIC.json
/regs/Solax-X3.json
/regs/Solax-X1.json
/regs/Growatt-SPH.json
/regs/Sofar.json
/web/Javascript.js
/web/style.css
/web/UpdateResponse.html
/web/PageHandleFS.js
/web/UpdatePage.html
/web/JsAjax.js

So the files should be there.

tobiasfaust, 2024-02-14:
hhh, no files are in your filesystem. Can you upload littlefs.bin again? You can check it by webpage "files"

kagehisa, 2024-02-14:
Alright! That did the trick, had to replace `partition-table.bin` with `partitions.bin` and `firmware_file.bin` with `firmware.bin` but thats about it. I can now change settings. I will do the other tests from here on out.

tobiasfaust, 2024-02-14:
Oh yes, that is only a template
I changed that in Wiki to the right one.

kagehisa, 2024-02-19:
Sorry it took me so long, I came down with a pretty bad flu.
Switching to Ethernet works fine, I configured the board connections and was able to "talk" to my solax X3 inverter.
So from my side everything works fine. Thanks again for the effort!

Do you want me to add a small sentence regarding flashing the wt32-eth01 board with the greater flash size and the changed flash adress for the `littlefs.bin` in the corresponding wiki page?

tobiasfaust, 2024-02-19:
Hi sounds well.
Thanks for your offer, what should I Write down at the Wiki Page?

kagehisa, 2024-02-19:
I would just specify that this board requires the 4 MB flash size instead of 2 and the required littlefs.bin adress change.

Or just like this:
For WT32-ETH01 (and similiar boards) use the following command to flash the device via the `esptool`:
```
esptool -p /dev/ttyUSB0 -b 921600 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x8000 partitions.bin 0x1000 bootloader.bin 0x10000 firmware.bin 0x290000 littlefs.bin
```

closed 2024-02-20 by tobiasfaust