tobiasfaust/SolaxModbusGateway #29: X3 "nicht-Hybrid"

thakikka, 2023-08-30:
Hallo,

ich habe einen Solax X3-8.0-T-D. Klappt es mit dem auch?

Als Laie habe ich nun einen fertigen Adapter eines Users gekauft, die mittels eines Lan-Splitters zusammen mit einem SmartMeter im Zählerschrank an den WR angeschlossen.

Ich bekomme aber leider keine Daten vom WR bzw. in "Raw Data" nur "Undefined".

siehe hier (mit Screenshots): https://www.photovoltaikforum.com/thread/177671-solax-x3-mic-modbus/?pageNo=3

tobiasfaust, 2023-08-30:
Ich würde einfach mal behaupten das das Modbus Protokoll des X3 Hybrid sich zu deinem normalen X3 unterscheidet und dein WR nur Müll zurückgibt weil er die Anfrage nicht versteht.

Kannst ja mal zum Test die anderen verfügbaren WR aus der Auswahlliste ausprobieren ob bei einem etwas sinnvolles zurückkommt.

weiterhin halte ich es für höchst zweifelhaft das am selben Modbus com Port deines WR sowohl der smartmeter als auch der ESP funktioniert. Es darf immer nur einen Modbus Master geben, der ESP ist einer. Und der WR ist auch Master der den Smartmeter abfragt.

thakikka, 2023-08-30:
[image]

Klappt leider auch nicht. Spannenderweise wird der Modbusadapter selbst in ioBroker erkannt, aber es kommen einfach keine (lesbaren) Daten an:

[image]

kommando828, 2023-08-30:
You need to ask Solax support for the modbus register for your exact model and firmware and then compare it to the ones already supported by Tobias. Also you may need to go into the inverters settings and turn on Modbus RTU and set the address to 1 as they may not be correctly set by default.

thakikka, 2023-08-30:
Thank you for your answers.

In the inverter settings, the address is already set to 1. In the meantime also on test to 2.

I have today written to Solax support with the request for a firmware update. I will write to them again and ask for a suitable modbus register.

kommando828, 2023-08-31:
Tobias has included a selection of the Solax Modbus pdf's

https://github.com/tobiasfaust/SolaxModbusGateway/tree/development/docs

So once you have the one from Solax for your inverter and can check against these to see the closest match.

Their naming on the inverters and the model numbers on the inverters don't always match exactly which causes some confusion. Not including the Generation of the inverter is another issue.

juwild83, 2024-01-25:
Hello everyone,

Has anything already happened on this topic? I also have an X3 inverter without a “hybrid”.
Best regards

tobiasfaust, 2024-01-26:
Hi, 
unfortunally - no.
you need the right modbus protocol description document before you are able to start

juwild83, 2024-01-28:
Hello, I probably got the relevant protocols from a post from the “Photovoltaikforum”.
I just don't know enough about the subject and won't start changing the code now.

Best regards

tobiasfaust, 2024-01-28:
Can you Upload it here please?

juwild83, 2024-01-28:
[1.On-grid INV-20240125T205749Z-001.zip](https://github.com/tobiasfaust/SolaxModbusGateway/files/14075884/1.On-grid.INV-20240125T205749Z-001.zip)

Of course I can do that

kommando828, 2024-01-28:
Just be aware that Solax have several generations of each inverter but fail to identify the generation in some cases. So in this case the PDF you have pointed to is for a X3-MIC Gen1, in your zip there is also a PDF for the Gen2. So its worth taking a bit longer to do a check which Generation your inverter is.

tobiasfaust, 2024-01-28:
As I saw in your documents, all MIC X3 types hold their live dataregister ap from 0x0400 by functioncode 0x04. So if you select Solax-MIC it should be fine basicly. 
How looks the rawdata page if you select "Solax-MIC"? Please post a screenshot

thakikka, 2024-01-28:
> You need to ask Solax support for the modbus register for your exact model and firmware and then compare it to the ones already supported by Tobias. Also you may need to go into the inverters settings and turn on Modbus RTU and set the address to 1 as they may not be correctly set by default.

[solax.zip](https://github.com/tobiasfaust/SolaxModbusGateway/files/14076055/solax.zip)

I have now received an e-mail from Solax confirming that the firmware update has been installed as requested. The following files have also been sent to me. Is there anything you can do with them?

thakikka, 2024-01-28:
[image]
[image]

(I also tried 9600 as Modbus Baudrate. Same result.)

juwild83, 2024-01-28:
same for me

kommando828, 2024-01-28:
The label on the inverter should identify the Gen with the exception of Generation 1, all other Gen's should have on the label G2/G3/G4 and so on. Also check the picture on the title page of PDF matches your inverter. For the X3-MIC there only seems to be Gen 1 and Gen 2 from what I can see but other types are at Gen 4.

kommando828, 2024-01-28:
Tips when comms are dead.

1. Check baud rate is correct, for X3-MIC G1 and G2 this is 9600 but be prepared to try other baud rates.
2. Swap the two RS485 wires around, not everyone labels them correctly, Janitza is one where this works,
3. Check the cable you are using is a good quality cable, some are copper plated aluminium, even solid copper fractures internally. I always use 24awg stranded copper wire network cable. This fixed my dead comms with an X1-Boost G3.

JMyrng, 2024-01-28:
> I know that Problem. Your inverter Firmware is too Old that you cannot request 64byte in one Single request. Please try to change it.

Hi, same problem here. 
I've asked the support for a firmware update but they replied that is was already the latest version. 

X3-MIC-10.0 TD

tobiasfaust, 2024-01-28:
> > I know that Problem. Your inverter Firmware is too Old that you cannot request 64byte in one Single request. Please try to change it.
> 
> Hi, same problem here. I've asked the support for a firmware update but they replied that is was already the latest version.
> 
> X3-MIC-10.0 TD

Please do the described changes to make sure that this is the problem.

JMyrng, 2024-01-28:
> Please do the described changes to make sure that this is the problem.

I can only find the "RequestLiveData" files in the dev branch, but I compield the new firmware. Is it possible to upload that over the web portal or should I do it via ESP32 Flash download tools?

tobiasfaust, 2024-01-28:
If you uses the dev branch, you have to flash via ESP32 flash download tool (see gitpos wiki)
If you used the master branch, you can also use the web portal upload. 
Currently I redesign the file handling, because that there are differences between dev and master

JMyrng, 2024-01-28:
I currently only have access via network. So where can I change these settings in the master branch?

thakikka, 2024-01-28:
> The label on the inverter should identify the Gen with the exception of Generation 1, all other Gen's should have on the label G2/G3/G4 and so on. Also check the picture on the title page of PDF [...]

[image]

thakikka, 2024-01-28:
So, ich habe mich da jetzt mal als absoluter Laie versucht durchzukämpfen.

[image]

Die Änderung habe ich für "Solax-Mic" vorgenommen. Danach habe ich unten in der Befehlszeile mittels "platformio run" das Ganze erneut erstellen lassen. Dann habe ich aus dem Ordner .pio/build/ die Datei firmware.bin heruntergeladen und auf den Adapter als Firmware hochgeladen.

[image]

Hat offensichtlich geklappt?

Leider weiterhin keine Daten bei RawData, egal was ich probiere bei den Einstellungen.

[image]

tobiasfaust, 2024-01-29:
Wir müssen jetzt rausfinden wo es hängt!
Stell mal bitte LogLevel auf 5 (Basis Config) und poste mal bitte das Log welches auf der seriellen Konsole ausgegeben wird.

thakikka, 2024-01-29:
Entschuldige meine Unwissenheit, aber muss ich dafür den Adapter an meinen PC anschließen? Oder lasse ich ihn am Wechselrichter? Ich habe in den FAQ gelesen, dass ich "hterm" installieren soll. Bin gerade offen gesagt etwas überfordert.

thakikka, 2024-01-31:
Danke für deine Hilfe, aber ich merke leider, dass ich hier an meinem persönlichen Skill Cap angekommen bin. Ich kriege weder auf dem Laptop noch testweise am PC eine Verbindung hin.

Ist der USB-Anschluss, an dem das Ladegerät hängt, auch der USB-Anschluss, den ich mit dem PC/Laptop verbinde? EInen anderen USB-Anschluss konnte ich nicht entdecken.

tobiasfaust, 2024-01-31:
Ja genau, also anstatt usb zum Ladegerät gehts an den Laptop
Versuch dich mal mit Google schlau zu machen, da gibt es ganz viele , auch bebilderte, howtos wie man starten kann :)

thakikka, 2024-01-31:
[output_2024-01-31_15-15-08.txt](https://github.com/tobiasfaust/SolaxModbusGateway/files/14112851/output_2024-01-31_15-15-08.txt)

Ich habe es geschafft. Danke für deine Geduld.

Dummerweise hatte ich "Timestamps" aktiviert, aber ich denke, es ist trotzdem aufschlußreich. Nämlich das egal was ich währenddessen an unterschiedlichen Einstellungen probiert habe, es nie zu irgendeiner Ausgabe von Daten kam :(

Ich hatte weiter oben die ZIP-Datei hochgeladen, die mir von Solax zugeschickt wurde. Passt das denn so überhaupt mit dem Modregister?

tobiasfaust, 2024-01-31:
Dein WR antwortet nicht. "No response"
Bitte überprüfe zuerst in den Einstellungen deines WR welche ModbusID und Baudrate dieser verwendet. Danach check auch unbedingt ob du über den RJ Stecker auch die korrekten Pins mit dem Modbussignal abgreifst. Schau dazu in die Doku des WR.
Auf der RS485 Platine müssen sowohl RX als auch TX aufblinken. Dazu must du mal alles (siehe dein Bild oben) aus dem Gehäuse herausholen.

thakikka, 2024-02-01:
[image]

Ich habe das mir von Solax zugeschickte ZIP nochmal durchgeschaut und meine, dass hier dürfte die für meinen WR passende PDF sein: 
[Solax Power X3-MIC three phase inverter Modbus-RTU comms protocolV1.05 - public version.pdf](https://github.com/tobiasfaust/SolaxModbusGateway/files/14124735/Solax.Power.X3-MIC.three.phase.inverter.Modbus-RTU.comms.protocolV1.05.-.public.version.pdf)

Ich weiss nun leider nicht, ob das ggf. eine Hilfe bei der Lösungsfindung sein könnte, aber zumindest konnte ich dem PDF entnehmen, dass die Baudrate wohl 9600 sein sollte. Im WR selbst konnte ich dazu keine Info/Einstellmöglichkeit finden.

[image]

Das habe ich zur der ModbusID finden können. Ich habe irgendwo den Tipp aufgeschnappt, dass ich in Verbindung mit dem Splitter, den ich verwenden muss, weil die Verbindung zu dem Stromzähler zwingend erforderlich bestehen bleiben muss, da ansonsten der WR direkt in einen Fehlermodus geht, die ModbusID im Gateway daher auf 02 stellen müsste.

Ausprobiert habe ich inzwischen aber sämtliche Varianten, also sowohl 01 als auch 02. Stets ohne Erfolg.

Hier zur Veranschaulichung ein Bild von dem Splitter:

[image]

Liegt da ggf. der Fehler? Muss ich da etwas gesondert beachten bzw. einstellen, dass die Verbindung endlich klappt?

Die Platine würde ich mir gerne (auch von unten) anschauen, aber noch war es mir nicht möglich, diese aus dem Gehäuse zu lösen. Zur Not muss ich das Gehäuse von unten vorsichtig aufschneiden oder so...

tobiasfaust, 2024-02-01:
Ich weiß nicht wie der Splitter funktioniert.
Probier erstmal ohne Splitter direkt am WR. Egal ob der WR in einen fehlermodus geht oder nicht. Du musst eine Fehlerquelle erstmal ausschließen.

lexxmm, 2024-03-28:
Gibt es hier News? Habe das selber Problem mit einem X3-MIC-4K-G2 und empfange nur 

undefined undefined undefined undefined undefined undefined undefined undefined undefined
undefined undefined undefined undefined undefined undefined undefined undefined undefined

juwild83, 2024-03-29:
Hallo,
ich hab das für mich jetzt anders gelöst.
Habe auf ESPHome auf den "Adapter" geflasht. Die Adressen dann mit in die Konfiguration gepackt und kann alles jetzt  mit Home Assistant auslesen.

`esphome:
  name: esphome-web-0ffdc4
  friendly_name: ESPHome Solax RTU

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-C96Ba8"
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxx"

substitutions:
  name: solax-x3
  device_description: "Monitor a Solax X3 Boost via RS485"
  tx_pin: GPIO17
  rx_pin: GPIO16

ota:

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
#mqtt:
#  broker: !secret mqtt_host
#  username: !secret mqtt_username
#  password: !secret mqtt_password
#  id: mqtt_client

# api:

uart:
  id: uart_0
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

modbus:
  - id: modbus0
    uart_id: uart_0
#    flow_control_pin: GPIO0

modbus_controller:
  - id: solax0
    address: 0x1
    modbus_id: modbus0
    setup_priority: -10
    update_interval: 15s

#text_sensor:
#  - platform: modbus_controller
#    modbus_controller_id: solax0
#    name: "${name} operation mode"
[... 371 lines cut ...]
  - platform: modbus_controller
    modbus_controller_id: solax0
    name: "${name} energy today"
    address: 0x425
    register_type: read
    value_type: U_WORD
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    accuracy_decimals: 1
    filters:
      - multiply: 0.001
    
  - platform: modbus_controller
    modbus_controller_id: solax0
    name: "${name} energy today MSB"
    address: 0x426
    register_type: read
    value_type: U_WORD
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    accuracy_decimals: 1
    filters:
      - multiply: 0.001`

closed 2024-03-29 by tobiasfaust

PatrickEscher, 2024-04-01:
@juwild83:
Hallo, versuche auch verzweifelt meinen X3 Gen1 anzubinden.
kannst du genauer erklären was du genau gemacht hast?
Danke und Gruß

juwild83, 2024-04-20:
Hi, 
ich benutze Home Assistant mit ESPHome und das ist der Code:
`esphome:
  name: esphome-web-0ffdc4
  friendly_name: ESPHome Solax RTU

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable Home Assistant API
api:
  encryption:
    key: "--Dein API-Key--"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-C96Ba8"
    password: "Passwort"

substitutions:
  name: solax-x3
  device_description: "Monitor a Solax X3 Boost via RS485"
  tx_pin: GPIO17
  rx_pin: GPIO16

ota:

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
#mqtt:
#  broker: !secret mqtt_host
#  username: !secret mqtt_username
#  password: !secret mqtt_password
#  id: mqtt_client

# api:

uart:
  id: uart_0
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

modbus:
  - id: modbus0
    uart_id: uart_0
#    flow_control_pin: GPIO0

modbus_controller:
  - id: solax0
    address: 0x1
    modbus_id: modbus0
    setup_priority: -10
    update_interval: 15s

#text_sensor:
#  - platform: modbus_controller
#    modbus_controller_id: solax0
#    name: "${name} operation mode"
#    address: 0x40f
#    register_type: read
[... 369 lines cut ...]
  - platform: modbus_controller
    modbus_controller_id: solax0
    name: "${name} energy today"
    address: 0x425
    register_type: read
    value_type: U_WORD
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    accuracy_decimals: 1
    filters:
      - multiply: 0.001
    
  - platform: modbus_controller
    modbus_controller_id: solax0
    name: "${name} energy today MSB"
    address: 0x426
    register_type: read
    value_type: U_WORD
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    accuracy_decimals: 1
    filters:
      - multiply: 0.001`

PatrickEscher, 2024-09-26:
@juwild83 
hab deinen Post erst kürzlich gesehen und hatte jetzt mal Zeit das umzubauen.
Also bei mir kommt irgendwie nichts an :(
Hab auf den Adapter auch ESPHome geflasht und in HomeAssistant eingebunden.
Deine Config habe ich soweit auch übernommen (notwendige Änderungen natürlich gemacht.)

Aber irgendwie bekomme ich keine Response vom modbus zurück. Es gibt folgende Fehlermeldungen
Modbus command to device=2 register=0x400 no response received - removed from send queue
Modbus command to device=2 register=0x413 no response received - removed from send queue
Modbus command to device=2 register=0x423 no response received - removed from send queue
mehr nicht :(

Ich hab dann mal das verbose log eingeschalten
z.B.[18:08:27][D][modbus_controller:038]: Modbus command to device=2 register=0x413 no response received - removed from send queue
[18:08:27][V][modbus_controller:042]: Sending next modbus command to device 2 register 0x423 count 4
[18:08:27][V][modbus:210]: Modbus write: 02.04.04.23.00.04.01.00 (8)
[18:08:27][V][modbus_controller:565]: Command sent 4 0x423 4 send_count: 1
[18:08:27][V][modbus_controller:042]: Sending next modbus command to device 2 register 0x423 count 4
[18:08:27][V][modbus:210]: Modbus write: 02.04.04.23.00.04.01.00 (8)
[18:08:27][V][modbus_controller:565]: Command sent 4 0x423 4 send_count: 2
[18:08:28][V][modbus_controller:042]: Sending next modbus command to device 2 register 0x423 count 4

Zu meinem Setup.
Das ist ein Solax X3 MIC GENERATION 1 (scheint die Dinger machen wohl doch Probleme)
Es hängt ein CHINT SmartMeter  am Modbus auf Pin 7&8 mit modbus-addresse 1
der ESP sollte eigentlich dann auch pin 4&5 hängen (laut Doku), ich hab aber auch schon mal Pin 7&8 sowie natürlich einige modbud-addresse  probiert

Irgendeine Idee was ich da falsch mache?

Danke und Gruß
Patrick

PatrickEscher, 2024-09-29:
ok, magically it worked after flashing and restarting a few times :octocat: 

But has anyone any idea about the values and what they really mean. I am comparing the values retrieved through modbus witht he values I retrieve through the REST API and thex show quite some different.

I am also missing some additional values like Total Yield, feedIn Energy, current comsumption etc. Tried to calculate them based on the various inputs, but as said, I am getting different results :(

tobiasfaust, 2024-09-29:
Wenn du „Solax-Mic“ in den Modbus settings ausgewählt und alles korrekt verkabelt hat, sollte es auf Anhieb funktionieren. Wenn man „undefined“ erhält  muss man entweder ein Firmware Update machen und/oder die Anzahl der auszulesenden Register in der  json herunterdrehen.
Auch die bezeichner sind alle selbsterklärend. Verstehe das Problem nicht.

PatrickEscher, 2024-09-29:
Hallo Tobias
verkabelt ist er richtig, Firmware ist die letzte Version, es gibt keine
Neuere. Trotz allem bekomm ich mit deiner Software nur undefined zurück
(btw, meine Kommentare bezogen sich auf den Post von juwild83 mit esphome.
Dort bekomme ich problemlos eine Verbindung und auch die Werte). Nochmals
der Hinweis, das es sich um einen X3 MIC Gen1 handelt, und die Dinger sind
wohl etwas zicking.
Wenn ich mir die json von MagicSven anschaue, dann werden dort Register
abgerufen die definitiv nicht in der Doku drinstehen. z.b. gibt es in der
offiziellen Doku kein Register für FeedIn Energy, in der JSON aber schon.
Stellt sich für mich daher die Frage, ob die Config für den Gen1 wirklich
funktioniert (bin ja nicht der einzige mit den Problemen ;)).

Nicht falsch verstehen, finde dein Projekt klasse und universell einsetzbar
dank MQTT, aber mir frisst das momentan viel zuviel Zeit das auch nur
Ansatzweise zum laufen zu bekommen, deswegen der Switch auf ESPHome, wobei
es hier dann auch an der sehr bescheidenen Doku von Solax hapert.

Wie auch immer, werde bei Gelegenheit das ganze nochmals testen, vielleicht
passiert ja auch wieder was magisches :D.

ich werde meine Posts bzgl. esphome denke ich auch löschen. Sie haben ja
mit deinem Projekt nichts zu tun und die Kommentare/Fragen dort sind denke
ich eher nur verwirrend

Gruß
Patrick

tobiasfaust, 2024-09-29:
Alles klar,
Bei mir gibt’s die jsons für einen mic und einen mic pro.
Das undefined gibt es in der Regel immer, wenn man Zuviel Register abfragt die der Solax an einem Stück zurückgeben kann.

`RequestLiveData": [
        ["#ClientID", "0x04", "0x04", "0x00", "0x00", "0x40"]
      ],`

die letzte 0x40 würde ich mal ändern auf 0x16. Dann sollte etwas kommen. Und dann kann man sukzessive erhöhen bis wieder ein undefined kommt ;)

wäre ein Test wert….

tobiasfaust, 2024-10-09:
Hat es etwas gebracht?

PatrickEscher, 2024-10-27:
Sorry für die späte Antwort, irgendwie bekomme ich keine updates per mail, ganz komisch.
Jein... also ich bekam jetzt teilweise Antworten zurück, allerdings musste ich die Register richtig runterdrehen.
Scheint bei >0x16 is schluss :( (Weiss jetzt auch nicht mehr genau ausm Kopf ob ich einen zweiten Abruf für ab 0x17 konfiguriert hatte)

Die Werte aus dem ModBus hab ich dann mit den Werten aus dem RestAPI verglichen, bin da aber noch auf keinen grünen Zweig gekommen. Es scheint die übertragen noch zusätzliche Daten um alles sauber berechnen zu können und um alles in HASS abbilden zu können fehlen hier doch einige Daten im ModBus.
Da ich aber nicht die kumulativen Daten bekomme, führt für mich kein Weg am RestAPI vorbei (5Min interval).
Bin parallel auch noch mit meinem Heizungsbauer zu Gange um zu sehen wie ich meine Solvis WP richtig damit steuern kann.
Bin daher noch am überlegen ob sich das Realtime überhaupt lohnt, weil ich muss sowieso irgendwie mitteln damit die WP nicht permanent an-aus geht. 
Die andere Idee wäre damit den E-Heizstab zu steuern, da juckt das nicht.
Sobald ich eine gescheite Config habe (Werte und Namen) (muss mir mal ein paar Stunden in Ruhe nehmen), lasse ich sie die zukommen

tobiasfaust, 2024-11-02:
Hi
Per Modbus werden sowohl die live Daten als auch kumulative Daten übermittelt. Man muss nur die richtigen Register abfragen ;)
Das steuern der WP incl Hysterese (damit die WP nicht ständig ein- und ausschaltet )muss man dann in seiner Hausautomatisierung umsetzen.
Spätestens bei Nutzung einer Wallbox braucht man livedaten im Sekundentakt