tobiasfaust/SolaxModbusGateway #30: Solax X3-PRO-G2

wm1962, 2023-09-04:
Hallo Tobias,
zunächst einmal mein Lob an deine tolle Software.
Ich habe einen Solax X3-PRO-G2 und in der  Modbus Config "Solax-MIC" eingestellt.
Über die Com-Schnittstelle des ESP32 kommen folgende Meldungen

> Request queue data to inverter 0x01 0x04 0x00 0x78 0x00 0x77 0x30 0x35 
Read Data from Queue 
0x01 0x84 0x02 0xC2 0xC1
Dataframe invalid

Scheinbar versteht der X3-PRO-G2 die Kommandos nicht.
Kannst Du mir weiterhelfen?

Viele Grüße Wilfried

wm1962, 2023-09-05:
Hi Tobias,
sorry, ich habe mich da verschrieben, ich hatte bereits "Solax X3" eingestellt.

tobiasfaust, 2023-09-05:
Ok, die Antwort bleibt leider dieselbe. Dein Request aus dem ersten Post ist aber von MIC.
Poste mal einen aktuellen Request.
Du kannst versuchen die Invertertypen nacheinander durchzuprobieren ob irgendeiner annähernd passt. Besser wäre es die ModbusProtokoll Doku vom SolaxSupport sich schicken zu lassen. Die antworten auch relativ schnell. Dann kann man eine eigene Sektion für den PRO anlegen wenn diese total abweichend von einer der anderen ist.

wm1962, 2023-09-05:
ok, den Support habe ich soeben angeschrieben.
Den aktuellem Request poste ich heute abend, da die Anlage nicht vor Ort ist.
Kann ich eigentlich im Reiter "Raw Data" vorab etwas testen?
Kannst du die Möglichkeiten dort noch etwas erklären?
[image]

tobiasfaust, 2023-09-05:
erst wenn korrekte Antworten kommen kann man auf der RAW Seite weiter arbeiten. 
Wie? Ist im Wiki unter Punkt "6. Advanced stuff" erklärt ;) Hoffentlich verständlich

wm1962, 2023-09-08:
Hi Tobias,
nun habe ich über meinen Händler die Doku von Solax bekommen.
Kannst du mir mir eine pn schreiben, dann kann ich die dir zukommen lassen.

kommando828, 2023-09-08:
You can add the PDF to a post by dragging it until its under the dotted line. Like so.

[Solax Power X1-BOOST Single phase inverter Modbus-RTU comms protocolV1.0 - Public version.pdf](https://github.com/tobiasfaust/SolaxModbusGateway/files/12562655/Solax.Power.X1-BOOST.Single.phase.inverter.Modbus-RTU.comms.protocolV1.0.-.Public.version.pdf)

wm1962, 2023-09-08:
sorry, at the moment I'm not allowed to publish the document

kommando828, 2023-09-09:
So instead compare the file you have with the Tobias posted pdf's and see which has the closest match.

https://github.com/tobiasfaust/SolaxModbusGateway/tree/development/docs

Then select that inverter and try a test.

wm1962, 2023-09-09:
Hi Tobias,
ich habe bei dem Dokument nun den Schutz entfernen können.
Es sieht analog zur V1.9 aus, ich komme damit leider nicht klar.
Kannst Du helfen?

[Solax X3-PRO Modbus-Protokoll V2.7.pdf](https://github.com/tobiasfaust/SolaxModbusGateway/files/12566239/Solax.X3-PRO.Modbus-Protokoll.V2.7.pdf)

tobiasfaust, 2023-09-10:
Hi,
wenn es wirklich dieses Dokument ist, stell mal den SOLAX MIC ein. Die Live-Register starten ab 0x400

wm1962, 2023-09-10:
Hi Tobias,
ich habs heute nachmittag hinbekommen mit den Live-Registern ab 0x400 und ID-Registern ab 0x300 und es kommen tatsächlich Daten. Zumindest mal die Seriennummer, die anderen Werte kommen auch, passen mit der  Doku von Solax jedoch nicht überein. Da muß ich morgen nochmals testen. Aktuell hat sich der WR abgeschaltet.
Ich melde mich wieder.
Vielen Dank soweit 👍

tobiasfaust, 2023-09-10:
Super, mach mal ein Firmwareupdate. Die Doku muss ja mit dem Stand der Firmware des WR zusammenpassen ;)

tobiasfaust, 2023-09-18:
gibt es etwas Neues?

wm1962, 2023-09-19:
Ich habe die Daten soweit eingepflegt. Beim Tages- und Gesamtertrag habe ich jedoch noch Probleme die Werte richtig zu interpretieren/darzustellen. Da muß ich noch etwas testen.

Noch eine Frage: wie funktioniert das mit dem "set-Befehl" und kann man neben der "InverterSN" weitere ID-Datas auf der webgui page darstellen?

tobiasfaust, 2023-09-21:
Welche WR-Settings hatten denn am besten gepasst? Oder musstest du eine neue Sektion bauen? Wenn du fertig bist nehme ich die JSON Änderung gerne an um sie ins Repo einchecken zu können

Zur Frage mit dem set Befehl:
Ich habe hier die Vorbereitungen im Code eingebaut, aber ohne Tests da ich selbst keine set´s absetze. 
Zu den ID-Data: Aktuell können ohne Codeänderung keine weiteren ID-Daten angezeigt werden. Da müsste ich erst noch ran.

wm1962, 2023-09-21:
Hallo Tobias,
ich habe nun die richtige Darstellung für die noch offenen Werte gefunden. Anbei die Datei mit dem Teil der Register-Daten für den Solax-X3-PRO.
[Solax-X3-PRO.txt](https://github.com/tobiasfaust/SolaxModbusGateway/files/12684523/Solax-X3-PRO.txt)

Grüße Wilfried

wm1962, 2023-09-21:
> Zur Frage mit dem set Befehl:
Ich habe hier die Vorbereitungen im Code eingebaut, aber ohne Tests da ich selbst keine set´s absetze.

Ich sollte beim meinem Solax die Ausgangsleistung setzen können, kannst Du mir das mit dem set Befehl etwas näher erläutern?

tobiasfaust, 2023-09-23:
Hi,
Als erstes musst du das korrekte Register kennen was du setzen willst.
Als absoluter erster Versuch ist auch hier das Example ein guter Start. Einfach die zu setzende registergruppe an einen schreibbefehl anpassen, genaueres findest du nur in deiner Modbus Doku.

Problem ist, das Solax nur ungern die Dokumente mit der Auflistung der schreibregister rausrückt.

MagicSven81, 2023-10-23:
Hier der Code für den X3 Pro G2:

`	"Solax-X3-PRO": {
		"config": {
			"RequestLiveData": [
				["#ClientID", "0x04", "0x04", "0x00", "0x00", "0x40"],
				["#ClientID", "0x04", "0x07", "0x00", "0x00", "0x0A"]
			],
			"RequestIdData": ["#ClientID", "0x03", "0x03", "0x00", "0x00", "0x54"],
			"ClientIdPos": 0,
			"LiveDataFunctionCodePos": 1,
			"LiveDataFunctionCode": "0x04",
			"IdDataFunctionCodePos": 1,
			"IdDataFunctionCode": "0x03",
			"LiveDataStartsAtPos": 3,
			"IdDataStartsAtPos": 3,
			"LiveDataErrorPos": 1,
			"LiveDataErrorCode": "0x84",
			"IdDataErrorPos": 1,
			"IdDataErrorCode": "0x83",
			"LiveDataSuccessPos": 1,
			"LiveDataSuccessCode": "0x04",
			"IdDataSuccessPos": 1,
			"IdDataSuccessCode": "0x03"
		},
		"data": {
			"livedata": [
				{
					"position": [3, 4],
					"name": "PvVoltage1",
					"realname": "Pv Voltage 1",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [5, 6],
					"name": "PvVoltage2",
					"realname": "PV Voltage 2",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [85, 86],
					"name": "PvVoltage3",
					"realname": "PV Voltage 3",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [7, 8],
					"name": "PvCurrent1",
					"realname": "PV Current 1",
					"datatype": "float",
					"factor": 0.1,
					"unit": "A"
				},
				{
					"position": [9, 10],
					"name": "PvCurrent2",
					"realname": "PV Current 2",
					"datatype": "float",
					"factor": 0.1,
					"unit": "A"
				},
				{
					"position": [87, 88],
					"name": "PvCurrent3",
[... 253 lines cut ...]
				},
				{
					"position": [160],
					"name": "MachineType",
					"realname": "Machine Type",
					"datatype": "integer"
				},
				{
					"position": [161],
					"name": "Power Ratio",
					"realname": "Percent of Powerlimits",
					"datatype": "integer",
					"factor": 0.01,
					"unit": "%"
				},
				{
					"position": [162],
					"name": "MpptScanMode",
					"realname": "MpptScanMode",
					"datatype": "integer"
				}
			]
		}
	},
`

closed 2024-02-07 by tobiasfaust