tobiasfaust/SolaxModbusGateway #31: Keine Verbindung zum X3 - Hybrid

ronk74, 2023-09-23:
Hallo, ich habe den Adapter am Com-Anschluss eines QCells- WR (baugleich X3 - Hybrid). Bis zu einigen Updates im ioBroker lief die Modbus-Kommunikation sehr gut. Der Adapter ist über sein Webinterface zu erreichen, alle Werte werden angezeigt, ich kann keinen Fehler in den Einstellungen finden. Trotzdem meldet ioBroker Verbindungsfehler. Auch der Radzio! Modbus Master Simulator kann keine Verbindung herstellen. Da im Interface alles richtig angezeigt wird, gehe ich von einer fehlerfreien Verbindung vom WR zum Adapter aus. Wie bekomme ich eine Verbindung?

ronk74, 2023-09-23:
Hat sich erledigt. Ein simpler Routerneustart hats gebracht.

ronk74, 2023-09-28:
Muss nochmal nachfragen. Die Verbindung ist weiterhin unzuverlässig. Was kann ich tun?

tobiasfaust, 2023-09-28:
Welches Release hast du? Aktuell ist 3.1.1. 
Hier läuft auch der automatische WLAN Reconnect zuverlässiger. Wenn du NUR den Router neustarten muss ohne am ESP32 etwas zu machen würde ich das Problem ev. woanders suchen?

ronk74, 2023-09-28:
Release ist 3.1.1 Die WLan Verbindung zum Adapter ist stabil. Wo kann ich noch suchen?

tobiasfaust, 2023-09-28:
Im Fehlerfall:
- die Weboberfläche ist erreichbar?
- die aktuellen Werte werden auf der Weboberfläche angezeigt und auch kontinuierlich aktualisiert?
- werden die Werte per MQTT versendet? Bitte einmal über zb.. http://mqtt-explorer.com/ testen

ronk74, 2023-09-28:
Danke für die Antwort.

die Weboberfläche ist erreichbar
die aktuellen Werte werden angezeigt und laufend aktualisiert

über den MQTT-Explorer bekomme ich ein Disconnected from Server

tobiasfaust, 2023-09-28:
Das bedeutet, auf der Statusseite der Weboberfläche steht bei "MQTT-Status" auch disconnected?
Mit dem MQTT-Explorer kannst dich nicht auf deinen MQTT-Server verbinden? Dann suche mal dort

ronk74, 2023-09-28:
Nein, auf der Statusseite steht bei "MQTT-Status" Connected. Trotzdem schreibt der MQTT-Explorer disconnected. ???

tobiasfaust, 2023-09-28:
ich würde als erstes beim MQTT Server suchen, ggf auf einen öffentlichen MQTT Server zum testen umstellen

ronk74, 2023-09-28:
mit test.mosquitto.org beim Broker Mosquitto bekomme ich sofort eine Verbindung und auch Daten.

ronk74, 2023-09-29:
Also MQTT funktioniert. Der Modbus bleibt immer noch ohne Verbindung?

tobiasfaust, 2023-09-30:
Wenn die Werte vom WR auf der weboberfläche des ESP passen und auch aktualisiert werden, ist die ModbusRTU Verbindung in Ordnung und arbeitet

ronk74, 2023-09-30:
Also muss es am Wechselrichter liegen?

tobiasfaust, 2023-09-30:
Wiso Wechselrichter? Wenn die Werte des WR am esp sauber ankommen macht der WR seinen Job! 
mehr als Hinweise geben kann ich nicht, testen und ausprobieren musst du schon selbst.

ronk74, 2023-09-30:
Naja, getestet und probiert habe ich sehr viel. Der WR macht seinen Job, der Meinung bin ich auch. Wenn ich nicht weiterkomme, frage ich gerne bei Leuten mit Erfahrung.
Also, der Adapter ist stabil im WLan. Er liefert auch zuverlässig Werte. Über die MQTT-Schnittstelle kann ich die Werte auslesen.
Weder im RMMS-Programm noch in ioBroker bekomme ich eine Verbindung.
Wo kann ich noch suchen, bzw. was kann ich testen?

tobiasfaust, 2023-09-30:
Beherzige dich die Ratschläge weiter oben und gehe systematisch vor. 
nach deinen Infos funktionieren WR und ESP32 sauber und sind bei der Fehlersuche raus.
Du musst beim MQTT Server, aber wohl eher beim iobroker ansetzen wenn der mqttserver die Daten ebenfalls sauber ausgibt

ronk74, 2023-09-30:
Den ioBroker würde ich erstmal rauslassen. Ich habe mir den Radzio!Modbus Master Simulator runtergeladen. Der ist zum Testen von Modbus-Verbindungen. Ich gebe dort die IP-Adresse des WR ein, dazu den TCP-Port 502 und hake Modbus TCP an.
Im Testfenster kommt Modbus Message TimeOut. Keine Verbindung.

kommando828, 2023-09-30:
The Modbus standard is very open, not all Modbus software is configured the same and with such an open standard none can cover all connections. I use several Modbus programs for that reason, sometimes one will fail where another would work. Try CAS Scanner

https://store.chipkin.com/products/tools/cas-modbus-scanner

and ModbusMechanic

https://github.com/SciFiDryer/ModbusMechanic

As long as one connects you do not have a Modbus issue except possibly a marginal signal ie a wiring issue on the edge of working. But as you already have achieved a Modbus connection and the failure is the MQTT server that would be point to investigate.

tobiasfaust, 2023-09-30:
Nochmal, beherzige die Ratschläge oben. 
Modbus via tcp ist hier völlig irrelevant, ggf wird es vieleicht sogar auch nicht unterstützt von deinem WR. Der ESP arbeitet mit deinem WR korrekt zusammen, daher kann dir hier nicht weiter geholfen werden.

ronk74, 2023-09-30:
Ich kann keine Ratschläge beherzigen, ich erkenne keine. Mein WR ist ein QCells Q.VOLT_HYB-G3-3P, baugleich Solax X-3. Die Modbusverbindung hat bereits funktioniert, aus mir nicht bekannten Gründen gehts jetzt nicht mehr. Ich bin kein Fachmann, weder bin ich IT-Spezialist, noch Elektriker. 
Trotzdem vielen Dank. Ich werde niemanden mehr um Hilfe bitten.

tobiasfaust, 2023-09-30:
Die Ratschläge waren folgende:
* bitte systematisch vorgehen, entlang der Datenfluss -Strecke
* Die Werte im ESP32 kommen vom WR an -> Verbindung WR zu ESP32 also ok
* Die Werte kommen via MQTT am MQTT-Server an, hast du ja kontrolliert -> MQTT Verbindung ESP32 zum MQTT-Server also ok
* Fehler bitte bei der Verbindung MQTT-Server -> IOBroker suchen, Daher ist das kein Thema für hier sondern eher für das IOBroker Forum

und trotz dieser ratschläge dokterst du mit ModbusTCP am WR rum? 
Der QCells Q.Volt_Hybrid ist hier schon mehrfach erfolgreich mit dem ESP32 im Betrieb genommen worden wurde berichtet.

ronk74, 2023-09-30:
Danke.
im ioBroker gibt es einen Modbus-Adapter. Der hat auch in meiner Konfiguration funktioniert. Nun eben (warum, weiß ich nicht) nicht mehr. In einem ähnlich gelagerten Fall wurde von den ioBroker-Experten geraten, die Kommunikation erstmal mit dem Radzio! Modbus Master Simulator (ein Tool zum Testen von Modbus- Geräten) sicherzustellen. Solange dieser Simulator keine Antwort gibt, braucht man auch in ioBroker nicht weiter zu probieren. So die Aussage. Dort war es aber wohl nur ein simpler IP-Adressenfehler.
Im ersten Versuch hat das bei mir auch funktioniert, danach konnte ich im modbus-Adapter in ioBroker meine Werte abgreifen.
Nun versuche ich zu ergründen, was die Ursache sein könnte. Wie ganz am Anfang geschrieben, half kurzzeitig ein Router-Neustart.
Ich habe hier eine Fritz!Box. Der ESP32 verbindet sich per WLan direkt mit der Box. Der Raspberry mit ioBroker verbindet sich auch mit der Fritz!Box per WLan. Der Windows-PC mit dem Radzio Simulator hängt per Lan an der Fritz!Box. Der WR hängt auch per WLan an der Fritz!Box.
Nochmal: ich brauche keinen MQTT-Server und ich will auch keinen. Die Modbus-Konfiguration im ESP32 ist auf Solax X-3 eingestellt. Sollte doch richtig sein?

tobiasfaust, 2023-09-30:
Hi, 
ich versuche es mal von ganz vorn zu erklären. Der ESP32 mit dem ModbusGateway benötigt IMMER einen MQTT-Server als Backend. Den kann man selbst prival lokal betreiben oder man nutzt einen öffentlichen. Diesen MQTT Server musst du auch in der Konfiguration des Modbusgateways einstellen. Denn genau dahin werden die empfangenen WR-Werte auch gesendet. Und von dort (dem MQTT-Server) holt sich jeder diese Werte wieder ab, auch dein ioBroker oder sonst wer.

Und wie kommen nun die Werte vom WR in den ESP32? Der ESP32 nutzt ModbusRTU (das ist etwas anderes als ModbusTCP).

Wenn es Probleme gibt muss man entlang des Datenflusses suchen:
1. Kommen die Werte im ESP32 an? Werden die Werte auf der Weboberfläche des ESP32 korrekt dargestellt und aktualisieren sie sich im eingestellten sekundentakt? Du sagst: ja. Also OK
2. zeigt der ESP auf der Statussseite eine VErbindung zum MQTT-Server an? Du sagst ja. Also ok
3. Kommen beim MQTT-Server die Werte vom ESP32 an? Kann man mit Tools selbst überprüfen, zb.  http://mqtt-explorer.com/
4. Wenn die Werte des WR im MQTT-Server ankommen, ist beim ESP32 alles in Ordnung, dann haben wir hier kein Problem.

Wenn bis hierhin alles in ordnung bringt der Radzio Simulator garnix denn die Datenverbindung funktioniert ja. Ansonsten würden im ESP auch keine Werte erscheinen.

Für den QCells WR ist Solax-X3 die korrekte einstellung, muss es ja auch sein da du schon berichtest hattest, das die Werte in der Weboberfläche des ESP ja korrekt sind.

Übrigens, der WR muss NICHT im WLAN hängen, diese telefonieren häufig nach Hause (China) deshalb trennt man diese normalerweise vom WLAN wenn man die Werte über ModbusRTU auslesen kann.

ronk74, 2023-10-11:
-