tobiasfaust/SolaxModbusGateway #32: Deye Temperaturwerte

gogo2001-2002, 2023-09-30:
Hallo Tobias,
der Deye WR liefert ja Temperaturwerte zurück, bei denen man den Wert 1000 abziehen muss. Habe ich hier irgendwie die Möglichkeit das umzusetzen innerhalb deiner Software?
Gruß
Georg

tobiasfaust, 2023-09-30:
Ja, das wird über die Register.h eingestellt, unter factor

gogo2001-2002, 2023-09-30:
Ja, da habe ich teilweise ja schon die Werte für den Deye hinzugefügt.
Aber das mit den Temperaturen ist leider nicht nur die Subtraktion sondern:
-> 1250 mean 25.0℃ 
Also, Wert -1000 * 0.1
Kann ich das auch unter factor so reinschreiben?

tobiasfaust, 2023-09-30:
Sorry, eine Subtraktion klappt nicht, es wird immer mit dem Faktor (als float wert) multipliziert.  Bist du sicher das du hier keine VErschiebung in den Registern hast? Solch eine Subtraktion ist a-typisch.... was steht denn in der Modbusprotokoll-Doku dazu?

gogo2001-2002, 2023-09-30:
Auszug aus der Spezi
z.B, Register 602
Beschreibung -> 1250 mean 25.0℃ 
Da ich das aktuell per Modbus TCP auslese, weiß ich auch, das der Wert wirklich so kommt.

gogo2001-2002, 2023-09-30:
ich habe mal 2 optionale Werte mit reingebracht (addbefore und addafter). Im Source sind das ja nur wenige Stellen. Ich teste das nachher mal, wenn das alles klappt, schicke ich dir das. Wenn du möchtest, kannst du das ja einbauen.

tobiasfaust, 2023-09-30:
ich hatte gerade an einen(!) weiteren Parameter gedacht, zb. valueadd. der kann ja auch negative Werte haben

tobiasfaust, 2023-09-30:
Anpassung gemacht, test mal bitte: 7d5fd46

Das optionale Feld heisst "valueAdd"

valueAdd: -1000

gogo2001-2002, 2023-09-30:
Hi Tobias,
ja ich hatte nur daran gedacht, die Addition bzw. Subtraktion vor bzw. nach der Multiplikation mit dem "factor" zu machen. 
(value * factor) + valueaddafter   -> Beispiel (1250*0,1)+(-1000) = -875
oder
(value + valueaddbefore) * factor -> Beispiel (1250+(-1000)) *0,1 = 25

Hast du ihn nach oder vor der Multiplikation eingebaut?

tobiasfaust, 2023-09-30:
Danach

also (wert * factor) + valueAdd

macht 
(1250 * 0.1)-100

ValueAdd wäre also -100

gogo2001-2002, 2023-09-30:
Ja, kann ich auch mit leben :-) 
Zum Testen komme ich erst heute abend. 
Info folgt dann
Vielen Dank schon mal für dein schnelle Hilfe.

gogo2001-2002, 2023-10-01:
Hallo Tobias,
funktioniert hervorragend. Temperatur wird nun korrekt angezeigt/übertragen. Vielen Dank für deine Hilfe. 
Wenn ich alle Werte für den Deye eingetragen habe schicke ich dir die zu. Dann haben ggfs. andere auch was davon.
Gruß
Georg

gogo2001-2002, 2023-10-03:
Hallo Tobias, wolltest du die angepasste register.h haben? Wenn ja, wie kann ich dir die zukommen lassen.
Gruß
Georg

tobiasfaust, 2023-10-04:
Ja, auf jeden Fall 👍:)

gogo2001-2002, 2023-10-04:
Hallo Tobias,

anbei die angepasste Datei.

Gruß

Georg

#ifndef SOLAXREGISTER_H
#define SOLAXREGISTER_H

/* https://jsonlint.com/ */
const char JSON[] PROGMEM = R"=====(
{
	"Solax-X1": {
		"config": {
			"RequestLiveData": [
				["#ClientID", "0x04", "0x00", "0x00", "0x00", "0x77"],
				["#ClientID", "0x04", "0x00", "0x78", "0x00", "0x77"]
			],
			"RequestIdData": ["#ClientID", "0x03", "0x00", "0x00", "0x00", "0x14"],
			"ClientIdPos": 0,
			"LiveDataFunctionCodePos": 1,
			"LiveDataFunctionCode": "0x04",
			"IdDataFunctionCodePos": 1,
			"IdDataFunctionCode": "0x03",
			"LiveDataStartsAtPos": 3,
			"IdDataStartsAtPos": 3,
			"LiveDataErrorPos": 1,
			"LiveDataErrorCode": "0x84",
			"IdDataErrorPos": 1,
			"IdDataErrorCode": "0x83",
			"LiveDataSuccessPos": 1,
			"LiveDataSuccessCode": "0x04",
			"IdDataSuccessPos": 1,
			"IdDataSuccessCode": "0x03"
		},
		"data": {
			"livedata": [{
					"position": [3, 4],
					"name": "GridVoltage",
					"realname": "Grid Voltage",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [5, 6],
					"name": "GridCurrent",
					"realname": "Grid Current",
					"datatype": "float",
					"factor": 0.1,
					"unit": "A"
				},
				{
					"position": [7, 8],
					"name": "GridPower",
					"realname": "Grid Power",
					"openwbtopic": "openWB/set/pv/W",
					"datatype": "integer",
					"unit": "W"
				},
				{
					"position": [9, 10],
					"name": "PvVoltage1",
					"realname": "Pv Voltage 1",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
[... 2174 lines cut ...]
					"unit": "V"
				},
				{
					"position": [374, 375],
					"name": "PV4DCCurrent",
					"realname": "PV4 DC Current",
					"datatype": "float",
					"factor": 0.01,			
					"unit": "A"
				}
			],
			"id": [
				{
					"position": [9,10,11,12,13,14,15,16],
					"name": "SN",
					"realname": "SerialNumber",
					"datatype": "string"
				}
			]
		}	
	}
}
)=====";

#endif

MagicSven81, 2023-10-22:
@gogo2001-2002 
für den Deye 5k hast du nicht zufällig die Zuordnung der Modbus Register?
Der scheint tatsächlich anderst zu sein als beim 12k

gogo2001-2002, 2023-10-22:
Schau mal hier, da wurde der 5er besprochen. In einem der Threads wurde ein Link zu einem Dokument gepostet. Vielleicht hilft dir das ja.

https://dy-support.org/community/technische-fragen/modbus-protokoll-fuer-sun-5k-sg03lp1-eu/

MagicSven81, 2023-10-22:
besten Dank, die Seite habe ich gerade auch gefunden.
Mal schauen ob ich es hinbekomme.

tobiasfaust, 2023-10-23:
Hi,
ICh habe es leider noch nicht geschafft den deye12k zu integrieren, steht aber noch auf meiner todo Liste.
Wenn der 5k Deye abweichend ist freue ich mich wieder über eine Register.h Sektion :) 👍

closed 2024-01-17 by tobiasfaust

tobiasfaust, 2024-04-20:
@gogo2001-2002 
Funktioniert es bei dir mit dem Deye? Wenn ja, kannst du mal bitte in #43 schauen und ggf unterstützen?

gogo2001-2002, 2024-04-20:
Hallo Tobias, kann da leider nicht mehr unterstützen, da ich meinen Deye verkauft habe. Sorry

MagicSven81, 2024-04-20:
Ich kann es mir mal anschauen. 
Läuft an meinem deye seit mehreren Wochen stabil.

tobiasfaust, 2024-04-20:
@gogo2001-2002 
Kein Problem, danke für die Rückmeldung 
@MagicSven81 
Danke, scheint also wohl nicht an meinem Programm zu liegen wenn es bei dir stabil läuft