tobiasfaust/SolaxModbusGateway #34: WiFi, nur mit USB-Kabel

truderinger, 2023-10-30:
Hallo zusammen,

danke für die Mühen der Software. Ich habe das Problem, das mit dem USB-Kabel alles über WiFi funktioniert. Die rote LED am ESP32 blinkt kurz und findet das Netz. Jedoch abgekoppelt vom PC  mit Platform.io ist keine WiFi-Verbindung möglich. Die rote LED blinkt immer im gleichen Takt vor sich hin. Ich nutze 3.1.1. Wo könnte der Fehler liegen?

Danke.

Herbert

tobiasfaust, 2023-10-30:
Ich würde mal ganz stark auf die Stromversorgung tippen, also stärkeres Netzteil und vor allem ein gutes Micro usb Kabel. 
mit einem billigen usb Kabel hatte ich auch schon dasselbe Problem.

truderinger, 2023-10-30:
Hallo, 

es funzt jetzt. Es war tatsächlich das Netzteil. Unglaublich hätte ich nie als Fehlerursache gesehen. Aber da habe ich mal wieder was gelernt.  

Herzlichen Dank für die Mühen und die schnelle Info.

truderinger, 2023-10-30:
Hallo,

 

danke. Ja das war tatsächlich der Fehler. Unglaublich. Dankeeeeeeee.

 

Ich habe einen Deye Sun 12kWp gem. SoalarAssistant kann der BMS-Port gesplittet werden, weil RS485 auf Pin 1-3 und CanBus auf Pin 4+5, liegt. Das Netzkabel Pin 1+2 beim HW-97 auf RO+DI (auch gedreht, und auch TR+RX = ESP32 Pin 17+16) gelegt. "register.h" für den Deye mit in das Image eingebaut (war ja so im ioBroker-Forum kommuniziert). MQTT-Server ist verbunden. So ca. 15 Punkte in der Item-Config aktiviert. Leider nichts.

 

RS485 B-     Netzkabel Pin 1

RS485 A-     Netzkabel Pin 2

kein GND

 

Also RX1/TX1 auf GPiO 16 + 17 oder RX2/TX2 auf 09+10 des ESP32 Wroom 32 in Modbus-Config gesetzt. Kein RTS-Pin, Client-ID = weil Deye ID 1, 9.600 baud, Intevall ID auf 3600 sec.

 

Eigentlich müsste er Daten empfangen und somit die rote LED am HW-97  flackern. Bleibt aber gleichmaässig, beständig rot. Jetzt bin ich da überfordert, wo es liegen könnte. Wäre nochmals eine kleine Hilfestellung möglich?

 

Danke.

 

mfg

Herbert

 

Gesendet: Montag, 30. Oktober 2023 um 09:22 Uhr

kommando828, 2023-10-30:
Not all network cables are the same, I had the same issue on my Solax and it was fixed using a network cable with stranded 24awg  full copper wires. There are copper plated wires around which will not conduct RS485. If you have the option then use the GND connection, RS485 is a differential standard between the 2 wires, GND provides a reference point for the signal.

truderinger, 2023-10-30:
Hallo,

 

nochmals danke  für Ihre Info. Dann verwende ich mal den GND als erste Rep.-Möglichkeit und dann ein besseres Netzkabel. Beim HW-97 sind es ja nur Schraubanschlüsse für A+ und B-. Dan ziehe ich den 3,3 Vcc vom ESP32 zum Netzkabel. 

 

Danke für Ihre Lebenszeit und Ihre Mühen. Ich teste jetz mal. die nächsten Tage.

 

mfg

Herbert Springer

 
 

Gesendet: Montag, 30. Oktober 2023 um 15:54 Uhr

truderinger, 2023-10-30:
Hallo nochmals und sorry. Meinte natürlich den GND und nicht den Vcc

 

best regards Herbert

 
 

Gesendet: Montag, 30. Oktober 2023 um 15:54 Uhr

kommando828, 2023-10-30:
You only use GND if there is one as part of the RJ45 wires at the inverter. The Solax does have a GND at its RJ45 socket.

closed 2024-01-17 by tobiasfaust