tobiasfaust/SolaxModbusGateway #35: Solax X3-10.0P-T-D

michacee, 2023-11-06:
Hat zufällig jemand schon mal die register.h für einen **Solax Mic Pro** angepasst? Mit der normalen Solax-Mic Konfiguration kann ich den WR leider nicht überreden, Daten auszugeben.

tobiasfaust, 2023-11-06:
Schau mal in #30 ob diese passt. Ich habe es nur noch nicht geschafft diese Config einzuarbeiten.

falls diese nicht passt kommst du über das passende Modbus Protokoll PDF nicht herum. Der Solax  Support  ist da aber sehr schnell und hilfsbereit.

michacee, 2023-11-07:
Leider kein Erfolg, ich werd mich wohl an Solax wenden müssen. Oder gibt es noch eine andere Möglichkeit?

kommando828, 2023-11-07:
Solax support used to be very reluctant to release the Register pdf's but at least in the last 18 months they are now very responsive, mine took 48 hrs to arrive. Other than that there is google, hits on home assistant forums can contain the PDF's but Solax's model naming can be deceptive ie your inverter name does not include the Generation ie Gen3 or Gen4 which is important.

tobiasfaust, 2023-11-07:
Leider nicht, du brauchst das Modbus Protokoll damit wir hier weiter kommen

michacee, 2023-11-07:
Ich hab dem Solax Support gerade mal eine Mail geschrieben.

michacee, 2023-11-20:
Hat noch jemand eine Idee, wie man an die Modbus Register kommt, Solax hat bis dato noch nicht reagiert.

kommando828, 2023-11-20:
If you have a laptop and a USB RS485 adaptor then this can be used to scan for register numbers.

https://github.com/SciFiDryer/ModbusMechanic

tobiasfaust, 2024-01-17:
closed due inactivity

michacee, 2024-05-31:
**Update** 

Nach langer Zeit konnte ich mich mal wieder damit beschäftigen. 
Die Inverter Config muss so angepasst werden:

`
"RequestLiveData": [
        [
          "#ClientID",
          "0x04",
          "0x04",
          "0x00",
          "0x00",
          **"0x35"**
        ]
      ],
`

michacee, 2024-05-31:
`
{
  "Solax-Mic-Pro": {
    "config": {
      "RequestLiveData": [
        [
          "#ClientID",
          "0x04",
          "0x04",
          "0x00",
          "0x00",
          "0x35"
        ]
      ],
      "RequestIdData": [
        "#ClientID",
        "0x03",
        "0x03",
        "0x00",
        "0x00",
        "0x07"
      ],
      "ClientIdPos": 0,
      "LiveDataFunctionCodePos": 1,
      "LiveDataFunctionCode": "0x04",
      "IdDataFunctionCodePos": 1,
      "IdDataFunctionCode": "0x03",
      "LiveDataStartsAtPos": 3,
      "IdDataStartsAtPos": 3,
      "LiveDataErrorPos": 1,
      "LiveDataErrorCode": "0x84",
      "IdDataErrorPos": 1,
      "IdDataErrorCode": "0x83",
      "LiveDataSuccessPos": 1,
      "LiveDataSuccessCode": "0x04",
      "IdDataSuccessPos": 1,
      "IdDataSuccessCode": "0x03"
    },
    "data": {
      "livedata": [
        {
          "position": [
            3,
            4
          ],
          "name": "PvVoltage1",
          "realname": "Pv Voltage 1",
          "datatype": "float",
          "factor": 0.1,
          "unit": "V"
        },
        {
          "position": [
            5,
            6
          ],
          "name": "PvVoltage2",
          "realname": "Pv Voltage 2",
          "datatype": "float",
          "factor": 0.1,
          "unit": "V"
        },
        {
          "position": [
            7,
            8
          ],
          "name": "PvCurrent1",
          "realname": "Pv Current 1",
          "datatype": "float",
[... 308 lines cut ...]
          "position": [
            4,
            3,
            6,
            5,
            8,
            7,
            10,
            9,
            12,
            11,
            14,
            13,
            16,
            15
          ],
          "name": "InverterSN",
          "realname": "Inverter SerialNumber",
          "datatype": "string"
        }
      ]
    }
  }
}
`

MagicSven81, 2024-07-07:
@michacee  - ich habe deinen Code genommen und entsprechend formatiert, aber dennoch erhalte ich keine Daten.
Hast du es bei dir zum laufen bekommen?
Falls ja, könntest du den Quelltext mal in deiner Formatierung hier posten?

michacee, 2024-07-08:
Passen denn auch die anderen Parameter?

[image]

MagicSven81, 2024-07-13:
@michacee  - ja die Einstellungen passen.
Wäre es dir möglich, deinen SourceCode oder noch besser die Datei "Solax-Mic-Pro" hier im RAW Format zu posten?
Ich denke dass ich hier einen Fehler drin habe.

Leider ist dein Code der oben gepostet ist nicht formatiert.

Das wäre super - danke

michacee, 2024-07-15:
`{
  "Solax-Mic-Pro": {
    "config": {
      "RequestLiveData": [
        [
          "#ClientID",
          "0x04",
          "0x04",
          "0x00",
          "0x00",
          "0x35"
        ]
      ],
      "RequestIdData": [
        "#ClientID",
        "0x03",
        "0x03",
        "0x00",
        "0x00",
        "0x07"
      ],
      "ClientIdPos": 0,
      "LiveDataFunctionCodePos": 1,
      "LiveDataFunctionCode": "0x04",
      "IdDataFunctionCodePos": 1,
      "IdDataFunctionCode": "0x03",
      "LiveDataStartsAtPos": 3,
      "IdDataStartsAtPos": 3,
      "LiveDataErrorPos": 1,
      "LiveDataErrorCode": "0x84",
      "IdDataErrorPos": 1,
      "IdDataErrorCode": "0x83",
      "LiveDataSuccessPos": 1,
      "LiveDataSuccessCode": "0x04",
      "IdDataSuccessPos": 1,
      "IdDataSuccessCode": "0x03"
    },
    "data": {
      "livedata": [
        {
          "position": [
            3,
            4
          ],
          "name": "PvVoltage1",
          "realname": "Pv Voltage 1",
          "datatype": "float",
          "factor": 0.1,
          "unit": "V"
        },
        {
          "position": [
            5,
            6
          ],
          "name": "PvVoltage2",
          "realname": "Pv Voltage 2",
          "datatype": "float",
          "factor": 0.1,
          "unit": "V"
        },
        {
          "position": [
            7,
            8
          ],
          "name": "PvCurrent1",
          "realname": "Pv Current 1",
          "datatype": "float",
          "factor": 0.1,
[... 306 lines cut ...]
        {
          "position": [
            4,
            3,
            6,
            5,
            8,
            7,
            10,
            9,
            12,
            11,
            14,
            13,
            16,
            15
          ],
          "name": "InverterSN",
          "realname": "Inverter SerialNumber",
          "datatype": "string"
        }
      ]
    }
  }
}`

michacee, 2024-07-15:
[Solax-Mic-Pro.json](https://github.com/user-attachments/files/16231179/Solax-Mic-Pro.json)

MagicSven81, 2024-07-21:
@tobiasfaust  - könntest du das in dein künftiges update mit integrieren?

tobiasfaust, 2024-07-21:
@michacee hast du noch mehr als "RequestLiveData" angepasst? 
Ich habe es erstmal als neues Modul übernommen. Bitte testet es mal

closed 2024-07-21 by tobiasfaust

michacee, 2024-07-22:
Ja hab ich.