tobiasfaust/SolaxModbusGateway #36: Die Daten werden nur im Mqtt Explorere angezeigt

Bimboboy007, 2023-11-18:
Habe das Problem dass die Sensoren beim Deye nur im Mqtt Explorer zu finden sind und nicht in Mqtt als Gerät mit den Sensoren. Muss bei allen Sensoren/Werten ein state_topic erstellen um diese überhaupt zu HA zu bekommen. Wenn dies geschehen ist werden die Werte mal angezeigt mal springen sie auf 0.

tobiasfaust, 2023-11-19:
Leider habe ich das Problem überhaupt nicht verstanden. Bitte auch Screenshots der weboberfläche vom ESP32 hier anzeigen.

Bimboboy007, 2023-11-19:
Habe momentan keinen Screenshot. Problembeschreibung wieso taucht das Gerät nicht im MQQTBroker auf ?
Muss die Sensoren mit dem MqttExplorer auslesen und dann jeweils ein state_topic erstellen um die Sensoren in Home Assistant zu haben.
[image]

tobiasfaust, 2023-11-19:
Da kann ich nicht helfen, da bist du im HA Forum besser aufgehoben. Wenn die Daten vom ESP32 korrekt alle dem MQTT Broker / Server zugestellt werden ist auf der ESP32 Seite alles in Ordnung

Bimboboy007, 2023-11-20:
Irgendwie reden wir aneinander vorbei. Genau das macht das Gerät eben nicht die Daten werden an den Mqtt Broker nicht weitergeleitet !  Die Daten sind nur im Mqtt Explorer verfügbar. Das sind 2 verschiedene Sachen.

tobiasfaust, 2023-11-20:
Was bezeichnet du als MQTT Explorer?
Was hast du als MQTT Server im ESP eingestellt? Screenshot?

tobiasfaust, 2023-11-20:
Da kann ich mich leider nur wiederholen:
Da die Deye Daten beim MQTT Server korrekt zugestellt werden, das zeigt ja dein MQTT Explorer Screenshot, ist beim ESP32 alles in Ordnung und du musst beim Broker suchen

Bimboboy007, 2023-11-20:
Das Problem ist das die Sensoren ausser die Messdaten nichts heraus gibt weder eine uniqe_id usw. und deshalb vom Mqtt Broker nicht als solches Gerät erkannt wird.
[image]

kommando828, 2023-11-20:
No, your broker is not aligned to the data as given out by the ESP, as each Broker needs a different format then you need to break out the message into its parts and then pass only what the broker wants in the format it needs.

I use Node-red to do this on the output from Yasdi2mqtt as this is based on JSON and has all the tools.

tobiasfaust, 2023-11-20:
Klar, weil im MQTT topic, im Pfad, schon die unique-id drin hängt.

Bimboboy007, 2023-11-20:
Eine frage habe ich noch. Verkaufen sie das bei keinanzeigen ?

tobiasfaust, 2023-11-21:
ich habe nochmal nachgesehen, Der MQTT Broker erwartet einen grossen JSON String in dem alle Elemente/MEsswerte gesammelt enthalten sind. Das ist dann wohl ein "Gerät". Immer dasselbe Topic mit einem Json-Datenset.
Das MQTT Gateway arbeitet anders, es sendet alle Messwerte isoliert raus, jeder Messwert für sich. Das hat den vorteil das Apps nur die Messwerte abonieren müssen die sie auch wirklich brauchen.
NAchteil eines JSONS ist auch, das der Json-String wirklich gross werden kann und viel - zu viel- Speicher braucht.

Alternative: die Daten per HTTP-Request holen. Dort wird dann auch ein grosses json array geliefert:

`http://<ip>/getitems`

Um auf die Frage zu antworten: Nein, das verkaufe ich nicht. Das lohnt die Zeit nicht - Die paar Kabel kann jeder selbst zusammen stecken.... ;)

tobiasfaust, 2023-12-01:
Logisch, hast ja auch die ip vergessen

Bimboboy007, 2023-12-01:
Hi, also das habe ich schon eingegeben http://<ip>/getitems .
Hab es wieder Gelöscht ist auch nicht so innovativ wie das ESP Projekt, damit kann man auch Änderungen und Einstellungen vornehmen.

tobiasfaust, 2023-12-01:
Nee, du verstehst es nicht, du musst schon deine richtige IP eintragen, oben im Beispiel ist es nur ein Platzhalter!

Bimboboy007, 2023-12-01:
Ich hab schon richtig verstanden und das hab ich auch so gemacht. Aber egal für mich hat sich das Project erledigt.

closed 2023-12-01 by tobiasfaust