tobiasfaust/SolaxModbusGateway #40: Hat zufällig jemand schon mal die register.h für einen Solax Mic Pro angepasst? Mit der normalen Solax-Mic Konfiguration kann ich den WR leider nicht überreden, Daten auszugeben.

Maver2k, 2023-12-29:


wwwiesel, 2024-01-17:
Hi, ich bin auch auf der Suche nach einer MOD Konfiguration... bzw. zumindest mal eine Basis. Das Beispiel aus dem Wiki funktioniert irgendwie nicht :(

tobiasfaust, 2024-01-17:
Funktioniert denn einer der vordefinierten WR sodass irgendetwas auf der RawData Seite ausgegeben wird?

tobiasfaust, 2024-01-17:
Hier steht’s aber…. Die Register starten ab 3000

https://objects.githubusercontent.com/github-production-repository-file-5c1aeb/226368338/10957121?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240117%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240117T191845Z&X-Amz-Expires=300&X-Amz-Signature=c55ef1a588ed4bc1d808e7c0df4e60a62e673e196b0f5efce21858577809c7a5&X-Amz-SignedHeaders=host&actor_id=9401139&key_id=0&repo_id=226368338&response-content-disposition=attachment%3Bfilename%3DNew-Modbus.RS485.RTU.Protocal.Latest.Ver.pdf&response-content-type=application%2Fpdf

tobiasfaust, 2024-01-17:
Probier mal das….
Das ließt die live Register ab stelle 3000

"RequestLiveData": [
				["#ClientID", "0x04", "0x0B", "0xB8", "0x00", "0x77"]
]

wwwiesel, 2024-01-17:
Der Chef persönlich! Vielen Dank für diese tolle Arbeit! Muss ich dies dann erst in einer neuen Firmware kompilieren? Hatte die bin genommen aus dem Release.

Und unter raw habe ich ein paar Daten, aber nicht alle denke ich.

Ich baue mir morgen mal eine neue Firmware.bin zusammen. 

Muss ich dann 3000x2+3 für das jeweilige Register machen?

Maver2k, 2024-01-18:
> Funktioniert denn einer der vordefinierten WR sodass irgendetwas auf der RawData Seite ausgegeben wird?

Hallo Tobias, 
es wird was auf der RawData Seite angezeigt, aber scheint so auszusehen wie bei wwwiesel. Kann aber im Moment auch nicht drauf schauen, da ich für die nächsten Tage unterwegs bin. 

@wwwiesel: Wenn du mit deiner .bin fertig wärst, würdest du die dann zur Verfügung stellen? Das wäre echt super.

tobiasfaust, 2024-01-18:
Wichtig ist erstmal, das auf der RAWData Seite unter LiveData jede Menge angezeigt wird. In meinem Beispiel oben von Position 3000 genau 119 Register, also 3000 - 3119
Bitte dazu in der register.h eine neue Sektion erstellen "Growatt 3000 TL3 XH"
Die korrekten Registernummern für die Register.h bekommt ihr am besten wie im Wiki beschrieben ebenfalls über die RawData seite heraus, darüber könnt ihr dann auch gleich testen ob der Wert plausibel ist.

Dann die Register.h erweitern, kompilieren, flashen und prüfen ob jetzt alles passt. Ggf das Spiel wiederholen um die register.h definition zu verfeinern.

Eigentlich ist das selbstständige Herausfinden der korrekten Register ein Kinderspiel wenn man die korrekte Modbus-Doku des WR hat ;)

wwwiesel, 2024-01-18:
Natürlich werde ich, wenn ich es hinbekomme, alles zur Verfügung stellen. 

@tobiasfaust Wie bist du auf 

"RequestLiveData": [
["#ClientID", "0x04", "0x0B", "0xB8", "0x00", "0x77"]
]

gekommen? ich will es ja verstehen, stehe aber etwas im Wald.

tobiasfaust, 2024-01-18:
77 hex = 119 dez.
Immer wenn ich mehr als 119 Register abfrage gab es einen Fehler. Kannst du aber ausprobieren ob du mehr abfragen kannst.

wwwiesel, 2024-01-18:
Ich bin nun ein wenig weiter... Baud Rate muss 9600 Sein sonst sieht man da nichts...
	"Growatt-MOD-Type": {
		"config": {
			"RequestLiveData": [
				["#ClientID", "0x03", "0x00", "0x00", "0x00", "0x77"],
				["#ClientID", "0x04", "0x0B", "0xB8", "0x00", "0x77"]
			],
			"RequestIdData": ["#ClientID", "0x03", "0x00", "0x00", "0x00", "0x14"],
			"ClientIdPos": 0,
			"LiveDataFunctionCodePos": 1,
			"LiveDataFunctionCode": "0x04",
			"IdDataFunctionCodePos": 1,
			"IdDataFunctionCode": "0x03",
			"LiveDataStartsAtPos": 3,
			"IdDataStartsAtPos": 3,
			"LiveDataErrorPos": 1,
			"LiveDataErrorCode": "0x84",
			"IdDataErrorPos": 1,
			"IdDataErrorCode": "0x83",
			"LiveDataSuccessPos": 1,
			"LiveDataSuccessCode": "0x04",
			"IdDataSuccessPos": 1,
			"IdDataSuccessCode": "0x03"

sind diese Daten von Relevanz?

wwwiesel, 2024-01-18:
Ah ok, das würde auch erklöären warum ich das ganze teil abschieße sobald ich den neuen sketch update?

wwwiesel, 2024-01-18:
Nabend, ich habe es nicht zum laufen bekommen, jedoch eine Alternative gefunden die für mich besser passt. Danke für die Mühe

tobiasfaust, 2024-01-19:
Moin,
Ich verstehe leider nicht, die 9600 baue stellst du doch in der weboberfläche ein, nicht im Sketch.
Im Sketch wird nur die Register.h angepasst.

Etwas tricky ist auch die korrekte Verkabelung zum WR sodass sowohl die RX als auch die TX LED des Rs485 Moduls blinkt.

Aber schade, ich hätte gerne die Register.h für deinen WR in mein Repo aufgenommen.

wwwiesel, 2024-01-19:
Guten Morgen,

Okay. Dann helfe ich gerne, ich habe noch ein paar esp32 und rs452 Platinen rumliegen.

Die baudrate 9600 habe ich auf der Webseite eingestellt, korrekt, sobald ich aber eine von mit kompilierte Version mit den MOD settings verwendet habe konnte ich den modbus nicht mehr anschließen, da sonst der esp nur noch per ping erreichbar war.

Mal was anderes, kann man das ggf nicht anders lösen als immer neu kompilieren? Das würde es einfacher machen. Zudem ist es etwas unübersichtlich wenn alle Hersteller in einer Datei sind. Ich bin jetzt nicht der große Entwickler, ich kann Python 😂 dann hörts auf.

wwwiesel, 2024-01-22:
Ich werde die woche mir einen raspberry zusammen bauen, dann kann ich den Fernbedienen mit der seriellen schnittstelle...

Die Arduino Test datei werde ich dann auch ausprobieren...

tobiasfaust, 2024-01-24:
ich habe alles auf Files umgestellt, dafür muss ab sofort bei der Installation immer auch das littlefs.bin geflashed werden. Das Wiki muss ich noch anpassen.
Eine eigene Seite auf der Weboberfläche des ESP zur Pflege des Inhaltes der Dateien kommt noch

tobiasfaust, 2024-02-02:
in der aktuellen Dev gibt es jetzt eine "Files" sektion, hier kann man via Web an den jsons arbeiten, lokal herunterladen und auch auf dem ESP wieder abspeichern

tobiasfaust, 2024-02-22:
Gibt es hier Fortschritte?

wwwiesel, 2024-02-23:
Hi. leider noch nicht, ich war die letzte Zeit etwas eingespannt, ich denke nächste Woche habe ich ein erstes Setup

nobbiman, 2024-02-25:
Muss ich davon ausgehen, dass der Growatt 4600 TL-XH nicht ausgelesen werden kann oder hat meine Fehlermeldung beim Verbindungsaufbau im ioBroker
`On error: {"errno":-111,"code":"ECONNREFUSED","syscall":"connect","address":"192.168.2.197","port":502}`
ein anderes Problem?
Hardwaremäßig habe ich einen ESP32 mit dem Breakout Board wie im Projekt vorgestellt verwendet.

tobiasfaust, 2024-02-26:
> Muss ich davon ausgehen, dass der Growatt 4600 TL-XH nicht ausgelesen werden kann oder hat meine Fehlermeldung beim Verbindungsaufbau im ioBroker `On error: {"errno":-111,"code":"ECONNREFUSED","syscall":"connect","address":"192.168.2.197","port":502}` eine andere Ursache? Hardwaremäßig habe ich einen ESP32 mit dem Breakout Board wie im Projekt vorgestellt verwendet.

Du versucht per Modbus-TCP drauf zu gehen. Mit dem Projekt funktioniert der Connect aber per Modbus-RTU

die Fehlermeldung kommt nicht aus meinem Projekt heraus ;)

nobbiman, 2024-02-26:
Hallo, Danke für deine Rückmeldung.

Ja, ich verstehe zu wenig davon, will aber unbedingt Zugriff auf die Holdingregister des WR haben um gewisse Parameter steuern zu können.

Deshalb suche ich Hilfe.

Nach deiner Antwort habe ich in der Konfiguration meines ioBrokers die im Screenshot eingestellten Werte verändert (von TCP auf RTU over TCP), im Ergebnis erhalte ich die gleiche Fehlermeldung.

 

Andererseits versuche ich den Zugang auf das WebUI zu bekommen, weiß aber nicht wie ich darauf zugreifen kann.

Kannst du mir da weiterhelfen?

nobbiman, 2024-02-26:
Nochmals, das vorhin Screenshot vergessen:

 

Hallo, Danke für deine Rückmeldung.

Ja, ich verstehe zu wenig davon, will aber unbedingt Zugriff auf die Holdingregister des WR haben um gewisse Parameter steuern zu können.

Deshalb suche ich Hilfe.

Nach deiner Antwort habe ich in der Konfiguration meines ioBrokers die im Screenshot eingestellten Werte verändert (von TCP auf RTU over TCP), im Ergebnis erhalte ich die gleiche Fehlermeldung.

 

Andererseits versuche ich den Zugang auf das WebUI zu bekommen, weiß aber nicht wie ich darauf zugreifen kann.

Kannst du mir da weiterhelfen?

tobiasfaust, 2024-02-26:
Bitte nicht per Email antworten!

einfach die ipadresse des esp im Router suchen und im Web aufrufen

nobbiman, 2024-02-26:
Die IP-Adresse habe ich ja, leider kann ich das WebUI nicht erreichen, ein Ping ist aber erfolgreich! ???
Ist evtl. ein Aderntausch hilfreich?

wwwiesel, 2024-02-28:
Hi, ich bin gerade am flashen, jedoch frist mein ESP32 die Files, die ich in GitPod kompiliert habe, nicht. 

BZW der flashed erfolgreich nur sehe ich kein WIFI.

��@;���WiFi access point started

WiFi access point  stopped

WiFi access point started

Completed scan for access points

[ 63603][E][WiFiSTA.cpp:317] begin(): connect failed! 0x300a

Completed scan for access points

[ 76632][E][WiFiSTA.cpp:317] begin(): connect failed! 0x300a

Completed scan for access points

[ 89661][E][WiFiSTA.cpp:317] begin(): connect failed! 0x300a

Completed scan for access points

[102691][E][WiFiSTA.cpp:317] begin(): connect failed! 0x300a

Completed scan for access points

[115920][E][WiFiSTA.cpp:317] begin(): connect failed! 0x300a

Completed scan for access points

[129149][E][WiFiSTA.cpp:317] begin(): connect failed! 0x300a

Completed scan for access points

[142377][E][WiFiSTA.cpp:317] begin(): connect failed! 0x300a

Releases:
https://github.com/tobiasfaust/SolaxModbusGateway/releases/tag/v.3.1.1--7566573188
.

nobbiman, 2024-03-21:
So, der ESP ist über das UI erreichbar, Probleme hat offenbar die instabile WLAN-Verbindung produziert.
Leider kommen noch keine Daten rüber, am COM-Stecker des WR habe ich die PINs 3+4 gewählt (s. Anhang)
Die Configuration der Modbus-SS habe ich als Screenshot der Anlage beigefügt.
Ob mir wohl jemand auf die Sprünge helfen kann?
[image]
[image]
[image]
[image]

tobiasfaust, 2024-03-21:
Eventuell die Modbus Adern tauschen, also nicht 3+4 sondern 4+3
Welchen Modbus Adapter hast du? Welche LEDs leuchten auf dem Adapter auf?
Da hilft leider nur testen und probieren.

nobbiman, 2024-03-22:
Bin wirklich fast am Ende mit meinen Nerven nach mehreren Stunden der Fehlersuche, da ich nun doch keine Verbindung mehr habe und einfach den Fehler nicht finde - ich befürchte, ich muss das Mosbus-Thema aufgeben. Ich habe alle Lötstellen und Jumperkabel überprüft, aber keinen Fehler gefunden.
Ich wollte nun nochmals den Adapter per ESP-Flashtool flashen, finde aber ums Verrecken nicht mehr die Dateien, die aufzuspielen sind. Im Moment geht gar nichts mehr :-( 
Ich habe einen ESP32 (32082A) mit dem Breakout Board HW-0519 wie im Projekt vorgestellt verwendet.
LED: auf dem ESP-Board leuchtet nur die PWR-LED, am Breakout Board blinkt nach dem Anstecken der Spannungsversorgun anfangs TXD, dann aber nicht mehr.
In den Settings des Inverters (MIN 4600 TL-XH) finde ich keine Einstellung zur Modbus-ID, wo finde ich diese Einstellung?
Ist in der Configuration die Inverter SerialNumber relevant? Wenn JA, wo ist diese einzutragen bzw. zieht sich der Adapter diese selbst?
Wer will mir helfen?

tobiasfaust, 2024-03-22:
Die Dateien bekommst du wenn du über gitpod kompilierst, siehe Wiki
Schau auch mal in die FaQ‘s, da steht auch viel (hoffentlich) drin. 
Eine der LEDs muss alle 5sek aufleuchten wenn du ein Intervall von 5sek eingestellt hast. Wenn diese nun zyklisch leuchtet ist zumindest das Modbus Board korrekt am ESP angeschlossen.

nobbiman, 2024-03-23:
Nun bin ich doch noch einen Schritt weiter gekommen, habe aber nicht den Eindruck, als ob das Flashen klappt:
beiliegender Screenshot zeigt keinen Fortschrittsbalken, am Breakout Board blinkt nun (ohne an der COM-SS angeschlossen) die TXD-LED.
[image]

tobiasfaust, 2024-03-23:
Geh mal bei dir in den Geräte Manager.
Ich wette das bei dir der Treiber für das Board fehlt. Dazu gibt aber Tonnen von Erklärungen im Netz ;)

closed 2024-04-25 by tobiasfaust