tobiasfaust/SolaxModbusGateway #44: v3.2.0 mit ESP32-S2 nicht verwendbar

Reinhard-M, 2024-02-10:
Ich habe versucht die v3.2.0 auf einen ESP32-S2 mini zu bringen. Inklusive littlefs.bin. Das funktioniert grundsätzlich, WIFI lässt sich konfigurieren und das Device ist dann in meinem LAN erreichbar. Allerdings sieht die Oberfläche anders aus. Statt der Tabelle mit gelbem Hintergrund scheint alles ohne Formatierung auf dem Bildschirm zu stehen. Außerdem lassen sich Änderungen an den Settings nicht abspeichern, mit einem "speichern" springt alles wieder auf 'Default'.
[image]

tobiasfaust, 2024-02-10:
Irgendetwas passt nicht mit der littlefs.bin.
Die ganzen Dateien sind nicht da. Sicher das du diese auf den ESP geschrieben hast? 
dasselbe auch in deinem anderen Issue

Reinhard-M, 2024-02-10:
Das ich alles hochgeladen habe - absolut sicher, hier die Terminalausgabe:

[image]

Auf dem Screenshot sieht man die v3.1.1, daher nur 3 Files. Ich habe das mehrfach auf verschiedenen Devices mit verschiedenen Versionen getestet soweit es ging )aktuellen 3.2.0 und 3.1.1). Die 3.1.1 lässt sich zumindest starten, es fehlen aber alle "Item Config" Einträge. Für alle Wechselrichter. Inzwischen habe ich VS installiert, gleiches Ergebnis. Mit der Vorgänger 3.1.1 vom Januar hat es noch funktioniert, keine Ahnung was da passiert ist.

tobiasfaust, 2024-02-10:
Dann schau mal bitte dir die debugausgaben auf der seriellen Konsole an. 
auf jeden Fall werden die files (css, js, Register) nicht ausgeliefert. Daran liegt es.

Reinhard-M, 2024-02-11:
Jetzt stehe ich tatsächlich ein wenig auf dem Schlauch. Bislang habe ich lediglich das Binary genommen und in den ESP geladen. Die v3.2.0 habe ich mit Gitpod erstellt und habe keinen Schimmer was da beim kompilieren abläuft. Was muss ich machen um die serielle Konsole (ich vermute vom ESP) zu aktivieren und auf welchen ESP Ports liegt sie? Sorry, ich weiß gerade wirklich nicht wie ich hier weitermachen soll.

tobiasfaust, 2024-02-11:
Bislang war alles css, JavaScript und html im Code hart eincodiert. Ab der 3.2.0 sind diese als echte Dateien im Flash Speicher ausgelagert. Erst damit ist es möglich, die registerdateien im laufenden Betrieb auf die eigenen Bedürfnisse anzupassen.

Ich habe gelesen, das der ESP32-S2 mit dem littleFS unter Umständen ein Problem hat. Siehe oben die Links. Muss man ausprobieren, Ich selbst habe keinen S2. Der normale ESP32 hat kein Problem.

wie man an die Logausgaben auf der seriellen Konsole kommt steht bei mir im Wiki unter den FaQ‘s

Reinhard-M, 2024-02-11:
Ok, schaue ich mir an. Danke für die schnelle Antwort

Reinhard-M, 2024-02-11:
Die Weboberfläche ist noch ein wenig nüchtern, aber die v3.2.0 läuft jetzt auch auf dem ESP32-S2 :-)
Ich muss noch das ein oder andere zurück bauen, im VSC konnte ich aber beobachten, dass das Filesystem auf 0x290000 statt auf 0x180000 geladen wird. Mit dem Umstellen der Adresse im Flash Download Tool funktionierte es dann auch bei mir. Super! Ich weiß nicht woher die Partition Table stammt, beim ESP32-S2 scheint sie aber anders belegt zu sein als bei ESP32. Falls du das ins Wiki übernehmen möchtest...

tobiasfaust, 2024-02-11:
Super, mache ich.

was meinst du mit „nüchtern“?

Reinhard-M, 2024-02-11:
So wie ich es bereits ganz oben gezeigt hatte. Hier ein Ausschnitt der Item Config Seite:
[image]
Kann natürlich auch daran liegen, dass bei mir noch nicht alles stimmt. Aufräumen kommt später, funktioniert, das ist wichtig.

tobiasfaust, 2024-02-12:
Ja, das css wird nicht ausgeliefert.
Schau mal die Seite als source an und öffne direkt die css.
Und bitte schauen was auf der seriellen Konsole der esp ausgibt

Reinhard-M, 2024-02-14:
Wie gesagt, beim ESP32-S2 ist die serielle Konsole nur im Download Mode zugänglich. Die Datenleitungen vom USB hängen direkt auf dem ESP32-S2. Wenn es außer dem keine andere serielle Konsole gibt habe ich keine Chance.
Was ich auf dem ESP32-S2 sehen kann ist folgendes:

[image]

Das style.css ist zumindest vorhanden. Die verschiedenen Werte meiner Solax werden auf der Item Config auch angezeigt. Leider kenne ich mich nicht CSS/HTML Umfeld im Grunde überhaupt nicht aus, was genau meinst du mit "wird nicht ausgeliefert"? Auf dem ESP ist es ja angekommen wie alle anderen Files aus dem data Verzeichnis auch. VSC mit allem Drum un Dran habe ich zwar bei mir gerade installiert, viel machen kann ich allerdings nicht damit. Beim Upload über Platformio sehe ich zwar eine Menge "depricated" und einige "Wrong Type" Warnings, das sollte aber eigentlich nichts ausmachen. Oder doch?
BTW, im "main.cpp" steht noch ein Passwort. Das hat zumindest bei mir jedes Mal Probleme mit der Ersteinrichtung gemacht.

tobiasfaust, 2024-02-14:
Das ist merkwürdig, seitens ESP scheint alles korrekt zu laufen. Warum dein Browser das css nicht anwendet kann ich nicht sagen. Ggf müsstest du mal in deinem Browser über die Entwicklerwerkzeuge die Konsole aufmachen ob der Browser darüber Fehlermeldungen ausgibt. CSS macht aber nur Optik.

das Passwort in der Main.cpp muss gesetzt sein da es das initiale Passwort im AP Mode angibt. Ansonsten würde der esp ein ungesichertes offenes Wlan aufmachen.

tobiasfaust, 2024-02-14:
Ich seh den Fehler. Der Name der Style.css muss mit großem S beginnen.
Kannst du über die Seite aber selbst ändern

Reinhard-M, 2024-02-14:
Perfekt, das passt, Fehler behoben :-)
Also nur 2 Dinge die nicht ganz richtig laufen beim ESP32-S2:
- Basisadresse für littlefs.bin ist 0x290000 statt 0x180000
- Der Name des css muss Style.css statt style.css lauten
Das einzige was ich noch nicht ganz verstanden habe - warum läuft im VSC der Build von bootloader, firmware, partitions getrennt vom Build für littlefs? Über gitpod werden alle 4 mit einem Start des Gitpod erzeugt. Ist aber wirklich nicht wichtig, besten Dank für die Hilfe und die schnuckelige Software :-)

closed 2024-02-14 by Reinhard-M

tobiasfaust, 2024-02-15:
- Basisadresse für littlefs.bin ist 0x290000 statt 0x180000
-> habe ich im Wiki schon angepasst
- Der Name des css muss Style.css statt style.css lauten
-> habe ich gerade im Repo angepasst
- warum läuft im VSC der Build von bootloader, firmware, partitions getrennt vom Build für littlefs? Über gitpod werden alle 4 mit einem Start des Gitpod erzeugt. 
-> das ist leider so. Kann ich nicht ändern. Im Gitpod habe ich beide Befehle dafür eingetragen, deswegen werkst du es nicht , siehe gitpod.yml