tobiasfaust/SolaxModbusGateway #46: Wrong openWB MQTT Support for v1.9.303.0

meisterwinter, 2024-02-14:
Here are the right MQTT topics for openWB v1.9.303.0. The nighly Build has the same Issue.

openWB/set/evu/W Bezugsleistung in Watt, int, positiv Bezug, negativ Einspeisung
openWB/set/evu/APhase1 Strom in Ampere für Phase 1, float, Punkt als Trenner, positiv Bezug, negativ Einspeisung
openWB/set/evu/APhase2 Strom in Ampere für Phase 2, float, Punkt als Trenner, positiv Bezug, negativ Einspeisung
openWB/set/evu/APhase3 Strom in Ampere für Phase 3, float, Punkt als Trenner, positiv Bezug, negativ Einspeisung
openWB/set/evu/WhImported Bezogene Energie in Wh, float, Punkt als Trenner, nur positiv
openWB/set/evu/WhExported Eingespeiste Energie in Wh, float, Punkt als Trenner, nur positiv
openWB/set/evu/VPhase1 Spannung in Volt für Phase 1, float, Punkt als Trenner
openWB/set/evu/VPhase2 Spannung in Volt für Phase 2, float, Punkt als Trenner
openWB/set/evu/VPhase3 Spannung in Volt für Phase 3, float, Punkt als Trenner
openWB/set/evu/HzFrequenz oder openWB/set/evu/Hz Netzfrequenz in Hz, float, Punkt als Trenner
openWB/set/evu/WPhase1 Leistung in Watt für Phase 1, float, Punkt als Trenner, positiv Bezug, negativ Einspeisung
openWB/set/evu/WPhase2 Leistung in Watt für Phase 2, float, Punkt als Trenner, positiv Bezug, negativ Einspeisung
openWB/set/evu/WPhase3 Leistung in Watt für Phase 3, float, Punkt als Trenner, positiv Bezug, negativ Einspeisung
openWB/set/evu/PfPhase1 Powerfaktor für Phase 1, float, Punkt als Trenner, positiv Bezug, negativ Einspeisung
openWB/set/evu/PfPhase2 Powerfaktor für Phase 2, float, Punkt als Trenner, positiv Bezug, negativ Einspeisung
openWB/set/evu/PfPhase3 Powerfaktor für Phase 3, float, Punkt als Trenner, positiv Bezug, negativ Einspeisung

openWB/set/pv/1/W PV-Erzeugungsleistung in Watt, int, positiv
openWB/set/pv/1/WhCounter Erzeugte Energie in Wh, float, nur positiv

openWB/set/pv/2/W PV-Erzeugungsleistung in Watt, int, positiv
openWB/set/pv/2/WhCounter Erzeugte Energie in Wh, float, nur positiv

openWB/set/houseBattery/W Speicherleistung in Watt, int, positiv Ladung, negativ Entladung
openWB/set/houseBattery/WhImported Geladene Energie in Wh, float, nur positiv
openWB/set/houseBattery/WhExported Entladene Energie in Wh, float, nur positiv
openWB/set/houseBattery/%Soc Ladestand des Speichers, int, 0-100

Can you edit this to the right Topics?

Thanks a lot.

mw

tobiasfaust, 2024-02-15:
name of style.css was wrong, its now fixed

One Deye Inverter is available

meisterwinter, 2024-02-15:
> name of style.css was wrong, its now fixed
> 
> One Deye Inverter is available

Thanks! 

Is there a compiled release?

closed 2024-04-14 by tobiasfaust