tobiasfaust/SolaxModbusGateway #48: Growatt WR SPH auslesen

thomas-99, 2024-02-17:
Hallo Tobias,
ich mochte meinen Growatt WR SPH über Modbus auslesen. Da ich mich mit Modbus nicht auskenne, habe ich mir bei eBay ein komplettes Modul gekauft. Der Typ hatte mir geschrieben, dass alles für meinen WR eingestellt ist. Habe gekauft und nichts funktioniert. Er meldet sich nicht mehr, na ja er hat ja das Geld :-(
Ich habe mich etwas eingelesen und hier deine Antworten gelesen. Du hast geschrieben, dass auf dem "RS485 to TTL" Modul RX und TX blinken müssen. Das passiert nicht. Ich wollte wissen, ob der WR überhaupt etwas sendet. Habe mir dazu einen USB Stick gekauft und siehe da, mit der richtigen Software und den korrekten Einstellungen antwortet der WR.
A+ und B- habe ich genauso mit dem „RS485 to TTL" Modul verbunden. Ich kann mich anmelden und die RAW Data senden. Dann müsste doch TX blinken, was nicht funktioniert.
[image]
Der Typ hatte mir geschrieben, dass der WR die Daten sendet , wenn bei RAW Data etwas angezeigt wird. Es wird immer etwas angezeigt, auch wenn der RJ45 Stecker im WR nicht eingesteckt ist. 
Kannst du mir bitte einen Tipp geben, wie ich weiter machen kann?
Danke Ciao thomas
[image]

tobiasfaust, 2024-02-17:
Dein Growatt kann wohl nicht soviel Register auf einmal senden wie angefordert. An deinem gekauften Modul scheint es nicht zu liegen. Die erhältst ja eine Antwort, auch wenn diese nur aus unknown‘s besteht.
Du musst die RequestLiveData anpassen.
Dazu ein besten den ESP mit den aktuellen github stand flashen und dann die growatt json entsprechend über die weboberfläche anpassen.

thomas-99, 2024-02-17:
Danke für deine Rückmeldung. Wenn ich das gekaufte Modul vom WR trenne, kommt die gleiche Rückmeldung. 
[image]
Das kann doch nicht sein? Da ist kein Unterschied, ob ich das Modul einstecke oder trenne. Es wird immer das Gleiche ausgegeben.
Ich habe den Verdacht, dass das "RS485 to TTL" Modul defekt ist. Wenn ich über USB Spannung an den ESP anlege, müßte die RX und TX Led auf dem RS485toTTL kurz aufleuchten (Initialisierung). RX und TX auf dem RS485 to TTL zeigen nie etwas an. 
Ich vermute, das Ding ist defekt.

tobiasfaust, 2024-02-17:
Wenn immer dasselbe angezeigt wird ist es der testmodus. Das bedeutet aber nur das von deinem WR keine Antwort kommt.
RX müsste wenn alles richtig verkabelt ist immer aufleuchten. Auch wenn der WR nicht angeschlossen ist. TX leuchtet nur dann wenn dein WR die Anfrage versteht und antwortet.

thomas-99, 2024-02-18:
Kann ich den Testmodus deaktivieren? 
Spannung liegt an nur keine der LEDs leuchtet, noch nicht einmal, wenn ich SPannung an den USB des ESPs anlege. 
Da der WR korrekt antwortet, was ich mit dem dem USB zu RS485 getestet hatte, liegt es an dem RS485toTTL. 
Habe ein neues RS485toTTL bestellt und dann see ich weiter.

tobiasfaust, 2024-02-18:
der testmodus ist in der modbus.h in deinem release aus versheen fest einkompiliert gewesen.
Du kommst nicht drum herum dein Modul auseinanderzubauen und eine aktuelle Version drauf zu flashen, steht aber alles im Wiki.

thomas-99, 2024-02-20:
Moin Tobias,
ich habe eine neue Firmware erstellt (habe keine Änderungen durchgeführt) und auf den ESP32 geladen (Firmware ESP32). Hat alles funktioniert. Allerdings wird der AP mode vom ESP32 nicht mehr angezeigt im Wlan. 
Es gibt andere Versionen: ESP32-S2, ESP32-C3, ESP32-S3.
Vielleicht habe ich die falsche Version geladen?
[image]
Kann ich irgendwie herausfinden, was ich falsch gemacht habe?
AP Mode hatt auf der alten Version funktioniert. 
DANKE

tobiasfaust, 2024-02-20:
Die Version ist richtig, sonst hätte der Upload Vorgang schon nicht funktioniert, imho
Der AP Mode wird nur gestartet wenn keine WLAN Zugangsdaten vorher eingegeben wurden und der esp sich nicht ins wlan einloggen konnte. 
ggf einfach mal ein paar Minuten warten

thomas-99, 2024-02-20:
Habe viele Minuten gewartet - funktionierte nicht. Dann alles auf Anfang, die Firmware neu erstellt + neu geflasht. Und jetzt wird AP Modus angezeigt. Leider ist keine Verbindung möglich. Gibt es ein neues PW für den AP Modus?
[image]
Notebook bekommt die gleiche Meldung. Habe dein PW kopiert, eingegeben, mehrmals ... es kommt immer, das es ein falsches PW ist.

tobiasfaust, 2024-02-20:
Hab es gerade getestet, ESP32 platt gemacht. Nach 3-5min ging der AP Mode an. Dann verbunden, passwort eingegeben. Alles io.
Wifi Zugangsdaten eigegeben und dann war der ESP im NEtzwerk.... also alles io

thomas-99, 2024-02-20:
OK, alles wieder von vorn + neu geflasht und jetzt kann ich den AP aktivieren + Wlan einrichten. Keine Ahnung, warum das alles so holprig läuft ...
Ich hatte mir ein neues RS485toTTL besogt, was jetzt blinkt auch TR. Test am WR erst heute Abend.  
Danke für deine GEDULD! ;-)

thomas-99, 2024-02-22:
Moin Tobias,
deine Software funktioniert, kann die Daten auslesen. DANKE gute Arbeit!
Kann dein Tool "irgendwie" Daten zurück schreiben? ... über Modbus Register im WR ändern? 

Und deine Meinung als Fachmann:
Aktuell habe ich min. 20 mal die Daten (ohne Änderung) auf den ESP32 neu geflasht und über AP alles neu eingerichtet. Nach 45 min ist alles Offline. Stecker raus und wieder rein hilft nicht. Muss alles neu flashen + einrichten. Und immer wieder das gezicke, PW bei AP falsch oder AP wird auch nach 15 min nicht angezeigt. 
Ist der ESP32 defekt? ... normal ist das sicher nicht ....

thomas-99, 2024-02-27:
Hi Tobias,
das Tool läuft bei mir sehr instabil, was an der Wlan Verbindung liegt. Ich habe gesehen, dass ich umstellen kann auf WT32-ETH01 und habe mir eins besorgt + geflasht. Habe dann in der Verwaltung umgestellt auf WT32-ETH01 + Neustart + Patchkabel zum Router. Jedefallst wurde keine IP per DHCP abgefragt und Wlan ist deaktiviert. 
Hast du das überhaupt schon implementiert oder kommt das erst noch?
DANKE

tobiasfaust, 2024-02-27:
Das ist in der aktuellen Dev schon integriert. Wenn du auf der BaseConfig Seite auf ETH umstellen kannst ist alles io.

im Zweifelsfall musst du bitte mal die Logs auf der  serielle Konsole beobachten.

closed 2024-04-25 by tobiasfaust