tobiasfaust/SolaxModbusGateway #49: Serienummer  Deye

tedesco1968, 2024-02-24:
Hallo Tobias,

habe dein Projekt umgestzt und mit dem Deye 12k verbunden, kann Daten sehen, werden auch zum Iobroker per MQTT übertragen.
Was mich wundert ist das dieInverter Serialnummer von meinem Deye nicht angezeigt wird.
Muss ich da noch irgendwas einstellen?
Gruß
Antonio

tobiasfaust, 2024-02-25:
Was steht denn auf der rawdata Seite? Wie sieht die ID Sektion im json aus?

tedesco1968, 2024-02-25:
Hallo Tobias,
Danke für deine Antwort.
Also erstmal danke für das super Projekt.
 Nun zu meine Problemchen, es ist die aktuellste Version auf meinem esp32, habe auch nichts verändert, nur auf der Auswahlseite auf Deye gestellt, sonst nichts verändert.
[image]

Ein größeres Problem ist das der Esp32 abstürzt, weiß nicht ob es ein Hitze Problem ist da alles in einem Gehäuse betrieben wird, habe heute morgen alles freiluft ohne Gehäuse am laufen, mal sehen ob es daran liegt.
Gruß 
Antonio

tedesco1968, 2024-02-25:
Hallo,

muss ich denn irgendwo was selbst eintragen? Inverter Seriennummer? In der
Deye_SUN_SG04LP3.json?

</body>
</html>

tobiasfaust, 2024-02-25:
Sorry, jetzt fällt es mir wieder ein nachdem ich mir das noch einmal angesehen habe.
Der deye hat eine merkwürdige Spezifikation, die Read Register liegen genauso wie die Input Register auf 0x03.
Normalerweise liegen die Read Register (wie zb. die Seriennummer) auf 0x04, die Inputregister (also die Livedaten) auf 0x03.
Dadurch das hier nicht getrennt wird funktioniert es mit dem aktuellen Code nicht. 
Ist beim Deye jetzt erstmal so….

tedesco1968, 2024-02-25:
Hallo Tobias,

wie kann ich denn irgenwelche Modbus Werte für meinen Deye hinzufügen?
Welche Dateien muss ich alles editieren, kannst du mal ein Beispiel durchgehen? Als Beispiel Adresse 661
[image]

tobiasfaust, 2024-02-26:
Ich kann es nicht besser als hier erklären.
https://github.com/tobiasfaust/SolaxModbusGateway/wiki/adding-new-inverters-or-items#find-out-live-data-get-new-items

Da gibt es auch ein Beispiel, Versuch mal das Beispiel nachzuvollziehen

tedesco1968, 2024-02-26:
Hallo Tobias,

Diese Seite hatte ich noch ni ht gefunden,  jetzt sieht es ganz anders aus.

Gruß 
Antonio

tobiasfaust, 2024-04-02:
Wenn du ein „neues“ Register.json für den Deye hast würde ich mich freuen wenn du es hier für die Nachwelt posten könntest.

tedesco1968, 2024-04-21:
Hallo Tobias,

sorry für die spte Antwort, Nein habe ich nicht, weil alles so schon passt!.

Einziges Problem ist das schon mehrere Personen, ewähnte spinnen, das heiß alle paar sekunden werden irgendwelche wirre zeichen übertragen, woher das kommt keine Ahnung

Gruß
Antonio

tobiasfaust, 2024-04-21:
Bei welchem Wert springen sie bei dir? Also welcher Name aus den Items? Ist die json unverändert?

tedesco1968, 2024-04-21:
> Bei welchem Wert springen sie bei dir? Also welcher Name aus den Items? Ist die json unverändert?

Hallo Tobias,
die springen alle paar sekunden undefiniert, da verändern sich alle Werte komplett.
Ja die json ist unverändert

tobiasfaust, 2024-04-21:
In #43 gab es den Hinweis, das der BMS Port anstatt des Modbus Ports genutzt werden soll

tedesco1968, 2024-04-21:
Hallo Tobias,Alle Items die json blieb unverändert also deine!Von meinem/meiner Galaxy gesendet

closed 2024-09-19 by tobiasfaust