tobiasfaust/SolaxModbusGateway #51: send data to esp and change values on the inverter over mqtt

Coolj1, 2024-02-27:
hallo Tobias,
ist es auch möglich Werte im Inverter zu verändern, und wenn ja was muß ich denn da über mqtt schicken?
Irgend wo habe ich mal gelesen das es geht, jetzt konnte ich dazu nichts mehr finden.
Es wäre top, wenn ich bei meinem Solax x3 g4 die Max-Einspeiseleistung ändern könnte, auch wäre das Ein- und Ausschalten des Inverters nicht schlecht.

danke 
Gruß
Hans

tobiasfaust, 2024-02-28:
Es ist zumindest vorbereitet, aber selbst noch nicht getestet.
Beim Solax-x1 ist ein Beispiel im json dabei.
Und auf der  Modbus Seite ist ein Schalter dafür um die Funktionalität einzuschalten 

und immer schön die serielle Konsole beobachten. 

Nichmal, ungerüstet und auf eigene Gefahr

Coolj1, 2024-02-28:
du meinst vermutlich diesen Teil 

"set": [
      {
        "name": "TargetBatSOC",
        "request" : ["#ClientID", "0x06", "0x00", "0x83"]

aber wie kann ich denn über Mqtt diesen Wert setzen?
Ist das in deiner Software nur für den X1 und nur für diesen Parameter oder auch schon für andere vorbereitet?
... sorry, aber ich bin kein Programmierer.☹️

tobiasfaust, 2024-02-29:
gesetzt wird dieses über ein MQTT Kommando, (kommt aus #1 )
für diesen Fall muss man an das Topic {MqttBasePath}/{HostName}/set/TargetBatSOC den gewünschten Wert senden. Den MqttBasePath und HostName musst du natürlich ersetzen auf deine Einstellungen (Basics ESP-WebSeite)

zb: /home/solax/set/TargetBatSOC  70

um 70% zu setzen

Damit das überhaupt funktioniert muss man in der Regel das Passwort des Wechselrichter vorher einmalig dort in das korrekte Register schreiben. Das habe ich noch nicht implementiert, muss man von Hand machen. Entweder über eine Modifikation des example Sketches (hier braucht man aber Programmierkenntnisse) oder über andere Tools mit denen man per Modbus-RTU Werte schreiben kann.

Für andere Wechselrichter ist es noch nicht implementiert, das muss jeder für deinen WR selbst herausfinden. Ich unterstütze gerne wenn es um Codeanpassungen geht die man vorher selbst eingebaut und getestet hat (-> Github PullRequest) und übernehme auch gerne funktionierende json-Konfigurationen damit andere diese nutzen können.

Coolj1, 2024-02-29:
ok danke schon mal... 
werde ich testen, ob mir das gelingt ist was anderes😉

tobiasfaust, 2024-09-19:
This issue was closed because it has been stalled for 14 days with no activity.

closed 2024-09-19 by tobiasfaust

benhubert, 2024-12-03:
Schade, dass dieses Ticket geschlossen wurde, das ist definitiv ein Feature, das ich für meine Zwecke benötige.

@tobiasfaust kannst du die Sache mit dem Passwort noch einmal näher erläutern? Was ist hier genau notwendig? Muss das Passwort über Modbus an den Wechselrichter geschickt werden? Findet man das in irgendeiner Solax Dokumentation?