tobiasfaust/SolaxModbusGateway #53: ESP32-S3 littlefs fehlt | Bootloader Register 0x0

Apolli, 2024-03-07:
Hi,

ich hab eine [ESP32-S3](https://de.aliexpress.com/item/1005006266375800.html?srcSns=sns_WhatsApp&spreadType=socialShare&bizType=ProductDetail&social_params=60530540629&aff_fcid=d562efa5563b461fabd4f2b2279690c7-1709836931474-01316-_EQo79NR&tt=MG&aff_fsk=_EQo79NR&aff_platform=default&sk=_EQo79NR&aff_trace_key=d562efa5563b461fabd4f2b2279690c7-1709836931474-01316-_EQo79NR&shareId=60530540629&businessType=ProductDetail&platform=AE&terminal_id=27c25e530e0d46dd812de5d5b9431c22&afSmartRedirect=y) und versuche ihn seit einigen Tagen zu flashen. Grundsätzlich funktioniert das auch, allerdings scheint das littlefs nicht gefunden zu werden und somit funktioniert auch die Oberfläche nicht.

Beim starten sehe ich im Log:
19:33:46.411 -> Attempting MQTT connection as Solax-46482 
19:33:46.504 -> connected... 
19:33:52.576 -> [ 16162][E][vfs_api.cpp:105] open(): /littlefs/web/Style.css does not exist, no permits for creation
19:33:52.576 -> failed to open requested file: web/Style.css[ 16181][E][vfs_api.cpp:105] open(): /littlefs/web/Javascript.js does not exist, no permits for creation
19:33:52.607 -> failed to open requested file: ��?.js[ 16486][E][vfs_api.cpp:105] open(): /littlefs/web/JsAjax.js does not exist, no permits for creation
19:33:52.901 -> failed to open requested file: web/JsAjax.js[ 39856][E][vfs_api.cpp:105] open(): /littlefs/web/Style.css does not exist, no permits for creation
19:34:16.264 -> failed to open requested file: web/Style.css[ 39875][E][vfs_api.cpp:105] open(): /littlefs/web/Javascript.js does not exist, no permits for creation
19:34:16.264 -> failed to open requested file: �a�?.js[ 39893][E][vfs_api.cpp:105] open(): /littlefs/web/JsAjax.js does not exist, no permits for creation
19:34:16.302 -> failed to open requested file: web/JsAjax.js

Im Browser bekomme ich einen Fehler 404 Not Found
[image]

Dabei ist mir aufgefallen das auch der Bootloader in das Register 0x0 geschrieben werden muss. Daher die Frage ob das Register für das littlefs auch woanders hin muss?

Aktuell sieht das bei mir so aus:
[image]

Danke für deine Hilfe,
Andreas

tobiasfaust, 2024-03-08:
Hi
Im Wiki steht’s bzw ist auch Bild vom flashtool mit den korrekten Einstellungen.

lattemacchiato, 2024-03-31:
Hallo,
ich habe auch das Problem, dass die littlefs.bin fehlt :-(
Wenn ich per Gitpod baue, wird die littlefs.bin mit gebaut.
Wenn ich das repo clone, und lokal per vscode baue, werden alle bins gebaut - bis auf die littlefs.bin
Ich habe sicher was übersehen (Wiki habe ich durchforstet...), sorry - wie bekomme ich die littlefs.bin lokal mitgebaut?
Danke Euch...
Paul

tobiasfaust, 2024-04-01:
Wiki muss ich noch anpassen, im VSCode muss man explizit „Build Filesystem“ anklicken. 
Im gitpod sind es auch 2 Befehle (siehe gitpod.yml)

lattemacchiato, 2024-04-01:
Danke!
--
// 42 //

--18e9a1d16176ac281079dbab8
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: quoted-printable

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.=
w3.org/TR/html4/loose.dtd"><html>
<head>
<meta http-equiv=3D"Content-Type" content=3D"text/html; charset=3Dutf-8"></=
head>
<body>
<div dir=3D"auto">
<div dir=3D"auto">Danke! </div><div id=3D"aqm-signature" dir=3D"auto" =
style=3D"color: black;"><div dir=3D"auto">--</div><div dir=3D"auto">// 42 /=
/</div><div dir=3D"auto">--</div></div><div dir=3D"auto"><br></div>
<div id=3D"aqm-original" style=3D"color: black;">

<div class=3D"aqm-original-body">
<div style=3D"color: black;">
<p style=3D"color: black; font-size: 10pt; font-family: sans-serif; margin:=
 8pt 0;">Am 1. April 2024 16:21:28 schrieb tobiasfaust ***@***.***=
thub.com&gt;:</p>
<blockquote type=3D"cite" class=3D"gmail_quote" style=3D"margin: 0 0 0 0.75=
ex; border-left: 1px solid #808080; padding-left: 0.75ex;">
<p></p>
<p dir=3D"auto">Wiki muss ich noch anpassen, im VSCode muss man explizit =
=E2=80=9EBuild Filesystem=E2=80=9C anklicken.<br>
Im gitpod sind es auch 2 Befehle (siehe gitpod.yml)</p>

<p style=3D"font-size:small;-webkit-text-size-adjust:none;color:#666;">=E2=
=80=94<br>
sfaust/SolaxModbusGateway/issues/53#issuecomment-2029835078">view it on Git=
HKVUE4WOYWUKYZBKY4NE3Y3FUOLAVCNFSM6AAAAABELQ47T2VHI2DSMVQWIX3LMV43OSLTON2WK=
Q3PNVWWK3TUHMZDAMRZHAZTKMBXHA">unsubscribe</a>.<br>You are receiving this b=
ecause you commented.<img src=3D"https://github.com/notifications/beacon/AM=
HKVUF3NUMCQ2MONB6VX3TY3FUOLA5CNFSM6AAAAABELQ47T2WGG33NNVSW45C7OR4XAZNMJFZXG=
5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTY7TJUM.gif" height=3D"1" width=3D"1" alt=
=3D""><span style=3D"color: transparent; font-size: 0; display: none; visib=
ility: hidden; overflow: hidden; opacity: 0; width: 0; height: 0; max-width=
odbusGateway/issues/53/2029835078</span><span>@</span><span>github</span><s=
pan>.</span><span>com&gt;</span></span></p>
</blockquote>
</div>
</div>

</div><div dir=3D"auto"><br></div>
</div></body>
</html>

--18e9a1d16176ac281079dbab8--

closed 2024-04-25 by tobiasfaust