tobiasfaust/SolaxModbusGateway #56: Solax X3 Hybrid G4 issies with 3.2.0

ingof, 2024-03-13:
Hello Tobias,

I created a pull request to add Cell_Voltage_High and Cell_Voltage_Low on my X3 Hybrid.
After building and burning to the ESP32 i have got two issues.

In the RAW-Data there are less values. With 3.1.5 i get over 400 Value (see screenshot in pull request).
But if i want to check now the new values i get 0xundefined instead of the data and get only 322 bytes RAW data.
[image]

But the new data in live view works fine:
[image]

But there is a problem with "Off Grid Current L1" it shows now randomly strange values and should be zero, because the EPS Mode is not active at the moment.

Did you change the data size of the modbus request in 3.2.0?
Is  it possible to set it in the web interface if i need more raw-data?

ingof, 2024-03-13:
Ok now i get only 319 bytes of data and it seems it now stable at 319 bytes.

tobiasfaust, 2024-03-15:
HI Ingo,
if you get "undefined", in the most cases you request more registers than your inverter is able to deliver. Did you change somthing at "RequestLiveData" Item in json?

Its now fine? Can i merge your pullrequest?

closed 2024-04-25 by tobiasfaust