tobiasfaust/SolaxModbusGateway #59: Letztes Register wirf falsch ausgegeben

lattemacchiato, 2024-04-01:
Hi,
ich will gar nicht ausschließen dass ich was falsch mache, aber für mein QCells (=Solax-X3) werden in der Item Config unplausible und stark variierende Werte angezeigt (sollte "0" sein). Nach einigem Herumexperimentieren habe ich gesehen dass der fragliche Wert in Register 0x77 steht - und das ist offenbar genau der letzte Wert der ersten gelesenen Reihe an Werten. Wenn ich ein Register mehr lesen lasse (sprich bis 0x78 statt 0x77) dann stimmt der Wert. Wenn ich nun alle nachfolgenden position-Werte um 2 erhöhe stimmen auch diese wieder...
Ist das mit 0x77 ein Fehler, oder habe ich einen Denkfehler?
Grüße,

Paul

tobiasfaust, 2024-04-01:
Hi,
Ich weiß das andere den Qcells WR schon erfolgreich mit Solax-X3 Einstellung am Laufen haben. 
77 heißt auch das ab der Startposition genau 119 Register gelesen werden

lattemacchiato, 2024-04-01:
Hi,

Ja, läuft auch - nur das eine (letzte) Register wird halt falsch angezeigt 
bzw ausgewertet. Die anderen stimmen alle...
--
// 42 //

--18e9a1ca3ce6ac28109f1adb7
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: quoted-printable

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.=
w3.org/TR/html4/loose.dtd"><html>
<head>
<meta http-equiv=3D"Content-Type" content=3D"text/html; charset=3Dutf-8"></=
head>
<body>
<div dir=3D"auto">
<div dir=3D"auto">Hi, </div><div dir=3D"auto"><br></div><div dir=3D"au=
to">Ja, l=C3=A4uft auch - nur das eine (letzte) Register wird halt falsch a=
ngezeigt bzw ausgewertet. Die anderen stimmen alle...</div><div id=3D"aqm-s=
ignature" dir=3D"auto" style=3D"color: black;"><div dir=3D"auto">--</div><d=
iv dir=3D"auto">// 42 //</div><div dir=3D"auto">--</div></div><div dir=3D"a=
uto"><br></div>
<div id=3D"aqm-original" style=3D"color: black;">

<div class=3D"aqm-original-body">
<div style=3D"color: black;">
<p style=3D"color: black; font-size: 10pt; font-family: sans-serif; margin:=
 8pt 0;">Am 1. April 2024 16:19:47 schrieb tobiasfaust ***@***.***=
thub.com&gt;:</p>
<blockquote type=3D"cite" class=3D"gmail_quote" style=3D"margin: 0 0 0 0.75=
ex; border-left: 1px solid #808080; padding-left: 0.75ex;">
<p></p>
<p dir=3D"auto">Hi,<br>
Ich wei=C3=9F das andere den Qcells WR schon erfolgreich mit Solax-X3 Einst=
ellung am Laufen haben.<br>
77 hei=C3=9Ft auch das ab der Startposition genau 119 Register gelesen werd=
en</p>

<p style=3D"font-size:small;-webkit-text-size-adjust:none;color:#666;">=E2=
=80=94<br>
sfaust/SolaxModbusGateway/issues/59#issuecomment-2029832384">view it on Git=
HKVUDMPVIESO4GPRVJMYLY3FUIBAVCNFSM6AAAAABFRTE4USVHI2DSMVQWIX3LMV43OSLTON2WK=
Q3PNVWWK3TUHMZDAMRZHAZTEMZYGQ">unsubscribe</a>.<br>You are receiving this b=
ecause you authored the thread.<img src=3D"https://github.com/notifications=
/beacon/AMHKVUEPO45EVHIVQF4NAUDY3FUIBA5CNFSM6AAAAABFRTE4USWGG33NNVSW45C7OR4=
XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTY7TEMA.gif" height=3D"1" width=
=3D"1" alt=3D""><span style=3D"color: transparent; font-size: 0; display: n=
one; visibility: hidden; overflow: hidden; opacity: 0; width: 0; height: 0;=
ust/SolaxModbusGateway/issues/59/2029832384</span><span>@</span><span>githu=
b</span><span>.</span><span>com&gt;</span></span></p>
</blockquote>
</div>
</div>

</div><div dir=3D"auto"><br></div>
</div></body>
</html>

--18e9a1ca3ce6ac28109f1adb7--

lattemacchiato, 2024-04-01:
PS: in der json die erste Wertereihe bis 0x78 lesen statt bis 0x77, dann 
wird das 0x77 (also dec 241,242) richtig gelesen bzw angezeigt - weil es 
dann das vorletzte ist...
--
// 42 //

--18e9a240e396ac2810c21445d
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: quoted-printable

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.=
w3.org/TR/html4/loose.dtd"><html>
<head>
<meta http-equiv=3D"Content-Type" content=3D"text/html; charset=3Dutf-8"></=
head>
<body>
<div dir=3D"auto">
<div dir=3D"auto">PS: in der json die erste Wertereihe bis 0x78 lesen statt=
 bis 0x77, dann wird das 0x77 (also dec 241,242) richtig gelesen bzw angeze=
igt - weil es dann das vorletzte ist... </div><div id=3D"aqm-signature=
" dir=3D"auto" style=3D"color: black;"><div dir=3D"auto">--</div><div dir=
=3D"auto">// 42 //</div><div dir=3D"auto">--</div></div>
<div id=3D"aqm-original" style=3D"color: black;">

<div class=3D"aqm-original-body">
<div style=3D"color: black;">
<p style=3D"color: black; font-size: 10pt; font-family: sans-serif; margin:=
 8pt 0;">Am 1. April 2024 16:19:47 schrieb tobiasfaust ***@***.***=
thub.com&gt;:</p>
<blockquote type=3D"cite" class=3D"gmail_quote" style=3D"margin: 0 0 0 0.75=
ex; border-left: 1px solid #808080; padding-left: 0.75ex;">
<p></p>
<p dir=3D"auto">Hi,<br>
Ich wei=C3=9F das andere den Qcells WR schon erfolgreich mit Solax-X3 Einst=
ellung am Laufen haben.<br>
77 hei=C3=9Ft auch das ab der Startposition genau 119 Register gelesen werd=
en</p>

<p style=3D"font-size:small;-webkit-text-size-adjust:none;color:#666;">=E2=
=80=94<br>
sfaust/SolaxModbusGateway/issues/59#issuecomment-2029832384">view it on Git=
HKVUDMPVIESO4GPRVJMYLY3FUIBAVCNFSM6AAAAABFRTE4USVHI2DSMVQWIX3LMV43OSLTON2WK=
Q3PNVWWK3TUHMZDAMRZHAZTEMZYGQ">unsubscribe</a>.<br>You are receiving this b=
ecause you authored the thread.<img src=3D"https://github.com/notifications=
/beacon/AMHKVUEPO45EVHIVQF4NAUDY3FUIBA5CNFSM6AAAAABFRTE4USWGG33NNVSW45C7OR4=
XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTY7TEMA.gif" height=3D"1" width=
=3D"1" alt=3D""><span style=3D"color: transparent; font-size: 0; display: n=
one; visibility: hidden; overflow: hidden; opacity: 0; width: 0; height: 0;=
ust/SolaxModbusGateway/issues/59/2029832384</span><span>@</span><span>githu=
b</span><span>.</span><span>com&gt;</span></span></p>
</blockquote>
</div>
</div>

</div><div dir=3D"auto"><br></div>
</div></body>
</html>

--18e9a240e396ac2810c21445d--

tobiasfaust, 2024-04-01:
Es kann sein das in der Modbus Doku steht das 77 und 78 zusammen gehört, weiß ich nicht. Dann muss man natürlich auch bis 78 lesen.
Das das letzte Byte grundsätzlich Probleme macht, davon habe ich noch nichts gehört.
Aber wenn es so bei dir passt macht dir eben ein qcells json für dich. Wäre aber schön wenn du es hier für die Nachwelt noch posten kannst wenn es fertig ist. ;)

lattemacchiato, 2024-04-01:
Hi,
lt Solax Doku ist 0x77 ein Einzelwert. Könnte es sein, dass der letzte Wert 
nicht ins Array geschrieben wird und deshalb wahllose Werte drinstehen (so 
was wie "schreibe bis kleiner als" statt "kleiner gleich")? Dafür spräche 
dass der Wert (241,242) stimmt wenn ich bis 0x78 lese, und dann in 243,244 
(0x78) halt was Zufälliges steht.
Ist nur eine Vermutung, vielleicht denke ich da falsch, Sorry.
Das sollte aber kein reines QCells Problem sein, müsste eigentlich beim 
Solax auch so sein.
Aber klar, meine "gepatchte" Json kann ich gerne posten.
--
// 42 //

--18e9a4f3bd86ac2810e2cc1e
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: quoted-printable

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.=
w3.org/TR/html4/loose.dtd"><html>
<head>
<meta http-equiv=3D"Content-Type" content=3D"text/html; charset=3Dutf-8"></=
head>
<body>
<div dir=3D"auto">
<div dir=3D"auto">Hi, </div><div dir=3D"auto">lt Solax Doku ist 0x77 e=
in Einzelwert. K=C3=B6nnte es sein, dass <span style=3D"font-size: 12pt;">d=
er letzte Wert nicht ins Array geschrieben wird und deshalb wahllose Werte =
drinstehen (so was wie &quot;schreibe bis kleiner als&quot; statt &quot;kle=
iner gleich&quot;)? Daf=C3=BCr spr=C3=A4che dass der Wert (241,242) stimmt =
wenn ich bis 0x78 lese, und dann in 243,244 (0x78) halt was Zuf=C3=A4lliges=
 steht. </span></div><div dir=3D"auto"><span style=3D"font-size: 12pt;=
">Ist nur eine Vermutung, vielleicht denke ich da falsch, Sorry. </spa=
n></div><div dir=3D"auto"><span style=3D"font-size: 12pt;">Das sollte aber =
kein reines QCells Problem sein, m=C3=BCsste eigentlich beim Solax auch so =
sein. </span></div><div dir=3D"auto"><span style=3D"font-size: 12pt;">=
Aber klar, meine &quot;gepatchte&quot; Json kann ich gerne posten. </s=
pan></div><div id=3D"aqm-signature" dir=3D"auto" style=3D"color: black;"><d=
iv dir=3D"auto">--</div><div dir=3D"auto">// 42 //</div><div dir=3D"auto">-=
-</div></div><div dir=3D"auto"><br></div>
<div id=3D"aqm-original" style=3D"color: black;">

<div class=3D"aqm-original-body">
<div style=3D"color: black;">
<p style=3D"color: black; font-size: 10pt; font-family: sans-serif; margin:=
 8pt 0;">Am 1. April 2024 17:07:41 schrieb tobiasfaust ***@***.***=
thub.com&gt;:</p>
<blockquote type=3D"cite" class=3D"gmail_quote" style=3D"margin: 0 0 0 0.75=
ex; border-left: 1px solid #808080; padding-left: 0.75ex;">
<p></p>
<p dir=3D"auto">Es kann sein das in der Modbus Doku steht das 77 und 78 zus=
ammen geh=C3=B6rt, wei=C3=9F ich nicht. Dann muss man nat=C3=BCrlich auch b=
is 78 lesen.<br>
Das das letzte Byte grunds=C3=A4tzlich Probleme macht, davon habe ich noch =
nichts geh=C3=B6rt.<br>
Aber wenn es so bei dir passt macht dir eben ein qcells json f=C3=BCr dich.=
 W=C3=A4re aber sch=C3=B6n wenn du es hier f=C3=BCr die Nachwelt noch poste=
n kannst wenn es fertig ist. ;)</p>

<p style=3D"font-size:small;-webkit-text-size-adjust:none;color:#666;">=E2=
=80=94<br>
sfaust/SolaxModbusGateway/issues/59#issuecomment-2029911617">view it on Git=
HKVUB347KOYIQL6C7NVATY3FZ3VAVCNFSM6AAAAABFRTE4USVHI2DSMVQWIX3LMV43OSLTON2WK=
Q3PNVWWK3TUHMZDAMRZHEYTCNRRG4">unsubscribe</a>.<br>You are receiving this b=
ecause you authored the thread.<img src=3D"https://github.com/notifications=
/beacon/AMHKVUBJOJIE3RFI32C34S3Y3FZ3VA5CNFSM6AAAAABFRTE4USWGG33NNVSW45C7OR4=
XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTY7X7EC.gif" height=3D"1" width=
=3D"1" alt=3D""><span style=3D"color: transparent; font-size: 0; display: n=
one; visibility: hidden; overflow: hidden; opacity: 0; width: 0; height: 0;=
ust/SolaxModbusGateway/issues/59/2029911617</span><span>@</span><span>githu=
b</span><span>.</span><span>com&gt;</span></span></p>
</blockquote>
</div>
</div>

</div><div dir=3D"auto"><br></div>
</div></body>
</html>

--18e9a4f3bd86ac2810e2cc1e--

tobiasfaust, 2024-04-02:
Danke, ich schau mir das mal an

tobiasfaust, 2024-04-21:
Da lag glaub ich ein Denkfehler meinerseits vor. Ich bin gerade den ganzen Code durchgegangen, konnte aber kein Codingproblem feststellen.

Der "OffGridCurrent L1" liegt im Register 77. Da die Register aber mit 0 anfangen muss ich also folglich 78 register lesen um auch das 77er register zu lesen. Da in der Antwort als letztes immer ein CRC angehangen wird, wurde als  "OffGridCurrent L1" der CRC versucht auszuwerten.

Problem erkannt, Gefahr gebannt. Ich passe die register jsons an und übernehme deines als neues QCells modul.
Warum als eigenes? Dadurch das du 3 Aufrufe in der RequestLiveData hast, hast du zwar auch mehr an Daten aber auch das Limit auf 3sek angehoben. Jeder Request braucht 1sek. ;)

im Anhang die angepasst Solax-X3.json.
Kann die mal bitte jemand testen? 
[Solax-X3.json](https://github.com/tobiasfaust/SolaxModbusGateway/files/15052669/Solax-X3.json)

lattemacchiato, 2024-04-21:
Hi,
so habe ich das auch gelöst - einfach ein Register mehr lesen ;-)
So ist halt der Wert nicht mehr konsistent, um 77 Register zu lesen muss als Länge 78 angegeben werden. Finde ich etwas unhandlich, ist aber sicher Geschmackssache.
Deine Solax-X3.json liefert bei mir (QVolt) plausible und so weit ich sehe richtige Werte - passt.
Mir ist noch was aufgefallen: Ich lese (ok, mit dem Preis von 3 Sekunden ;-) ) als dritten Bereich die Register ab 0x100. Wenn ich im dritten Bereich als Startregister 0x100 angebe, sind die ausgegebenen Werte Unsinn. Ändere ich das auf 0xFF (also ein Register vorher), dann passt alles. Nur eine Vermutung: Kann das am dreistelligen Wert (0X100) liegen? 
Danke auf jeden Fall!

tobiasfaust, 2024-04-22:
wenn du ab 0x100 ließt musst du eintragen: 0x01 0x00
Ist immer 2-stellig ;) Was gibt es da?
Also entweder liesst man mit dem ersten mehr register (imho sind es max 128 register in einem Request) oder man muss die Requests aufteilen.

ich checke dann das qvolt.json ein

closed 2024-04-22 by tobiasfaust

lattemacchiato, 2024-04-22:
Hi,

Danke erst mal!

Die maximal mögliche Leselänge ist x7D (empirisch ermittelt mit Verdacht 
auf 254 minus 3 für den Header minus 1 für die Checksum) - drüber 
funktioniert es nicht mehr.
Mit 2 mal x7D kommt man bis xFA, also nicht ganz bis x100. Also bleiben nur 
3 Bereiche. xFA liegt noch dazu in einem REV Bereich (xD1 - xFF), also läse 
man etliche ungenutzte Register.
Von x55 bis x65 liegen auch ungenutzte REV Register - die wollte ich mir 
einfach auch sparen ;-)
Über x100 liegt nicht mehr so viel, (Dis)ChargeableElectricCapacity finde 
ich interessant. Außerdem wollte ich die lesen... weil sie da sind ;-)
Anbei ein Screenshot der Werte ab x100, ich hab mir das aus der Spec 
(v3.29) in Excel gezogen damit ich die "positions" nicht händisch berechnen 
muss...

CU

--18f06cceedc54ef28107867615
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: quoted-printable

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.=
w3.org/TR/html4/loose.dtd"><html>
<head>
<meta http-equiv=3D"Content-Type" content=3D"text/html; charset=3Dutf-8"></=
head>
<body>
<div dir=3D"auto">
<div dir=3D"auto">Hi, </div><div dir=3D"auto"><br></div><div dir=3D"au=
to">Danke erst mal! </div><div dir=3D"auto"><br></div><div dir=3D"auto=
">Die maximal m=C3=B6gliche Lesel=C3=A4nge ist x7D (empirisch ermittelt mit=
 Verdacht auf 254 minus 3 f=C3=BCr den Header minus 1 f=C3=BCr die Checksum=
) - dr=C3=BCber funktioniert es nicht mehr. </div><div dir=3D"auto">Mi=
t 2 mal x7D kommt man bis xFA, also nicht ganz bis x100. Also bleiben nur 3=
 Bereiche. xFA liegt noch dazu in einem REV Bereich (xD1 - xFF), also l=C3=
=A4se man etliche ungenutzte Register. </div><div dir=3D"auto">Von x55=
 bis x65 liegen auch ungenutzte REV Register - die wollte ich mir einfach a=
uch sparen ;-) </div><div dir=3D"auto">=C3=9Cber x100 liegt nicht mehr=
 so viel, (Dis)ChargeableElectricCapacity finde ich interessant. Au=C3=9Fer=
dem wollte ich die lesen... weil sie da sind ;-) </div><div dir=3D"aut=
o">Anbei ein Screenshot der Werte ab x100, ich hab mir das aus der Spec (v3=
.29) in Excel gezogen damit ich die &quot;positions&quot; nicht h=C3=A4ndis=
ch berechnen muss... </div><div dir=3D"auto"><br></div><div dir=3D"aut=
o">CU</div><div dir=3D"auto"><br></div><div dir=3D"auto"><br></div><div id=
=3D"aqm-signature" dir=3D"auto" style=3D"color: black;"><div dir=3D"auto">-=
-</div><div dir=3D"auto">// 42 //</div><div dir=3D"auto">--</div></div><div=
 dir=3D"auto"><br></div>
<div id=3D"aqm-original" style=3D"color: black;">

<div class=3D"aqm-original-body">
<div style=3D"color: black;">
<p style=3D"color: black; font-size: 10pt; font-family: sans-serif; margin:=
 8pt 0;">Am 22. April 2024 16:55:38 schrieb tobiasfaust ***@***.***=
ithub.com&gt;:</p>
<blockquote type=3D"cite" class=3D"gmail_quote" style=3D"margin: 0 0 0 0.75=
ex; border-left: 1px solid #808080; padding-left: 0.75ex;">
<p></p>
<p dir=3D"auto">wenn du ab 0x100 lie=C3=9Ft musst du eintragen: 0x01 0x00<b=
r>
Ist immer 2-stellig ;) Was gibt es da?<br>
Also entweder liesst man mit dem ersten mehr register (imho sind es max 128=
 register in einem Request) oder man muss die Requests aufteilen.</p>
<p dir=3D"auto">ich checke dann das qvolt.json ein</p>

<p style=3D"font-size:small;-webkit-text-size-adjust:none;color:#666;">=E2=
=80=94<br>
sfaust/SolaxModbusGateway/issues/59#issuecomment-2069764260">view it on Git=
[... 783 lines cut ...]
Bg4cOEBvb68ohw4MDFy0Y+3ChYunzyJDQdBqCjmtl3oIBT2hEHjmcDjEl/3GjRtFb8+ZM2fIzc3F
x8eHQ4cOodfrKSwsZM+ePWKmG0Fb/GXbe1IE/abNZhN1rUu1K/xf0FQu5P79+1y8eJHLly8THx/P
tm3bftA6c/irltfT0/Oh91Pob5vNJqbBlUgkhISEsGXLFurr6ykvL+fChQtkZmZiNps5fvw4o6Oj
PPvss2zfvl1czAnt+fj4IJVKH7u9J0XQngvxNkuNH0EHDoha4YX9ND8/Ly5ybTYbGo0GlUrF6Ogo
LS0tVFVVMTw8zM9+9jOngM/vI4L0ICAggJ6eHtrb20VDYXJyko6ODiYmJlAqlXR0dNDU1CQuvIzz
8/T39zM7O0tCQoK4KPoihLiX/v5+Kioq0Ov1JCcni9mpLBYL165d48CBA8zMzPDyvn288uqrREVF
IZFImJiYIDw8nD/+8Y9cvXrVybvZ0NBAfX09ycnJ/PKXv6SwsBB3d3ccDgf3798nJCSE//3f/+XK
lSusX7+e5ORk5ufnKS8v5/Tp08hkMnbv3s2ePXvQaDSi/EqtVvP+++9TWlpKbm4u4eHhZGZmkpqa
ikKhEHelVqxYwZ49e0RN+qNIT08nLS2NM2fO0N7RgcFgEBflJpOJuro6BgYGeOaZZ0SZR09PD6dO
naKmpobCwkJ+9atfiV5gs9lMdnY2f/jDH7h16xZlZWWkpaU9dnY84R1069YtmpqacHNzIzY2lqCg
IAwGA6dPn+b06dP4+fnx2muvsX37dvF9MjExgVar5f/+7/84c+YMaWlpaLVaAgIC0Gq1+Pj40Nzc
TH9/v2gojI2NiXVT5HI5dXV16PV6YmNjkUgkGAwGenp6sFqtJCUlERQUhNVq5c6dOxw7dozBwUH2
7dvHa6+9RmxsLDKZjOnpaaKjo3njjTcoLS1l1apVaLVaJ4eCn58fL7zwArt27SIwMFCcL5504Xvj
xg2qq6uJi4vjF7/4BevWrRMX+XNzc4SGhvKHP/yBa9eusW3bNpKTk50MBalUSk5ODj/72c9IS0tD
qVSK5zY2NsaxY8eoqqoiIyODX/7yl6xduxYvLy8xQcibb77J+++/z/T0tPibVquV69evc+zYMe7d
u8ff//3f8+qrrxIdHY1UKmVqaorw8HD+53/+h3PnzlFQUEBkZCQ5OTlkZ2cTFBTEhQsX6OvrY82a
NWzfvl0c2y5DwYWLvy1LvkHsdjsSieSRulKJRILD4RAD9ODBwjo6OppXXnmFpqYmampqOHToEBqN
htOnTxMVFcXOnTud0vYJ2Uce1d7CYwvbexIWnvvj6GeFzwsYjUbq6+s5deoUSqWS7du3s3x59g9+
0rLb7V/osRfu5+f7VCqVkpKSwu7du+no6OD27dscO3YMs9nM1atX0el0Yi2Npdp71PiBxffwSfj8
2H+czy/8rEqlIj4+ntWrV+Pp6UlBQQGJiYkoFAr6+/spLS2ltLSUgwcPEhoays9//vPvZMrOx0Wl
UhEVFUVYWBhNTU10dXVhMhpRqlRitpPIiAiUKhUjIyO0trYyPT2NWq1m9N499Ho93t7exMXFLdlP
RqORnp4e6urqgL8G6I6OjnLt2jVOnjyJu7s7xcXFpKSkAA9yt1+/fp3u7m7y8vLY9Jl3X1hUeXh4
sHnzZnH3tLW1lYmJCQIDA0lNTeUnP/kJqampFBQUOO1SqFQqcnNzOXv2LLdu3RJjIoaGhqisrGR0
dJTi4mKKi4uJiYkRx29iYiJbt26ls7NTlGKYTCaxFoFCoXjwbAFymeyx8+lrNBrS09MpKyujqamJ
o

commit b7c284ac (2024-04-22): fix #59