tobiasfaust/SolaxModbusGateway #63: 404  not found

lattemacchiato, 2024-04-26:
Hallo,
möglicherweise mache ich beim Kompilieren was falsch, aber egal ob ich die bins lokal oder via Gitpod baue, nach dem Flashen bekomme ich beim Aufruf das GWs nur ein "404 not found".
littlefs haben ich mit gebaut und geflashed....

tobiasfaust, 2024-04-26:
Wenn du einen 404 bekommst ist die littlefs.bin nicht oder nicht richtig geflashed. Der ESP findet keine Web-files
Ich habe auch eben im Wiki die flashadressen noch angepasst. Aufgrund der neuen Partitionen haben sich diese gändert

https://github.com/tobiasfaust/SolaxModbusGateway/wiki/build-with-gitpod

lattemacchiato, 2024-04-26:
Perfekt, das wars - die neue flash address für die littlefs hatte ich nicht mitbekommen, sorry.
Dankeschön!

closed 2024-04-26 by lattemacchiato