tobiasfaust/SolaxModbusGateway #65: Add support for board

McMacaron, 2024-05-15:
Hello Tobias,
would it be possible to add support for this board?
https://github.com/LaskaKit/ESPlan/tree/main
I think it is all in one solution for your project.

tobiasfaust, 2024-05-16:
Hi,
it should still work well. Just define the right pins for Modbus Pins according pinout documentatation of ESPLan. https://github.com/LaskaKit/ESPlan/blob/main/HW/ESPlan_v1_1.pdf
https://www.studiopieters.nl/esp32-pinout/

It is: 
Modbus-RX-Pin = RX1 = Sensor_VP = GPIO36 = D36
Modbus-TX Pin = TX1 = GPIO4 = D4
https://github.com/tobiasfaust/SolaxModbusGateway/wiki/configuration-modbus

McMacaron, 2024-06-05:
Hello Tobias,

I have spend whole weekend just to put libraries for your ESP32 devices. But I was not able to do build it. I have spent about 16hours and still nothing. My idea was to build it in your way and after working model to do it in my way.

There so many missing dependencies that I am not able to do it. And it is confusing because looking randomly on the internet trying libraries which could work is not way how to do it. So what i supposed to do? I can help you to make it compacta and useful for others but what should I do?

S pozdravem,

Robert Košulič

mob. +420 737 685 211

https://www.studiopieters.nl/esp32-pinout/

https://github.com/tobiasfaust/SolaxModbusGateway/wiki/configuration-modbus

tobiasfaust, 2024-06-06:
Try this documentation: 
https://github.com/tobiasfaust/SolaxModbusGateway/wiki/build-with-gitpod

closed 2024-09-19 by tobiasfaust