tobiasfaust/SolaxModbusGateway #66: Solax X3 MIC zu Cerbo GX hinzufügen

Michax1979, 2024-06-05:
Hi Tobias,

versuche das o.g. Gerät in Cerbo einzufügen. Dein Solax X1 Skript hat funktioniert, zeigt halt nur die gesamte Leistung auf eine Phase an.

Hast du eine Idee?

Gtrüße Michael

tobiasfaust, 2024-06-05:
Hi,
Das ist auch klar, da X1 nur einphasig ist. Du musst die X3 Datei benutzen.

Michax1979, 2024-06-05:
Ok.. danke. Wo finde ich diese?

tobiasfaust, 2024-06-05:
Probier man die „Solax-X3-Pro“ Einstellung ob diese schon ausreicht. 

Andernfalls musst dir eine neue json Datei erstellen, ggf auf Basis schon vorhandener. Anleitung dazu steht im Wiki.

Michax1979, 2024-06-05:
Entschuldige... bin da was blöde. Kannst du mir bitte links schicken dazu...🙈

tobiasfaust, 2024-06-05:
https://github.com/tobiasfaust/SolaxModbusGateway/wiki

tobiasfaust, 2024-07-08:
Wenn du es geschafft hast, wäre es toll, wenn du bei mir im Wiki ein howto erstellen könntest, wie du vorgegangen bist, dieses in die Victron Welt zu integrieren, dann können andere auch davon profitieren. 
alternativ geht auch eine Textdatei und Bilder dazu.