tobiasfaust/SolaxModbusGateway #69: ESP Webinterface nicht mehr erreichbar, teils erst nach 1-2 Wochen

crazy-ahmet, 2024-07-09:
Betreibe das Teil an einem Deye Inverter. Signalqualität über 90% bzw. -53dB
Habe das merkwürdige Problem, dass nach 1-2 Wochen irgendwann keine Daten mehr per MQTT kommen.
Versucht man aufs Webinterface per IP Adresse zuzugreifen bekommt man keine Verbindung.

Man könnte meinen das Teil hat sich irgendwie aufgehangen, das merkwürdige ist aber, dass es in meinem Router bzw. Unifi System noch als online angezeigt wird. Es wird auch angezeigt dass da 50-60kBit/s Daten drüber gehen nur wohin? Auf dem MQTT Server kommt nichts an und aufs Webinterface kommt man ja auch nicht mehr. Anpingen funktioniert ebenso nicht. Habe das Teil dann aus dem Netzwerk geschmissen bzw. übers Unifi System gesagt es soll sich neu verbinden. Es war dann offline und hat sich sogar neu verbunden, hat wieder die gleiche IP laut Anzeige nur nach wie vor nicht erreichbar. Es hilft nur ein Power Reset. Hat jemand schon ähnliches beobachtet?

MagicSven81, 2024-07-12:
Ich konnte das Problem mit einer FritzBox reproduzieren, bei welcher das 2,4Ghz & 5Ghz Netzwerk die selbe SSID hatte.
Die ESP32 Module können allerdings nur über das 2,4Ghz arbeiten.
Versucht es nun auf das 5Ghz Netz zu connecten, bricht die Verbindung ab und das Gerät ist nicht mehr erreichbar.

Abhilfe war hier, das 5Ghz Netzwerk anders zu benennen.

crazy-ahmet, 2024-07-13:
Das ist es bei mir aber nicht. Es hängt definitiv im 2,4GHz WLAN. Habe extra ein IOT WLAN nur unter 2,4GHz eingerichtet auf das solche Geräte connecten. Ich sehe ja dass es noch online ist und dann 50-60KBit/s Daten drüber laufen lt. meinem Unifi System. Wenn ich es rauswerfe verbindet es sich sogar erneut und ist trotzdem nicht erreichbar. Jetzt wo es gerade normal arbeitet sind es nur 3-4 KBit/s Daten also deutlich weniger, ist auch merkwürdig. Das läuft meist schon so 1-2 Wochen durch bis es auftritt.

crazy-ahmet, 2024-08-14:
So das Teil lief jetzt fast eine Monat wieder ohne Probleme. Nun wieder nix mehr. Webseite nicht erreichbar und liefert auch keine Daten mehr per MQTT. Lt. meinem Unifi System ist das Teil aber nach wie vor unter der IP online. Macht wieder zwischen 50-60KBit/s Traffic. Was wohin hab ich keine Ahnung. Im Normalbetrieb sinds ja nur 3-4KBit/s. Läuft das bei allen anderen ewig durch? Habe viele Geräte bei mir im Netz die das tun nur dieses macht Mucken.

tobiasfaust, 2024-08-14:
Meiner lief schon mal ein halbes Jahr durch. Da ich aber öfters das neuste Update drauf flashe komme ich meistens nicht über 1-2 Monate durchlauf.
Ich kann den Fehler überhaupt nicht nachvollziehen. 
Du schreibst oben das auch ein anpingen nicht funktioniert. Falls dem so ist bezweifle ich das ein neuverbinden über die UniFi Konsole möglich ist.

hast du schon mal den esp gewechselt? 
du kannst auch mal auf LogLevel 4 wechseln. Dann wird per mqtt alle 10sek der freie Speicher übermittelt.

crazy-ahmet, 2024-08-14:
Habe gerade nochmal per Unifi neu verbinden angestoßen. Jetzt läuft er sogar wieder. Das hat vorher noch nicht funktioniert. Zeigt an dass er seit 34 Tagen online ist. Die Platine wurde nicht angefasst oder resettet, trotzdem läuft er erstmal wieder und liefert auch mqtt Daten. Loglevel 4 habe ich eingestellt aber wo sehe ich das? Auf der Statusseite is nix und per MQTT wird auch nix neues miteliefert.

MagicSven81, 2024-08-14:
Also meiner läuft seit dem 03.12.2023 Problemlos durch. Da habe ich zuletzt die Firmware aktualisiert ;-)

crazy-ahmet, 2024-08-14:
Habe es jetzt gefunden und logge das memory mal mit. Übermittelt gerade so 195660 +/- ein bissi was

crazy-ahmet, 2024-09-04:
Seit einem Tag läuft es schon wieder nicht mehr. Letzte freie Speichermeldung 193904. Hat immer so in dem Bereich geschwankt, auch vor dem Ausfall nichts auffälliges.

tobiasfaust, 2024-09-04:
da wird dir leider nichts anderes übrigbleiben als die echte serielle ausgabe mitzuloggen. Ich habe zwar gestern eine Webserial version eingecheckt, die hilft aber nicht viel wenn das wifi abreisst.

Entweder du hängst die ganze Zeit ein Laptop mit serial monitor dran, oder du müsstet über eine transparent bridge über einen ESP8266 die serielle Ausgabe über wifi zugänglich machen. 
zb. das hier: https://github.com/jeelabs/esp-link

tobiasfaust, 2024-09-08:
hab das wiki um das "Advanced Logging" erweitert

https://github.com/tobiasfaust/SolaxModbusGateway/wiki/advanced-logging

Lazyjack1031, 2024-09-09:
> Seit einem Tag läuft es schon wieder nicht mehr. Letzte freie Speichermeldung 193904. Hat immer so in dem Bereich geschwankt, auch vor dem Ausfall nichts auffälliges.

Hallo, mich würde interessieren, ob du in deinem WLAN Passwort ein Sonderzeichen hast? 
Dies war bei mir der Grund, warum der ESP nach einer Zeit nicht mehr erreichbar war. Er war nicht abgestürzt, sondern nur nicht mehr in der Lage HTTP Requests zu beantworten. Manchmal ging es dann wieder (~nach einem Tag) und man konnte anhand der fortgeschrittenen Zeit sehen, dass er weitergelaufen ist. Auch erhöhte sich der Stromverbrauch leicht (mit Shelly PlugS gemessen 0,84 Watt auf 1,2 Watt), wenn er nicht erreichbar war.     
Viele Grüße Lazyjack

tobiasfaust, 2024-09-29:
Hi,
Mit meiner Pumpcontrol konnte ich den Fehler nachstellen, da scheint es ein Problem zu geben im Falle eines schwachen wlan Signals.
Ich werde schauen ob ich nächste Woche ein fix hinbekommen kann.

tobiasfaust, 2024-10-06:
problem should be solved by 38960dc3e3bfd6eb67bcc70e998b3b101de27f61

Thoralfo, 2024-10-12:
Hallo, woher weiß ich welche ESP-Version ich bei dem Adapter habe, um ein Firmware Update zu machen ?

tobiasfaust, 2024-10-13:
> Hallo, woher weiß ich welche ESP-Version ich bei dem Adapter habe, um ein Firmware Update zu machen ?

Steht auf dem ESP Chip direkt drauf.

closed 2024-11-26 by tobiasfaust