tobiasfaust/SolaxModbusGateway #70: Growatt Wechselrichter Modbus Adressen 1000 und 3000

Gustavw82, 2024-07-14:
Hallo. 
Ich habe einf Frage. Ich habe das Projekt jetztcerfolgreich an meinem Wechselrichter (Growatt mod Sph) laufen. Ich habe jetzt seit einiger Zeit einen Akku am Wechselrichter. Die Werte davon würde ich gern auch abfragen.
Leider liegen die Modbus adressen im Bereich größer 1000 bis über 3000.
Wenn ich die Adressen versuche im Testmenü abzurufen bekomme ich nur 0, oder einen Fehler. Ich vermute mal, ich muß dazu in der Configdatei für den Wechselrichter irgendwas anpassen. 
Ich hoffe, es kann mir jemand helfen.

tobiasfaust, 2024-07-14:
Ja, du musst die config Datei erweitern.
Im Wiki gibt es eine Anleitung dazu.

sobald sie fertig ist würde ich mich freuen wenn du die komplette Datei hier postest damit andere diese auch benutzen können bzw ich sie ins Repo einchecken kann

Gustavw82, 2024-07-14:
Ich verstehe es leider nicht. Ich bekomme wie gesagt nir 0 oder einen Fehler. Was haben die Einträge bei Request Live Data in der  growatt-sph.json zu bedeuten?

tobiasfaust, 2024-07-14:
Schau dir mal hier das Attribut „RequestLiveData“ an . Das musst du mit einem weiteren request , nämlich ab Position 1000 ( oder ab der startposittion die du willst bei der Batteriewerte stehen) erweitern. Du kannst immer maximal 0x77 Register mit einer query abfragen.

https://github.com/tobiasfaust/SolaxModbusGateway/wiki/configuration-register

Gustavw82, 2024-07-18:
Hallo. Also momentan funktioniert es erstmal bei mir. In dem Bereich ab 1000 werden keine Daten ausgegeben. 
[image]

tobiasfaust, 2024-07-18:
Was heißt denn das es erstmal funktioniert?
Hast du das JSON angepasst? Wenn ja, kannst du die Datei hier posten wenn du final fertig bist?

Gustavw82, 2024-07-18:
Ich habe mich jetzt näher damit beschäftigt. Man muß die Werte, welche man anzeigen lassen will immer relativ zu dem vorher angegebenen Bereich abfragen. Wenn ich mit dem ersten request von 0-240 einlese, und dem zweiten request ab 3100 dann lautet die richtige Position zum beispiel 3171(State of charge) 388. Das ergibt sich in dem man 71+3 für den header rechnet, dann mal 2, weil es immer 2 stellen sind, und dann plus 240 für die Daten aus dem vorherigen request.
[image]
[image]
[image]

Gustavw82, 2024-07-18:
Ich habe mich jetzt näher damit beschäftigt. Man muß die Werte, welche man anzeigen lassen will immer relativ zu dem vorher angegebenen Bereich abfragen. Wenn ich mit dem ersten request von 0-240 einlese, und dem zweiten request ab 3100 dann lautet die richtige Position zum beispiel 3171(State of charge) 388. Das ergibt sich in dem man 71+3 für den header rechnet, dann mal 2, weil es immer 2 stellen sind, und dann plus 240 für die Daten aus dem vorherigen request.

Gustavw82, 2024-07-18:
[Growatt-SPH.txt](https://github.com/user-attachments/files/16279695/Growatt-SPH.txt)

Gustavw82, 2024-07-18:
Die Datei müßte noch in .json umbenannt werden. Ich versuche die nächsten Tage noch mehr Werte einzubauen.

tobiasfaust, 2024-07-18:
Hast du Einträge angepasst für dich oder nur neue hinzugefügt?
Ich frage deshalb ob ich einen neuen WR-Eintrag machen muss oder die bestehende growatt ersetzen kann.

Gustavw82, 2024-07-18:
Ein Problem ist noch. Die Seriennummer aus den IDDaten wird nicht korrekt gelesen. Es steht nur unknown. Unter Raw Data steht sie aber korrekt drin.

Gustavw82, 2024-07-18:
> Hast du Einträge angepasst für dich oder nur neue hinzugefügt? Ich frage deshalb ob ich einen neuen WR-Eintrag machen muss oder die bestehende growatt ersetzen kann.

Ich habe Einträge hinzugefügt, und manche auch angepasst. Für meinen Wechselrichter 
Passt es so. Ob Das für andere auch so ist kann ich nicht sagen.
[image]

tobiasfaust, 2024-07-18:
Alles klar, meld dich wenn du fertig bist. Ich würde dann eine neue Sektion für deinen WR anlegen

Gustavw82, 2024-07-18:
So, Ich bin jetzt erstmal soweit fertig. Ein par Fehlermeldungen konnte ich nicht vernünftig übersetzen, deshalb in englisch. Das einzige, was noch offen wäre, wäre die Anzeige der Seriennummer. Das bekomme ich aber wahrscheinlich nicht selbst hin. Meines erachtens nach ist in der Json-datei alles okay.
[Growatt-SPH.json](https://github.com/user-attachments/files/16289444/Growatt-SPH.json)

tobiasfaust, 2024-07-19:
Irgendwie passt dein Screenshot oben mit den ID Data nicht zum json. 
im json steht das du 20 Register abfragst (0x14), damit hätte ich 40 ID Einträge in der rawData Seite erwartet. Es sind aber über 100 ;)

Damit ich mir das ansehen kann, kannst du bitte mal  sowohl die Werte von ID-Data als auch LiveData hier als komplette lange Strings posten? Plus bitte den dazugehörigen RawData Screenshot

Gustavw82, 2024-07-19:
Das kann durchaus sein. Ich hatte einiges probiert.
Hier die richtigen Daten:
IDData:
0x01 0x03 0xEE 0x00 0x01 0x00 0xBD 0xFF 0xFF 0x00
0x64 0x00 0x00 0x27 0x10 0x00 0x00 0x9C 0x40 0x06
0x40 0x44 0x4E 0x31 0x2E 0x30 0x00 0x5A 0x42 0x44
0x42 0x00 0x02 0x00 0x02 0x00 0x00 0x06 0x40 0x00
0x3C 0x00 0x3C 0x00 0x5A 0x00 0x5A 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x01 0x00 0x64 0x00 0x00 0x00
0x00 0x20 0x20 0x20 0x50 0x56 0x20 0x49 0x6E 0x76
0x65 0x72 0x74 0x65 0x72 0x20 0x20 0x00 0x00 0x15
0x18 0x02 0x03 0x07 0xE8 0x00 0x07 0x00 0x13 0x00
0x08 0x00 0x10 0x00 0x05 0x00 0x05 0x0C 0x73 0x13
0x73 0x12 0x8E 0x14 0x1E 0x07 0x00 0x13 0x73 0x12
0x8E 0x14 0x1E 0x07 0x00 0x13 0x73 0x12 0x8E 0x14
0x1E 0x0D 0x61 0x10 0xF6 0x12 0x9D 0x13 0x8D 0x00
0x98 0x00 0x32 0x00 0x32 0x00 0x32 0x00 0x32 0x00
0x32 0x00 0x32 0x00 0x32 0x00 0x32 0x00 0x32 0x00
0x32 0x00 0x32 0x11 0x1E 0x00 0x00 0x44 0x4E 0x41
0x41 0x30 0x31 0x35 0x31 0x30 0x30 0x30 0x32 0x01
0x31 0x00 0x00 0x00 0x00 0x13 0x9C 0x00 0x32 0x10
0x07 0x10 0xA6 0x0F 0x18 0x0E 0x79 0x00 0x14 0x00
0x05 0x10 0x57 0x0F 0x90 0x26 0xFD 0x26 0xFD 0x26
0xA5 0x27 0xC1 0x27 0xC1 0x28 0x21 0x00 0x0A 0x00
0x00 0x01 0xE4 0x00 0xFF 0x4E 0x20 0x00 0xFF 0x4E
0x20 0x00 0xFF 0x4E 0x20 0x00 0xFF 0x4E 0x20 0x07
0x00 0xFC 0xE5

LiveData:
00x01 0x04 0xEE 0x00 0x01 0x00 0x00 0x32 0x8F 0x07
0xF5 0x00 0x00 0x00 0x00 0x00 0x00 0x0B 0xC2 0x00
0x2B 0x00 0x00 0x32 0x8F 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x05 0xED 0x13 0x86 0x09
0x5B 0x00 0x06 0x00 0x00 0x05 0x9D 0x09 0x5C 0x00
0x06 0x00 0x00 0x05 0x9D 0x09 0x4D 0x00 0x06 0x00
0x00 0x05 0x94 0x10 0x54 0x10 0x1B 0x10 0x1B 0x00
0x00 0x00 0x03 0x00 0x00 0x5F 0x9F 0x01 0xA8 0xF8
0x7A 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x0B 0x00 0x00 0x67 0x99 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x67 0x99 0x01
0xBD 0x01 0x63 0x01 0x54 0x00 0x00 0x01 0x79 0x0C
0xD4 0x0C 0xCA 0x4E 0x20 0x00 0x00 0x00 0x00 0x9C
0x40 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x03 0x00 0x3C 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x07 0xFA 0x01 0x04 0xEE 0x4E 0x20 0x00 0x03
0x00 0x00 0x9C 0x40 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x3C 0x00 0x00
0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00
0x0A 0x25 0x00 0x00 0x00 0x05 0x00 0x00 0x00 0x02
0x00 0x00 0x20 0x19 0x00 0x00 0x00 0x08 0x00 0x00
0x21 0xD5 0x00 0x00 0x00 0x02 0x00 0x00 0x00 0x51
0x00 0x00 0x66 0xD9 0x00 0x00 0x00 0x05 0x00 0x00
0x50 0x16 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x27 0x10
0x00 0x00 0x19 0xA0 0x00 0x00 0x00 0x00 0x01 0x01
0x00 0x00 0x00 0x00 0x53 0x00 0x00 0x30 0x00 0x12
0x19 0x9E 0x0D 0x4A 0x00 0x15 0x00 0x15 0x01 0x61
[... 32 lines cut ...]
0x00 0x07 0xFA 0x01 0x04 0xEE 0x4E 0x20 0x00 0x03
0x00 0x00 0x9C 0x40 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x3C 0x00 0x00
0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00
0x0A 0x25 0x00 0x00 0x00 0x05 0x00 0x00 0x00 0x02
0x00 0x00 0x20 0x19 0x00 0x00 0x00 0x08 0x00 0x00
0x21 0xD5 0x00 0x00 0x00 0x02 0x00 0x00 0x00 0x51
0x00 0x00 0x66 0xD9 0x00 0x00 0x00 0x05 0x00 0x00
0x50 0x16 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x27 0x10
0x00 0x00 0x19 0xA0 0x00 0x00 0x00 0x00 0x01 0x01
0x00 0x00 0x00 0x00 0x53 0x00 0x00 0x30 0x00 0x12
0x19 0x9E 0x0D 0x4A 0x00 0x15 0x00 0x15 0x01 0x61
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x28 0x14
0x00 0x00 0x20 0x19 0x00 0x00 0x21 0xD5 0x00 0x00
0x00 0x03 0x0C 0x54 0x00 0x39 0x00 0x10 0x00 0x00
0x01 0x01 0x00 0xE1 0x00 0x02 0x00 0x10 0x00 0x00
0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x31 0x00 0x31
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x12 0x52 0x94
0x01 0xCC 0x01 0xA5 0xD1 0x6F

Gustavw82, 2024-07-22:
> Irgendwie passt dein Screenshot oben mit den ID Data nicht zum json. im json steht das du 20 Register abfragst (0x14), damit hätte ich 40 ID Einträge in der rawData Seite erwartet. Es sind aber über [...]

Benötigst Du noch weitere Informationen?

Gustavw82, 2024-07-23:
Hallo, und Guten Morgen. 
Es gibt neue Erkenntnisse. 
Die Seriennummer steht bei meinem Wechselrichter ab der position 209. Sie ist 10 Stellen lang. Ich lese die IdDaten ab 0x78. Die richtige Stelle ist dann 181, bis 190.
Was mir aufgefallen ist, wenn man bei den IDDaten mehrere Bereiche angibt, so wie bei den Livedaten, dann wird unter Rawdata nichts mehr angezeigt. Nur mal so als info.
Ich hänge mal meine aktuelle Json Datei hier an.

{
  "Growatt-SPH": {
    "config": {
      "RequestLiveData": [
        [
          "#ClientID",
          "0x04",
          "0x00",
          "0x00",
          "0x00",
          "0x77"
        ],
        [
          "#ClientID",
          "0x04",
          "0x0C",
          "0x1C",
          "0x00",
          "0x77"
        ]
      ],
      "RequestIdData": [
        "#ClientID",
        "0x03",
        "0x00",
        "0x78",
        "0x00",
        "0x77"
      ],
      "ClientIdPos": 0,
      "LiveDataFunctionCodePos": 1,
      "LiveDataFunctionCode": "0x04",
      "IdDataFunctionCodePos": 1,
      "IdDataFunctionCode": "0x03",
      "LiveDataStartsAtPos": 3,
      "IdDataStartsAtPos": 3,
      "LiveDataErrorPos": 1,
      "LiveDataErrorCode": "0x84",
      "IdDataErrorPos": 1,
      "IdDataErrorCode": "0x83",
      "LiveDataSuccessPos": 1,
      "LiveDataSuccessCode": "0x04",
      "IdDataSuccessPos": 1,
      "IdDataSuccessCode": "0x03"
    },
    "data": {
      "livedata": [
        {
          "position": [
            3,
            4
          ],
          "name": "InverterRunState",
          "realname": "Inverter Status",
          "datatype": "integer",
          "mapping": [
            [
              0,
              "WaitMode"
            ],
            [
              1,
              "NormalMode"
            ],
[... 626 lines cut ...]
          "unit": "KWh"
        }
      ],
      "id": [
        {
          "position": [
            181,
            182,
            183,
            184,
            185,
            186,
            187,
            188,
            189,
            190
          ],
          "name": "InverterSN",
          "realname": "Inverter SerialNumber",
          "datatype": "string"
        }
      ]
    }
  }
}

closed 2024-08-10 by tobiasfaust