tobiasfaust/SolaxModbusGateway #71: Not able to get data from foxess kh8 inverter

iqbaal77, 2024-08-18:
I am beginner trying to get data from foxess kh8 inverter but getting very long data in decimal or HEX using below code, can anyone help me to parse it or get correct readable form or this response is correct or not or how I can get correct response?

rs485 to max485 to esp-wroom-32 ch340 kit
inverter modbus address: 247

I connected:
max485 -> esp32
RO -> RX
(DE + RE common)  -> 5 pin
DI -> TX
GND -> GND

My code: 
#include <Arduino.h>

#define RX 16      // UART2 RX
#define TX 17      // UART2 TX
#define RTS_pin 5  // RS485 Direction control
#define RS485Transmit HIGH
#define RS485Receive LOW

void setup() {
  pinMode(RTS_pin, OUTPUT);
  digitalWrite(RTS_pin, RS485Receive);

  // Start the Modbus serial Port
  Serial2.begin(9600, SERIAL_8N1, RX, TX);
  Serial.begin(9600);  // Ensure Serial Monitor is set to 9600
  delay(1000);

  Serial.println("Test RS485 Connection");

  while (Serial2.available()) {
    Serial.print(Serial2.read(), HEX);
    Serial.print(" ");
  }

  delay(1000);
}

uint16_t Calc_CRC(uint8_t* message, uint8_t len) {
  uint16_t crc = 0xFFFF;
  for (int pos = 0; pos < len; pos++) {
    crc ^= (uint16_t)message[pos];
    for (int i = 8; i != 0; i--) {
      if ((crc & 0x0001) != 0) {
        crc >>= 1;
        crc ^= 0xA001;
      } else {
        crc >>= 1;
      }
    }
  }
  return crc;
}

String Hex2String(uint8_t num) {
  char hexCar[2];
  sprintf(hexCar, "0x%02X", num);
  return hexCar;
}

void loop() {
  byte request[] = { 0xF7, 0x03, 0x00, 0x00, 0x00, 0x07, 0x00, 0x00 };

  uint16_t crc = Calc_CRC(request, sizeof(request) - 2);
  request[sizeof(request) - 1] = highByte(crc);
  request[sizeof(request) - 2] = lowByte(crc);

  Serial.println("Schreibe Daten ....");
  for (uint8_t i = 0; i < sizeof(request); i++) {
    Serial.print(Hex2String(request[i]));
    Serial.print(' ');
  }
  Serial.println("");

  digitalWrite(RTS_pin, RS485Transmit);
  Serial2.write(request, sizeof(request));
  Serial2.flush();
  digitalWrite(RTS_pin, RS485Receive);

  delay(100);
  Serial.println("Lese Daten ....");

  while (Serial2.available()) {
    Serial.print(Serial2.read(), HEX);
    Serial.print(" ");
  }
  Serial.println("");
  delay(2000);
}

Serial2.read response:
Schreibe Daten ....
0xF7 0x03 0x00 0x00 0x00 0x07 0x10 0x9E 
Lese Daten ....
F7 3 0 0 0 7 10 9E 
or

Schreibe Daten ....
0xF7 0x03 0x00 0x00 0x00 0x07 0x10 0x9E 
Lese Daten ....
0 0 0 0 6 0 1 1 C0 E0 E0 60 E1 60 D0 10 70 50 40 40 E0 F1 60 F0 70 60 10 0 70 20 30 28 18 18 18 C C 10 1A 1C E 4 5 2 4 80 E0 C0 60 C0 F0 18 38 18 C E 2 6 4 3 1 E0 C0 E0 54 B0 30 20 70 58 18 1C 18 1C A 2 E E 3 2 C4 B8 0 C0 C0 60 B0 30 78 B0 28 1A C F 60 E0 28 30 18 58 B 1C 8 C 2 B 6 6 E E 7 C0 E0 60 30 38 10 30 38 18 19 8 18 18 A C0 60 30 30 78 98 18 8 4 6 2 E 6 6 7 6 6 42 C0 30 70 20 18 18 C 6 4 3 C0 E0 60 40 20 30 16 0 4C 6 6 7 7 80 C0 60 60 60 30 32 70 90 10 70 70 38 30 18 14 4C 2 2 7 1 1 C0 C0 E0 D0 70 38 0 C 4 7 87 3 1 44 E0 A8 60 70 31 78 30 8 8 A 7 3 1 C0 A0 E0 74 B8 18 4 48 98 C C E 7 3 80 E0 50 D0 40 60 30 70 38 20 30 28 38 38 30 18 10 18 14 1C 6 3 C0 C0 F0 61 10 30 38 30 78 E4 30 B0 F4 E0 E0 E0 F0 F7 E5 F0 A0 F8 E0 E1 E0 F0 E8 F0 F0 F8 78 78 F8 E8 60 F0 B0 E1 E0 F4 F0 F4 F8 34 70 FA 78 7C FE 58 7A FC 7E 78 70 FA F0 7C FC F4 78 F2 F9 E8 FB 9C 7F BE 1C 7E 3F FF 3F 9E 3E C7 BD 7F FF FF FF 9F FB 7E FF BF FF 7F FF DF FF FF FF FF FF BF FF FF FF 7F FF FF FF FF FF FF FF FF BF FF 0 3 0 0 0 0 0 0 6 2 0 0 2 0 0 0 0

iqbaal77, 2024-08-20:
@tobiasfaust did you get chance to look fix of my inverter?

tobiasfaust, 2024-08-20:
It looks a little bit strange because the first 2 hex of answer should be the same of first 2 hex of request.
F7 03 ….
But at first, it’s good to get an answer :)

tobiasfaust, 2024-08-20:
Ok, can you please try function code 0x04? 
I couldn’t find function code 0x03 in your linked document.

0xF7, 0x04, 0x00, 0x00, 0x00, 0x07, 0x00, 0x00

iqbaal77, 2024-08-20:
Schreibe Daten ....
0xF7 0x03 0x00 0x00 0x00 0x07 0x10 0x9E 
Lese Daten ....
20 80 0 18 F4 FF 
Schreibe Daten ....
0xF7 0x03 0x00 0x00 0x00 0x07 0x10 0x9E 
Lese Daten ....
7 0 0 0 7 98 FF 
Schreibe Daten ....
0xF7 0x03 0x00 0x00 0x00 0x07 0x10 0x9E 
Lese Daten ....
30 80 0 10 E4 FF 
Schreibe Daten ....
0xF7 0x03 0x00 0x00 0x00 0x07 0x10 0x9E 
Lese Daten ....
20 60 80 F 10 DF 

Got this output.

tobiasfaust, 2024-08-20:
As suggested, please exchange 0x03 to 0x04 at position 2
Your current output is random stuff.

iqbaal77, 2024-08-20:
Schreibe Daten ....
0xF7 0x04 0x00 0x00 0x00 0x07 0xA5 0x5E 
Lese Daten ....
84 80 0 0 F A5 FF 
Schreibe Daten ....
0xF7 0x04 0x00 0x00 0x00 0x07 0xA5 0x5E 
Lese Daten ....
6 80 0 0 F9 DF 
Schreibe Daten ....
0xF7 0x04 0x00 0x00 0x00 0x07 0xA5 0x5E 
Lese Daten ....
86 80 80 0 E8 FF 
Schreibe Daten ....
0xF7 0x04 0x00 0x00 0x00 0x07 0xA5 0x5E 
Lese Daten ....
86 0 80 80 D8 BE

tobiasfaust, 2024-08-21:
Look into your Modbus docu ;) the read register starts at 11000, that’s 2AF8

So let’s try with that request:

0xF7, 0x04, 0x2A, 0xF8, 0x00, 0x07, 0x00, 0x00

iqbaal77, 2024-08-21:
Sure I will try and update, Thanks for being with me :-)

iqbaal77, 2024-08-24:
Response:
0xF7 0x04 0x1E 0x0A 0x3E 0x00 0x39 0x05 0xE1 0x0A 0x68 0x00 0x39 0x05 0xFA 0x00 0x08 0x00 0x00 0x00 0x00 0x00 0x00 0x08 0xFC 0x00 0x7C 0x00 0x00 0x13 0x87 0x00 0x00 0x24 0x27
Received CRC: 0x2724
Calculated CRC: 0x2724
CRC check passed.

I got this response, Do you know how to convert it into readable values, because I can't match it with kh series registers manual ?

tobiasfaust, 2024-08-24:
That looks  very well!
What was your exact requested string?

iqbaal77, 2024-08-24:
> That looks very well! What was your exact request?

0xF7 0x04 0x79 0x18 0x00 0x0F 0x3D 0xC3

tobiasfaust, 2024-08-24:
Hi,
attached basic json file, has to be extend with all items by you.
Please look at this wiki page to do that: https://github.com/tobiasfaust/SolaxModbusGateway/wiki/adding-new-inverters-or-items

[Foxess-KH8.json](https://github.com/user-attachments/files/16736621/Foxess-KH8.json)

currently, id section cannot be used, i need apply a patch #58 at first of @TigerGrey

tobiasfaust, 2024-08-26:
Can you please share your example program which produces the crc validation output? I need this for further investigation.

tobiasfaust, 2024-09-01:
Any news? I would like to add your final json file to the repository for public use.

closed 2024-10-16 by tobiasfaust