tobiasfaust/SolaxModbusGateway #73: Incorrect data periodically

den-localnet, 2024-08-24:
Hello,

deye sun-12k-sg04lp3-eu
ESP32 + HW-97

Every time I get data one or more fields have incorrect values.
[image]
[image]
[image]

I tried changing Baudrate from 9600 to 115200 - it does not help to solve the problem.

Please tell me what else I can try.

tobiasfaust, 2024-08-25:
Hi,
Please Check #43 if it fits to your problem. (German language)
Please use the BMS Port and not the COM Port of your Deye

tobiasfaust, 2024-08-26:
Hi,
Please keep in mind to keep Modbus id’s unique. It’s an often doing problem.

1. can you please share your example program which produces the crc validation output? I need this for further investigation.

2. please share raw hex outputs of correct and failure values for further investigation.

tobiasfaust, 2024-09-01:
Any news with the latest repository update? I added a CRC check

den-localnet, 2024-09-01:
> Any news with the latest repository update? I added a CRC check

Hello
After update - no data(
Debug log: https://pastebin.com/r20prEmP

den-localnet, 2024-09-01:
rollback to 3.2.1 - work but randomly incorrect data (line 225, 2765, 3046, 3071, 3861, 3873)
debug log: https://pastebin.com/h5TWmmwh

thank you

tobiasfaust, 2024-09-01:
Hi,
you can flash the newest firmware from repository. The problem is, the CRC Check is failing continously. Please disable the CRC Check at page Modbus Settings. (I dont know why your CRC is different)
<pre>Received CRC: 0x9D 0x5B
Calculated CRC: 0xCC 0xEC
CRC check  failed!</pre>
Please check also your log of 3.2.1. and look for "Dataframe valid, Dateframe size:". You will see, mostly the final size is 378 bytes, but sometimes, the size is different, in your example only 193 Bytes. Thats raises your wierd values. 
That behaviour of Deye is well known. The Deye COM port is very buggy and unstable. Please use the BMS Port instead.  Maybe you have to setup a different modbus ID because 01 is still in use by your battery BMS.

If the data via BMS Port will deliver, please try to enable CRC Check again and check the log.

den-localnet, 2024-09-01:
I flash latest firmware, upload data directory and disable CRC Check.
Now work as 3.2.1

Can you add minimal Dateframe size as parametr Modbus Config?
Dataframes smaller than the specified size will not be parsed and sent via mqtt.
I understand that this is a "crutch" and not a solution, but so far I can't use the BMS port.

I tried turning the inverter off and on, tried changing the modbus ID to 2, tried disconnecting the battery - still no data from the BMS port.

tobiasfaust, 2024-09-01:
Hi,
Fine…..
Your inverter sends the correct size at position 3. So it makes no sense to specify it itself.
Please check the log for length check failures ;) 

on the other hand, the bms ports works! Many other deye users have switched their modbus connection from com port to bms port. 
please keep in mind to select the right pins and don’t exchange A+ and A- ;)

den-localnet, 2024-09-01:
> Your inverter sends the correct size at position 3. So it makes no sense to specify it itself.
> Please check the log for length check failures ;)

If dataframe size 243/378 bytes - data correct
if 242/377 or another size - data incorrect

> please keep in mind to select the right pins and don’t exchange A+ and A- ;)
Pinout Modbus port and BMS port are same (as for RS485).

den-localnet, 2024-09-01:
on BMS port:

Query Live Data into Queue:
0x01 0x03 0x01 0xF4 0x00 0x77 0x02
0x01 0x03 0x02 0x6B 0x00 0x41 0x02
Request queue data to inverter: 0x01 0x03 0x01 0xF4 0x00 0x77 0x45 0xE2
Read Data from Queue:
no response from client
Request queue data to inverter: 0x01 0x03 0x02 0x6B 0x00 0x41 0xF5 0x9E
Read Data from Queue:
no response from client
Query Live Data into Queue:
0x01 0x03 0x01 0xF4 0x00 0x77 0x02
0x01 0x03 0x02 0x6B 0x00 0x41 0x02
Request queue data to inverter: 0x01 0x03 0x01 0xF4 0x00 0x77 0x45 0xE2
Read Data from Queue:
no response from client
Request queue data to inverter: 0x01 0x03 0x02 0x6B 0x00 0x41 0xF5 0x9E
Read Data from Queue:
no response from client

den-localnet, 2024-09-01:
if i change modbus id to 2
in BMS port - same as previous message
in modbus port:

Query Live Data into Queue:
0x01 0x03 0x01 0xF4 0x00 0x77 0x02
0x01 0x03 0x02 0x6B 0x00 0x41 0x02
Request queue data to inverter: 0x01 0x03 0x01 0xF4 0x00 0x77 0x45 0xE2
Read Data from Queue:
0x01 0x03 0xEE 0x00 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xA6 0x00 0x00 0x00 0x70 0x00 0x41 0x04 0x81 0x00 0x00 0x04 0x18 0x00 0x00 0x00 0x84 0x00 0x00 0x0B 0xEC 0x00 0x00 0x00 0x02 0x00 0x00 0x01 0x3A 0x16 0x28 0x00 0x00 0x01 0x0B 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x0C 0x94 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x04 0xE2 0x05 0xB1 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x0A 0x00 0x00 0x01 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x78 0xFF 0x72 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x02 0x00 0x00 0x00 0x00 0x1F 0x7C 0x00 0x2B 0x08 0x00 0x15 0x40 0x5D 0xC0 0x00 0x00 0x5D 0xC0 0x14 0x14 0x02 0xCA 0x01 0x65 0x02 0x81 0x00 0x02 0x00 0x00 0x00 0xFA 0x04 0xD8 0x14 0xA5 0x00 0x50 0x00 0x00 0x05 0x9B 0x0A 0x9C 0x01 0x90 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 0x00 0x00 0x00 0x01 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x83 0x30
Dataframe invalid
Query Live Data into Queue:
0x01 0x03 0x01 0xF4 0x00 0x77 0x02
0x01 0x03 0x02 0x6B 0x00 0x41 0x02
Request queue data to inverter: 0x01 0x03 0x01 0xF4 0x00 0x77 0x45 0xE2
Read Data from Queue:
0x01 0x03 0xEE 0x00 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xA6 0x00 0x00 0x00 0x70 0x00 0x41 0x04 0x81 0x00 0x00 0x04 0x18 0x00 0x00 0x00 0x84 0x00 0x00 0x0B 0xEC 0x00 0x00 0x00 0x02 0x00 0x00 0x01 0x3A 0x16 0x28 0x00 0x00 0x01 0x0B 0x00 0x00 0x00 0x20 0x00 0x00 0x00 0x00 0x0C 0x94 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x04 0xE2 0x05 0xB1 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x0A 0x00 0x00 0x01 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x78 0xFF 0x75 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x02 0x00 0x00 0x00 0x00 0x1F 0x80 0x00 0x2B 0x08 0x00 0x15 0x40 0x5D 0xC0 0x00 0x00 0x5D 0xC0 0x14 0x14 0x02 0xCA 0x01 0x65 0x02 0x81 0x00 0x02 0x00 0x00 0x00 0xFA 0x04 0xD8 0x14 0xA5 0x00 0x50 0x00 0x00 0x05 0x7A 0x0A 0x5D 0x01 0x90 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xC4 0x1B
Dataframe invalid

tobiasfaust, 2024-09-01:
Disable crc check, but enable length check

den-localnet, 2024-09-01:
> Disable crc check, but enable length check

[image]

Don't save Dataframe Length Check and disable reading inverter relay.

den-localnet, 2024-09-01:
How update code on gitpod?

den-localnet, 2024-09-01:
> Don't save Dataframe Length Check and disable reading inverter relay.

But in config - ok

{
  "data": {
    "pin_rx": 16,
    "pin_tx": 17,
    "pin_rts": 5,
    "clientid": "1",
    "baudrate": 115200,
    "txintervallive": 10,
    "txintervalid": 3600,
    "invertertype": "Deye SUN-xx-SG04LP3",
    "EnableRelays": "0",
    "enable_openwbtopic": 0,
    "enableCrcCheck": 0,
    "enableLengthCheck": 1,
    "enable_setters": 0
  }
}

den-localnet, 2024-09-01:
```
      if (this->enableCrcCheck) {
        //CRC Check
        uint16_t crc = this->Calc_CRC(this->DataFrame, dataFrameStartPos, this->DataFrame->size()-2);
        
        if (Config->GetDebugLevel() >=4) {
          Serial.printf("Received CRC: 0x%02X 0x%02X\n", this->DataFrame->at(this->DataFrame->size()-2), this->DataFrame->at(this->DataFrame->size()-1));
          Serial.printf("Calculated CRC: 0x%02X 0x%02X\n", lowByte(crc), highByte(crc));
        }

        if (this->DataFrame->at(this->DataFrame->size()-2) != lowByte(crc) ||
            this->DataFrame->at(this->DataFrame->size()-1) != highByte(crc)) {
          valid = false;
          if (Config->GetDebugLevel() >=2) Serial.println("CRC check  failed!");
        }

        // Check datalength
        if (this->DataFrame->at(dataFrameStartPos+2) != this->DataFrame->size()-dataFrameStartPos-5) {
          valid = false;
          if (Config->GetDebugLevel() >=2) Serial.printf("data length check failed, should be %d but is %d\n", this->DataFrame->at(dataFrameStartPos+2), this->DataFrame->size()-dataFrameStartPos-5);
        }
      } 
```
How Check datalength work if CRC Check disabled? :)

tobiasfaust, 2024-09-02:
Please checkout again, you work an the previous version

den-localnet, 2024-09-07:
Hello!

I have been trying to get data from the BMS port for several days but have had no success.

Analysis of the data from the modbus port showed that even with the correct data frame length, the data can be corrupted inside the data block. This looks like a bug in the inverter itself.
Now I have two more inverters of the same model and within a week I will install them in parallel mode. I want to check the operation on them.

I am also interested in what data I will receive from the master inverter and what from the slave inverter.

den-localnet, 2024-09-24:
Hi!

The problem was in the Board HW-97.
After replacing it with the Board HW-0519, the problem disappeared in the Modbus port.

P.S. Dataframe CRC Check started working.
In the console, I see periodic broken data packets, but now they are not passed by the CRC check.

den-localnet, 2024-09-24:
> I am also interested in what data I will receive from the master inverter and what from the slave inverter.

Each inverter needs a separate MobdasGateway. Each inverter only gives out data about the telemetry it has.

closed 2024-09-29 by tobiasfaust