tobiasfaust/SolaxModbusGateway #76: ID-Daten per MQTT

Lazgar, 2024-09-14:
Hey, 
Sollten die definierten Datenfelder aus den ID-Daten auch per MQTT übermittelt werden? Unter Item Config tauchen die Felder nicht auf, somit kann ich sie auch nicht aktivieren. Falls die nicht übermittelt werden sollten, wäre das ein Feature Request von mir.

LG Lazgar

tobiasfaust, 2024-09-14:
Hi, 
Works-as-designed. Es war nicht geplant die ID Daten zu übermitteln da sich diese ja nie ändern. Die ID-Daten werden einmal pro Stunde abgefragt.
Hast du einen entsprechenden Use-Case warum du diese ständig übermittelt haben möchtest?

Lazgar, 2024-09-14:
Was ich gerne ausgelesen hätte wäre z.B. SolarChargerUseMode oder Manual Mode. Es sind da ja auch alle Einstellungen drin die man auch ändern kann, zumindest beim Solax. Warum fragt das Gateway die ID-Daten überhaupt per Modbus ab wenn es weder per MQTT noch am Gateway selbst ersichtlich ist was abgefragt wird?

tobiasfaust, 2024-09-14:
Es fragt die ID Daten hauptsächlich ab um auf der Hauptseite die Inverter Seriennummer anzuzeigen.

wie gesagt, die ID Daten werden alle 60min abgefragt, die live Daten je nach Einstellung um 10sek Takt.
Mit dem letzten Update kannst du aber auch ID Daten im LiveRequest Array mit angeben. Die würden dann auch alle x Sekunden mit abgefragt.

das ID Daten sowohl mit in der itemconfig angezeigt als auch ausgewählt werden können müsste erst eingebaut werden. Hier würde das senden der ID Daten per MQTT aber auch alle 60min, je nach Einstellung, erfolgen.

Lazgar, 2024-09-14:
Ich werde das mit dem zusätzlichen array mal probieren. Danke

Lazgar, 2024-09-17:
Hey, hätte noch eine Frage, im LiveRequest kann ich nur maximal 5 Bereiche abfragen. Weißt du, ist da die Limitierung das Gateway oder der Inverter?

tobiasfaust, 2024-09-17:
Bzgl ID-Daten per MQTT kommt noch ;)
Die Queue habe ich im Code aktuell auf 5 begrenzt. Falls man es höher setzt kann man auch nur langsamer abfragen. Damit wäre eine live Steuerung nur noch schwer möglich da träger ;)

Lazgar, 2024-09-17:
Das Problem hab ich gerade, beschränkt auf 5 Abfragen muss ich die Bereiche so groß machen (um alles zu kriegen was ich gerne hätte), dass mir der ESP aussteigt :D 
Dachte ich könnte mehr aber dafür kleinere Bereiche definieren, damit mir die Antwort vom Inverter nicht den Rahmen sprengt. Ich will wohl einfach zuviel :P

Lazgar, 2024-09-17:
Danke! Werde ich probieren, hab selbst schon danach gesucht und nicht gefunden ;)

Hast du jemanden der das Gateway am Solax X3 testet? 
Hab bei dem einen oder anderen vordefinierten Item unmögliche Werte zurückbekommen...

tobiasfaust, 2024-09-17:
Leider nicht, das kam über ein Forum

Lazgar, 2024-09-17:
Ich schick dir die Stellen die ich ändern musste, damit bei mir die korrekten Werte kamen.

tobiasfaust, 2024-09-17:
D.h. Du hast ein X3? Dann trage ich dich gerne als author ein und bist Ansprechpartner bei Erweiterungen oder Änderungen 👍

Lazgar, 2024-09-17:
Ja, hab einen X3 Hybrid mit 12kW. Ich hab dir den Request aufgemacht für die potenzialfreien Kontakte, danke nochmal fürs übernehmen ;) Kannst mich gerne eintragen 👍

Lazgar, 2024-09-18:
> Bzgl ID-Daten per MQTT kommt noch ;)

darf ich da noch einen Vorschlag machen? 
Toll wäre eine Abfrage die einmalig nach dem booten ausgeführt wird (für die statischen Werte, die sich nie ändern können: InverterSN, MachineType, etc).
Eine Abfrage für die Live-Daten wie jetzt schon integriert.
Zusätzlich eine Abfrage für die Settings vom Inverter im definierbaren Minuten-Intervall. 
Jeweils mit einer QueueSize 5 damit die Rückmeldung nicht zu groß wird.

tobiasfaust, 2024-09-18:
Verstehe ich noch nicht, die ID Datenabfrage wird aktuell beim booten ausgeführt und dann alle X Minuten, Konfigurierbar im frontend.

Interval for Live Datatransmission in sec (Default: 5)	
Interval for ID Datatransmission in sec (Default: 3600)

Lazgar, 2024-09-18:
klar, ich bin bei dir...unglücklich formuliert. Mein Gedanke war, Dinge kontinuierlich abzufragen die sich gar nicht ändern können, macht keinen Sinn. Deshalb der Vorschlag eine Abfrage zu haben die nur einmalig nach dem booten läuft, damit man die Infos hat aber im weiteren Betrieb den ESP nicht unnötig damit beschäftigt ...

tobiasfaust, 2024-10-30:
This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 5 days.

Lazgar, 2024-12-19:
Hey Tobias,

Ich versuche das Feature ID-Daten per MQTT zu übermitteln, selbst einzubauen. Leider bekomme ich immer ein Problem mit den Tooltips wo man eigentlich das MQTT Topic sehen sollte... Sieht dann so aus:

[image]

Hast du vielleicht einen Tip für mich?

Danke und LG,
Lazgar

Lazgar, 2024-12-20:
Ich kann nur grob erfassen, was der Code macht (bin kein Programmierer), habe die Funktionen, die für die Live-Daten da sind dupliziert und auf ID-Daten geändert. In der Weboberfläche befülle ich eine zweite Tabelle für die ID-Daten und wird mir auch angezeigt. Mir werden die Daten auch per MQTT geschickt. Nur was er mir zamhaut ist, der Mouseover Tooltip, wo das Topic drinsteht, NICHT das OpenWallbox Topic.

Ich habe versucht auch im json bei den ID-Abfragen ein Fake owtopic einzubauen, falls er sich erwartet, dass es zumindest einen Wert damit geben muss. Ändert sich leider nicht...
Mir kommt so vor, als würde er beim befüllen der Webseite nicht sauber trennen können zwischen LIVE Daten update und ID-Daten update.

Und irgendwann schmiert er ab, ich nehme an wegen dem Fehler.

Lazgar, 2024-12-21:
Vielleicht jage ich auch einem Gespenst nach... hab den Fehler mit dem Mouseover Tooltip auch in der 3.3.0 stable nach einer frischen Installation. Kannst du das bestätigen?

tobiasfaust, 2024-12-21:
stimmt, da passt etwas nicht...... muss ich mir ansehen...
Das MouseOver bei den OpenWB-Topics passt aber.

tobiasfaust, 2024-12-21:
ersetze mal bitte Zeile 960-965 in modbus.cpp mit folgendem:
(hinzugekommen ist 2x `std::move(...)`

    doc["name"]       = this->InverterLiveData->at(i).Name.c_str();
    doc["realname"]   = this->InverterLiveData->at(i).RealName.c_str();
    doc["value"]      = std::move(this->InverterLiveData->at(i).value + " " + this->InverterLiveData->at(i).unit);
    doc["active"]["checked"] = (this->InverterLiveData->at(i).active?1:0);
    doc["active"]["name"] = this->InverterLiveData->at(i).Name.c_str();
    doc["mqtttopic"]  = std::move(this->mqtt->getTopic(this->InverterLiveData->at(i).Name, false));

Lazgar, 2024-12-21:
Läuft damit *juhu* Ich werde es mal eine Zeit laufen lassen um zu sehen ob es stabil ist. Wenn ich darf mach ich dir dann einen PR auf :) 
Danke dir!

tobiasfaust, 2024-12-22:
Super, 
Versuche aber bitte so wenig wie möglich Code zu duplizieren. Wenn notwendig in parametrisierte Funktionen auslagern.
Denk auch dran das die Funktion „getlivedataasjson“ alle items incl deiner neuen items zurückliefern muss.

Lazgar, 2024-12-22:
Das ist jetzt die Frage, getlivedataasjson liefert eine liste aller items für die weboberfläche. Mein Ansinnen war es eine zweite Tabelle nur für die ID-Items einzubauen und damit brauch ich eine getiddataasjson. Aber das ist Geschmackssache, wenn du es lieber in einer Liste möchtest, kann ich das umbauen.

tobiasfaust, 2024-12-22:
Ja, das hat einen einfachen Grund. Diese Liste über diese Funktion wird ausgegeben wenn man im Browser 
Http://<espIP>/getitems (oder so ähnlich) eingibt.
Mit einem einfachen http call wird alles ausgegeben, wird für homeassistent und andere weiterverarbeitenden Systeme die nicht mit MQTT arbeiten (wollen oder können) benötigt.

Lazgar, 2024-12-22:
ok, gut zu wissen. Dann bau ich das in die bestehende Tabelle ein :)

tobiasfaust, 2024-12-26:
Merged mit #107

Lazgar, 2024-12-29:
Mir ist  aufgefallen dass er es nach einem Reboot eine Zeit lang nicht schafft die ID Daten abzufragen. Kann es sein, da meist der Intervall für die ID Daten ein vielfaches vom Intervall der Live Daten ist, dass er es verwirft weil schon eine Live Abfrage in der Queue ist?

tobiasfaust, 2024-12-29:
> Mir ist aufgefallen dass er es nach einem Reboot eine Zeit lang nicht schafft die ID Daten abzufragen. Kann es sein, da meist der Intervall für die ID Daten ein vielfaches vom Intervall der Live Daten ist, dass er es verwirft weil schon eine Live Abfrage in der Queue ist?

probier mal bitte den letzten Build, ist ungetestet. Im Endeffekt habe ich nur die Reihenfolge in der Loop getauscht

Lazgar, 2024-12-30:
Scheint es jetzt zu tun. Somit schmeißt er die Live Abfrage weg wenn er gerade eine ID Abfrage in der Queue hat.
Dass man die Abfrage einfach in die Queue dazu packt geht nicht?

Ich weiß nicht ob es beabsichtigt ist, wenn ich einmal einen selbst compilierten build aus meinem fork einspiele funktioniert ota mit den precompiled  firmwares nicht mehr.

tobiasfaust, 2024-12-30:
nein nicht ganz. Wenn in der Queue schon ein Request liegt, wird nicht noch ein weiterer dazu gepackt. Ansonsten funktioniert das parsen der Antwort mit den RegisterJsons nicht mehr.
Es muss immer erst die Queue abgearbeitet sein bevor wieder ein neuer Request in die Queue kommen kann.

> Ich weiß nicht ob es beabsichtigt ist, wenn ich einmal einen selbst compilierten build aus meinem fork einspiele funktioniert ota mit den precompiled firmwares nicht mehr.

Ja, das ist normal da dem Elegant OTA über dem "getDeviceInfo" die Repository Daten mitgegeben werden. Das ist notwendig damit die Daten aus dem richtigen Repo geladen werden können. Das müsstest du auf mein Repo anpassen. 
Aber macht das sinn? Mit einem vorkompilierten Image machst du ja deinen eigenen Fork mit deinen Änderungen wieder obsolet

Lazgar, 2024-12-30:
> nein nicht ganz. Wenn in der Queue schon ein Request liegt, wird nicht noch ein weiterer dazu gepackt. Ansonsten funktioniert das parsen der Antwort mit den RegisterJsons nicht mehr.
Es muss immer erst die Queue abgearbeitet sein bevor wieder ein neuer Request in die Queue kommen kann. 

Ich probier mal was aus, vielleicht klappt es :D

> Ja, das ist normal da dem Elegant OTA über dem "getDeviceInfo" die Repository Daten mitgegeben werden. Das ist notwendig damit die Daten aus dem richtigen Repo geladen werden können. Das müsstest du auf mein Repo anpassen.
Aber macht das sinn? Mit einem vorkompilierten Image machst du ja deinen eigenen Fork mit deinen Änderungen wieder obsolet

Wenn ich es mit einem PR wieder in deinem Repo habe, würde ich schon gern wieder auf den normalen Release zurück. Aber nicht schlimm, dann muss ich es über den Webinstaller machen

Lazgar, 2024-12-30:
Hab einen PR aufgemacht, ich glaub so würden wir keine Abfragen verlieren...

tobiasfaust, 2024-12-30:
du müsstest den PR nochmal machen, ich musste leider auf meinen vorletzten commit zurücksetzen.

Lazgar, 2025-01-02:
Sorry, hab deine Rückmeldung völlig übersehen. Ich schau es mir an 👍

Lazgar, 2025-01-02:
Ich habe das Problem auch, bekomme den Watchdog Trigger und zack -> reboot.
Das yield() löst aber zumindest bei mir das Problem nicht. Es sieht auch nicht so aus als wäre die ID Abfrage das Problem. Es ist gerade mal ein Wert im Json definiert, würde mich wundern wenn er deshalb in ein Timeout fällt. Es passiert mir aber auch nur in diesem Json (stichprobenartig getestet).

Lazgar, 2025-01-02:
Es sind schon sehr sehr viele Live Werte definiert im Json vom Deye, wenn ich einfach stumpf die Hälfte davon raus schmeiße, kann Item Config ganz normal geöffnet werden....

tobiasfaust, 2025-01-02:
Naja, es muss auch Laufen mit einer riesigen Item-Liste ;)
Ich bin gerade dabei die Funktion komplett umzubauen auf ein chunked Response. Brauchst also erstmal nix machen

tobiasfaust, 2025-01-22:
ist das Thema jetzt erledigt?

Lazgar, 2025-01-22:
Ja, kann man schließen 👍

closed 2025-01-22 by Lazgar