tobiasfaust/SolaxModbusGateway #80: Authorization/authentication on the web

den-localnet, 2024-09-25:
Please add simple web authorization. Additionally, you can add IP-based access control.
Sometimes inverters are connected to public Wi-Fi networks and third-party access is highly undesirable.
Thank you!

tobiasfaust, 2024-09-25:
So anyone who integrates their inverter into a public WiFi has only themselves to blame and, to be honest, can't be helped.

private and secured WiFi but public Mqtt broker is ok.

den-localnet, 2024-09-25:
Even in a home network, it is useful to manage access because children can simply accidentally get into the settings and change them.
It happens that guests or neighbors are given access to home Wi-Fi, in our country this is normal.

Even a simple PIN code is enough.

And there are also viruses that can infect devices that get access to the Wi-Fi network.

MagicSven81, 2024-09-25:
you can also create a "v-lan" with separate wifi key, so that no one can connect to this ssid.

den-localnet, 2024-09-25:
> you can also create a "v-lan" with separate wifi key, so that no one can connect to this ssid. Don´t forget to get access only to the 2,4 Ghz (some ESP32 have problems with 5 Ghz Network)

Good advice if your home router has these features. Most of them don't.

P. S. I realized that the developer does not want or cannot do this. Well, let it be as it is.

tobiasfaust, 2024-09-29:
I’ll looking for a suitable solution

den-localnet, 2024-09-29:
> But please keep this hint in your mind: it’s better to split your networks between secured internal and guests, neighbours or children network. Connection between both only with firewall for needed ip’s and ports. It is in your own interest

Thank you!

However, the lock on your safe is installed even though the front door is locked :)

tobiasfaust, 2024-10-09:
Does it work now?
After configuration please restart the esp

den-localnet, 2024-10-09:
> Does it work now? After configuration please restart the esp

Hello,

Work fine, thank you!

P,S, <input size="30" maxlength="40" name="auth_pass" id="auth_pass" placeholder="password" **type="password"**>

closed 2024-10-09 by tobiasfaust