tobiasfaust/SolaxModbusGateway #81: Add support emperature sensor DS18B20

den-localnet, 2024-09-29:
Hello,

Thank you for your work and excellent program for working with the inverter.

Consider adding support for temperature sensors on the 1-Wire bus. 
It would be convenient to be able to transmit the temperature of external sensors to the MQTT broker.

These could be ambient temperature sensors, temperature in the electrical panel (electrical box) and other useful installation locations.

tobiasfaust, 2024-09-29:
Sorry, but that’s meets not the core functionality regarding Modbus.

den-localnet, 2024-09-29:
Thanks for the answer

I don't understand what this has to do with Modbus.
Use a separate library to poll sensors and add topics with data from temperature sensors to the general flow of publications on the broker.
For example, this one: https://github.com/DavidAntliff/esp32-ds18b20

tobiasfaust, 2024-09-29:
Because that, it has nothing to do with Modbus. That functionality belongs not to that project.
There are other projects which can handle one wire better. Maybe tasmota, EspHome or EspEasy

closed 2024-10-06 by tobiasfaust