tobiasfaust/SolaxModbusGateway #82: Add multilingual interface functionality

den-localnet, 2024-09-29:
Hello,

I suggest adding multilingualism to the web interface.
The interface is currently in English, but there are words in German. The bad thing is that it is hardcoded.
If you can make reading from the language files, then I could prepare translations in Russian and Ukrainian.
In the future, anyone can translate the interface and suggest adding it.

It is important that there are separate language files for each translation. You can also read in the settings what language files are available and offer to choose from the available ones.

What do you think about this?

tobiasfaust, 2024-09-29:
Good idea, feel free to eliminate the last German words ;)

just fork the repo, eliminate the german words and create based on your changes a pull request to update my repo.

multiple and dynamic webpages are not a suitable solution due we are using a microcontroller and not a full machine with lots of CPU and Memory ;)

den-localnet, 2024-09-29:
> multiple and dynamic webpages are not a suitable solution due we are using a microcontroller and not a full machine with lots of CPU and Memory ;)

You still use javascript on web interface pages. There is no load for esp32 to read language variables from a file and substitute them in .innerText

In web page
This:
`<td>Select Inverter Type (Default: Solax X1)</td>`
Change:
`<td id='inverter_type'></td>`
Then load page, javascript change .innerText of element document.getElementById('inverter_type')

P.S. I'm still that "programmer", please don't judge me too harshly :)

tobiasfaust, 2024-09-29:
Hi,
Yes, of course, it would be possible only in JavaScript.
You need a user friendly language selection in the header and a dependency to the proper json language file with the well known json structure like as it now for the values.

But I think with english language most of / all people are able to handle this. The priority is not the UI, it’s the functionality ;)

closed 2024-11-14 by tobiasfaust