tobiasfaust/SolaxModbusGateway #84: Bootloop mit eth01 Board

Cracker0815, 2024-10-10:
Hi, ich habe das Problem das der ETH01 in einen Bootloop läuft wenn der Modbus angeschlossen ist bevor der Esp komplett hochgefahren ist.
Zu erkennen is der loop wenn an der RJ45 buchse die LEDs ausgehen.
Ich hab diverse GPIOs probiert, alle 3 RS485-to-TTL Boards versucht, immer das selbe Verhalten.
Kennt jemand das Problem bzw. weiß jemand wie ich das Problem lösen kann.

tobiasfaust, 2024-10-10:
Ich glaube das die die GPIOs für den Modbus ändern musst da das eth01 Board teilweise dieselben gpio benutzt

Cracker0815, 2024-10-10:
Ich hab bereits diverse Kombinationen ausprobiert. Leider keine Änderung. Hast du einen Vorschlag welche GPIOs ich benutzen soll/kann?

tobiasfaust, 2024-10-10:
Schau mal hier, diese gpio‘s dürfen bei Verwendung des lan shields nicht verwendet werden
https://github.com/tobiasfaust/SolaxModbusGateway/blob/ddab0931e1325709e3d39ddd221b0bfa34540822/src/mqtt.h#L39

Cracker0815, 2024-10-10:
Ok danke. Die GPIOs sind auf dem Board gar nicht
herrausgeführt. Ich hab es bereits mit allen GPIOs ausprobiert die auch herrausgeführt sind. (ausser io1) Es ist bei allen IOs das selbe Verhalten. 

Ich hänge mal das pinout mit an.
[image]

tobiasfaust, 2024-10-11:
achte bitte auch drauf, die option "Serial Log Output: TX-Pin (Default: D1)" nicht auf D1 zu setzen

Cracker0815, 2024-10-12:
Sorry für die Anfänger Frage wo find ich denn die option

tobiasfaust, 2024-10-12:
Auf der BasisConfig Seite

Cracker0815, 2024-10-12:
ich hab die latest Version 3.2.2 auf meinem ETH01 aber die Option wird bei mir nicht angezeigt!
Das mit dem Wiki für den ETH01 ist keine schlechte Idee evtl gehts ja aderen auch wie mir 👍 

Screenshot
[image]

tobiasfaust, 2024-10-12:
Ok, ich meine die aktuelle Dev Version. Die kannst du dir nur mit gitpod (siehe Wiki) bauen.
Die nächsten Tage teste ich es selbst mal….

tobiasfaust, 2024-10-12:
ein schuss in Blaue ohne noch selbst getestet zu haben ..... 
1. nach dem Firmwareupload alles einmal stromlos machen und dann neustarten (https://github.com/espressif/arduino-esp32/issues/1912)
2. ein stärkeres Netzteil nehmen (https://www.reddit.com/r/esp32/comments/ivtdzf/esp32_keeps_disconnecting_and_reconnecting_from/?rdt=35885)

Cracker0815, 2024-10-13:
Danke für die Tips. Leider blieben auch die ohne Erfolg.
Als Stromversorgung hab ich schon ein 2,5A Netzteil und der reboot bring auch nix. 
Ich hab jetzt die DEV Version geflasht und die Option wie in deinem Vorschlag gesetzt - wieder mit dem selben Ergebnis. 
Zur Verdeutlichung hab ich auch mal ein Video von dem Verhalten gemacht. 

Video: https://drive.google.com/file/d/12nI8-S4Li-G83UCe9btlJi4aBdBtdlBJ/view?usp=drivesdk

Beachte die LEDs vom ETH01 und dem Breakout Board

tobiasfaust, 2024-10-13:
Danke,
Ich konnte den Fehler mittlerweile schon nachvollziehen. Ich kümmere mich drum. Hab allerdings noch keine Ahnung ab welchen change das LAN nicht mehr funktioniert hat :(

tobiasfaust, 2024-10-13:
ich denke nicht das es am SolaxModbusGAteway Sketch liegt. 
Probiere es mal bitte mit dem offiziellen ETH_LAN8720 example sketch:
https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino
Für das ESP32_ETH01 Board müssen die Parameter im Sketch wie folgt angepasst werden:
<pre>
#define ETH_PHY_TYPE  ETH_PHY_LAN8720
#define ETH_PHY_ADDR  1
#define ETH_PHY_MDC   23
#define ETH_PHY_MDIO  18
#define ETH_PHY_POWER 16
#define ETH_CLK_MODE  ETH_CLOCK_GPIO0_IN
</pre>

Cracker0815, 2024-10-13:
Der LAN Anschluss funktioniert ja grundsätzlich.
wenn der ETH01 bootet und die verbindung zum Wechselrichter/ModBUS nicht besteht dann passt alles - WebIF ist erreichbar usw.
Steckt man dann den Modbus an wenn der ETH01 läuft kann man auch daten auslesen und alles funktioniert so wie es sein soll, nur wenn der Modbus bereits beim booten angeschlossen ist kommt es zu dem Bootloop.
Deshalb denke ich nicht das ich das Problem mit dem offiziellen sketch nachstellen kann.

tobiasfaust, 2024-10-14:
Konnte es jetzt endlich nachstellen. 
Es muss definitiv bei dir an der Konfiguration liegen. Bei mir wird sauber über ESP32_ETH01 und Modbus-BreakOutboard die Daten vom WR ausgelesen.
Im Anhang mal meine konfiguration. Wichtig ist, das du die Pins 16,23,18 nirgends irgendwo in der Weboberfläche definierst.  Es gab noch einen Fehler bei der (De-)Aktivierung der PinRelay Abfrage, bitte teste mal bit dem aktuellen RepoStand
[baseconfig.json](https://github.com/user-attachments/files/17361647/baseconfig.json)
[modbusconfig.json](https://github.com/user-attachments/files/17361648/modbusconfig.json)

Cracker0815, 2024-10-16:
Kann ich leider erst morgen abend testen, bin Dienstlich unterwegs. Wenn ich Neuigkeiten hab meld ich michVon: tobiasfaust ***@***.***>Gesendet: Mittwoch, Oktober 16, 2024 3:29:18 PMAn: tobiasfaust/SolaxModbusGateway ***@***.***>Cc: Cracker0815 ***@***.***>; Author ***@***.***>Betreff: Re: [tobiasfaust/SolaxModbusGateway] Bootloop mit eth01 Board (Issue
 #84)
Funktioniert es jetzt?

Cracker0815, 2024-10-16:
Hi ich wollte eben schon mal die Firmware compilieren bekomme aber folgenden Fehler.
[image]
mach ich was falsch oder gibt es ein anderes Problem

tobiasfaust, 2024-10-17:
Kann ich mir überhaupt nicht vorstellen da die Actions sauber durch kompiliert und die partitions.csv aktuell ist.

Cracker0815, 2024-10-17:
OK keine Ahnung warum aber heute hat es ohne Probleme funktioniert.
Ich hab also die Firmware gebaut und auf den ETH01 geflasht.
Deine config hab ich übernommen, ich hab nur Wechselrichter und MQTT credentials angepasst.
Leider hat sich das Verhalten nicht verändert, der ETH landet immer noch im Bootloop bei angeschlossenem ModBus.
Ich hab mir jetzt nochmal einen neuen ETH01 bestellt evtl hat ja der ne macke....
Sobald der da ist werde ich nochmal testen und dir dann Rückmeldung geben

Cracker0815, 2024-10-18:
Also heute ist der neue ETH01 angekommen.
Ich also eben die Firmware geflasht und erneut getestet leider wieder mit dem gleichen Ergebnis.
somit kann ich jetzt folgendes auf jedenfall ausschließen:
Hardware: sowohl der ETH als auch das Breakoutboard -> neu
Config: hab ich von dir komplett übernommen nur der Wechselrichter und die MQTT credentials wurden angepasst.
hast du noch Möglichkeiten den Fehler zu finden?

tobiasfaust, 2024-10-19:
Entferne mal das Modbus breakout board und teste wie es dann ist. Hast du die aktuell developer Version im gitpod gebaut und dann geflashed?

Ich baue mein Test Board Anfang nächster Woche nochmal auf und mache ein kurzes Video, eventuell hilft es.

Cracker0815, 2024-10-19:
Ja ich hab die aktuelle developer Version mit gitpot gebaut.
Ohne Breakout Board fährt der Esp ohne Probleme hoch, deshalb auch mein Vorschlag die erste Abfrage zu verzögern.
Am Aufbau kanns nicht liegen sonst würde es ja garnicht funktionieren.

tobiasfaust, 2024-10-19:
Check auch mal bitte die serielle Ausgabe des ESP. Macht dieser wirklich einen kompletten reboot oder restartet das lan nur?
Poste auch mal bitte Screenshots der Basis- und Modbusconfig

Cracker0815, 2024-10-19:
Hier die gewünschten Infos:

Basecfg
[image]

ModBuscfg
[image]

und hier der Serial output: (so wie ich das interpretiere bootet er komplett neu)
[SerialOutput.log](https://github.com/user-attachments/files/17445234/SerialOutput.log)

tobiasfaust, 2024-10-21:
Ich kann den Fehler leider nicht nachstellen, bei mir läuft es :(

installiere mal VSC mit der platformio und github Extension. Dann kompiliere mal den Code, lade ihn auf den esp hoch und beobachte den serial Monitor.
Im besten Fall gibt der esp32_decoder eine lesbare Fehlermeldung aus. Dieser ist in der platformio.ini schon aktiviert.

tobiasfaust, 2024-10-22:
Was denn für Fehler…. ?
VSC, git und Python installieren.
In vsc die python und GitHub Erweiterungen installieren. Fertig…. Eigentlich

Cracker0815, 2024-10-23:
Eigentlich hab ich alles installiert... bei mir siehts aber immer so aus
[image]

tobiasfaust, 2024-10-23:
Das sieht nach einer falschen/anderen ETH Library aus. Da ist die parameterstellung im begin() anders. Allerdings weiß ich noch nicht wo die herkommt.

Cracker0815, 2024-10-24:
Nach langem hin und her hab ichs nun hinbekommen! Es lag an der Git Installation! Ich hoffe die Fehlermeldungen helfen dir weiter:

`abort() was called at PC 0x401719fb on core 1

Backtrace: 0x40083799:0x3ffb1bb0 0x4008da35:0x3ffb1bd0 0x40092f59:0x3ffb1bf0 0x401719fb:0x3ffb1c70 0x40171a42:0x3ffb1c90 0x401719a3:0x3ffb1cb0 0x40177b77:0x3ffb1cd0 0x400da559:0x3ffb1f70 0x400de6a9:0x3ffb1f90 0x400def02:0x3ffb21d0 0x400defcb:0x3ffb2250 0x400d9e35:0x3ffb2270 0x400fd1a9:0x3ffb2290
  #0  0x40083799 in tick_hook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/int_wdt.c:74
      (inlined by) tick_hook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/int_wdt.c:62
  #1  0x4008da35 in xRingbufferReceiveUpTo at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_ringbuf/ringbuf.c:1255
  #2  0x40092f59 in multi_heap_realloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:297        
  #3  0x401719fb in get_rf_freq_init at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1314
  #4  0x40171a42 in get_rf_freq_init$part$2 at phy_chip_v7_ana.c:?
  #5  0x401719a3 in get_rf_freq_init at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1302
  #6  0x40177b77 in std::char_traits<char>::copy(char*, char const*, unsigned int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/char_traits.h:352
      (inlined by) std::basic_streambuf<char, std::char_traits<char> >::xsgetn(char*, int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/streambuf.tcc:56
  #7  0x400da559 in modbus::Calc_CRC(unsigned char*, unsigned char) at src/modbus.cpp:861
  #8  0x400de6a9 in modbus::ParseData() at src/modbus.cpp:693 (discriminator 1)
  #9  0x400def02 in StringSumHelper::~StringSumHelper() at C:/Users/Kastn/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:357
      (inlined by) modbus::ParseData() at src/modbus.cpp:656
  #10 0x400defcb in modbus::ParseData() at src/modbus.cpp:788
  #11 0x400d9e35 in handleFiles::HandleAjaxRequest(ArduinoJson::V720PB22::JsonDocument&, AsyncResponseStream*) at src/handleFiles.cpp:75
  #12 0x400fd1a9 in uartAvailableForWrite at C:/Users/Kastn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:578`

tobiasfaust, 2024-10-24:
irgendetwas mit dem uart, der seriellen Kommunikation. Kann ich leider nicht nachstellen. siehe unten. Bei mir läuft es. Siehe unten, siehe Bild.

Du kannst auch testweise in der mail.cpp -> loop die Zeile "mb->loop();" herausnehmen, dann wird der Modbus nur initialisiert, aber nicht mehr zyklisch abgefragt.

Du baust die Firmware aber auch für den **ESP32**, nicht für S, oder S2 oder andere....

<pre>
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Start of Solar Inverter MQTT Gateway
Starting BaseConfig
reading config file
opened config file
{
  "data": {
    "mqttroot": "Solax",
    "SelectConnectivity": "eth",
    "SelectLAN": "WT32-ETH01",
    "mqttserver": "test.mosquitto.org",
    "mqttport": "1883",
    "mqttbasepath": "home",
    "UseRandomClientID": "yes",
    "serial_rx": 3,
    "serial_tx": 1,
    "sel_auth": "off",
    "debuglevel": 2
  }
}
ready
Starting Wifi and MQTT
Ethernet started
Ethernet connected
ETH MAC: B0:A7:32:05:8E:FF, IPv4: 192.168.10.236, FULL_DUPLEX, Mbps: 100
Wait for connect
Starting MQTT (test.mosquitto.org:1883)
[ 81685][E][vfs_api.cpp:105] open(): /littlefs/modbusitemconfig.json does not exist, no permits for creation
MQTT aktiviert
Starting WebServer
WebServer started...
Attempting MQTT connection as Solax-40236
connected... 
MQTT resubscribed to: home/Solax/set/TargetBatSOC
no response from client
no response from client
no response from client
no response from client
no response from client
no response from client
no response from client
no response from client
no response from client
no response from client
no response from client
no response from client
</pre>

tobiasfaust, 2024-10-25:
ersetze mal in der main.cpp diese zeile:
<pre>
mb->loop(); 
</pre>
mit dieser hier
<pre>
if(millis()>(2*60*1000)) mb->loop(); 
</pre>

dann werden wir sehen ob es ein timing problem ist

Cracker0815, 2024-10-25:
OK ich habs getestet, die Verzögerung um 2 min hat funktioniert aber dann genau das selbe verhalten :-(
Ich hatte gehofft es klappt...

tobiasfaust, 2024-10-25:
Probier, wie bei mir im Test, den WR nicht anzustecken. Sodass du ein „no response“ bekommst.
Du musst den Fehler sukzessive eingrenzen um an die Stelle zu kommen die den Fehler auslöst.
Es hilft auch ungemein an sehr vielen Stellen im code testweise logmeldungen auszugeben um zu wissen, bis wohin der Code kommt

Cracker0815, 2024-10-25:
Also wenn der WR nicht angesteckt ist siehts genau so aus wie bei dir:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Start of Solar Inverter MQTT Gateway
Starting BaseConfig
reading config file
opened config file
{
  "data": {
    "mqttroot": "Test",
    "SelectConnectivity": "eth",
    "SelectLAN": "WT32-ETH01",
    "mqttserver": "test.mosquitto.org",
    "mqttport": "1883",
    "mqttbasepath": "home",
    "UseRandomClientID": "yes",
    "serial_rx": 1,
    "serial_tx": 1,
    "sel_auth": "off",
    "debuglevel": 2
  }
}
ready
Starting Wifi and MQTT
Ethernet started
Ethernet connected
ETH MAC: E8:6B:EA:D8:30:1B, IPv4: 192.168.0.113, FULL_DUPLEX, Mbps: 100
Wait for connect
Starting MQTT (test.mosquitto.org:1883)
MQTT aktiviert
Failed to parse JSON Register <set> Data: EmptyInput
Starting WebServer
WebServer started...
no response from client
Attempting MQTT connection as Test-27296 
connected... 
no response from client
no response from client
no response from client
no response from client
```
steck ich dann im laufenden Betrieb den WR an läuft alles wie es soll und es gibt auch in der Seriellen Ausgabe keinerlei Fehler

Wie mach ich das mit den Logmeldungen im code? und an welchen stellen?

tobiasfaust, 2024-10-25:
als erstes setze mal als erstes in jeder funktion ein **Serial.println()** logstatement,
zb.
<pre>
void modbus::ReceiveReadData() {
Serial.println("ReceiveReadData");
</pre>

dann siehst du welche Funktion wann aufgerufen wird. Bei der letzten Logausgabe vor dem Reboot tritt dann der fehler auf. Dann muss man innerhalb dieser funktion wieder mehrere Logstatements setzen bis man auf den problemantischen Befehl trifft

Cracker0815, 2024-10-25:
OK hab ich soweit gemacht - ich konnte feststellen das, dass Programm bei der Funktion ParseData abbricht.
hier mal das Log:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Start of Solar Inverter MQTT Gateway
Starting BaseConfig
reading config file
opened config file
{
  "data": {
    "mqttroot": "Test",
    "SelectConnectivity": "eth",
    "SelectLAN": "WT32-ETH01",
    "mqttserver": "test.mosquitto.org",
    "mqttport": "1883",
    "mqttbasepath": "home",
    "UseRandomClientID": "yes",
    "serial_rx": 1,
    "serial_tx": 1,
    "sel_auth": "off",
    "debuglevel": 2
  }
}
ready
Starting Wifi and MQTT
Ethernet started
Ethernet connected
ETH MAC: E8:6B:EA:D8:30:1B, IPv4: 192.168.0.113, FULL_DUPLEX, Mbps: 100
Wait for connect
Starting MQTT (test.mosquitto.org:1883)
modbus
LoadInvertersFromJson
LoadJsonConfig
init
LoadInvertersFromJson
LoadInverterConfigFromJson
LoadRegItems
LoadRegItems
LoadJsonConfig2
QueryIdData
enableMQTT
MQTT aktiviert
GenerateMqttSubscriptions
Failed to parse JSON Register <set> Data: EmptyInput
Starting WebServer
WebServer started...
loop
QueryLiveData
QueryQueueToInverter
Calc_CRC
ReceiveReadData
Calc_CRC2
ParseData

abort() was called at PC 0x40171c17 on core 1

Backtrace: 0x40083799:0x3ffb1bb0 0x4008da35:0x3ffb1bd0 0x40092f59:0x3ffb1bf0 0x40171c17:0x3ffb1c70 0x40171c5e:0x3ffb1c90 0x40171bbf:0x3ffb1cb0 0x40177d93:0x3ffb1cd0 0x400da639:0x3ffb1f70 0x400de851:0x3ffb1f90 0x400df0b1:0x3ffb21d0 0x400df17f:0x3ffb2250 0x400d9ee1:0x3ffb2270 0x400fd3c5:0x3ffb2290
  #0  0x40083799 in tick_hook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/int_wdt.c:74
      (inlined by) tick_hook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/int_wdt.c:62
  #1  0x4008da35 in xRingbufferReceiveUpTo at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_ringbuf/ringbuf.c:1255
  #2  0x40092f59 in multi_heap_realloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:297        
  #3  0x40171c17 in bt_i2c_write_set at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1398
  #4  0x40171c5e in bt_i2c_write_set at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1404
  #5  0x40171bbf in bt_i2c_write_set at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1396
  #6  0x40177d93 in void std::locale::_Impl::_M_init_facet_unchecked<std::collate<wchar_t> >(std::collate<wchar_t>*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/locale_classes.h:617
      (inlined by) std::locale::_Impl::_M_init_extra(std::locale::facet**) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/cow-locale_init.cc:119
  #7  0x400da639 in modbus::PrintDataFrame(unsigned char*, unsigned char) at src/modbus.cpp:925
  #8  0x400de851 in ArduinoJson::V720PB22::detail::MemberProxy<ArduinoJson::V720PB22::JsonDocument&, char const*>::MemberProxy(ArduinoJson::V720PB22::JsonDocument&, char const*) at .pio/libdeps/firmware_ESP32_WebSerial/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:21
      (inlined by) ArduinoJson::V720PB22::detail::enable_if<ArduinoJson::V720PB22::detail::IsString<char const*, void>::value, ArduinoJson::V720PB22::detail::MemberProxy<ArduinoJson::V720PB22::JsonDocument&, char const*> >::type ArduinoJson::V720PB22::JsonDocument::operator[]<char const>(char const*) at .pio/libdeps/firmware_ESP32_WebSerial/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:194
      (inlined by) modbus::ParseData() at src/modbus.cpp:736
  #9  0x400df0b1 in modbus::QueryQueueToInverter() at src/modbus.cpp:388
  #10 0x400df17f in modbus::QueryQueueToInverter() at src/modbus.cpp:398
  #11 0x400d9ee1 in myMQTTCallBack(char*, unsigned char*, unsigned int) at src/main.cpp:31
  #12 0x400fd3c5 in uart_ll_is_tx_idle at C:/Users/Kastn/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/hal/esp32/include/hal/uart_ll.h:789    
      (inlined by) uartFlushTxOnly at C:/Users/Kastn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:698
      (inlined by) uartFlushTxOnly at C:/Users/Kastn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:691
```

Hilft das evtl schon weiter? bzw. wie mach ich das mit dem loggen innerhalb der Funktion? Hast du evtl eine Beispielzeile für mich?

tobiasfaust, 2024-10-25:
dann nimm mal zeile 805 heraus oder kommentiere sie testweise aus.

sicherheitshalber entferne auch mal bis auf 2 oder 3 items alle anderen aus der data sektion in deinem json

Cracker0815, 2024-10-25:
Sieht auf den ersten blick schon mal gut aus! 👍 
- erster Versuch mit 2 Items und Zeile "805" auskommentiert
- zweiter Versuch mit allen benötigeten Items und Zeile "805" auskommentiert
bei beiden Versuchen kein crash!
Für was ist die Zeile 805 zuständig, kann man die einfach weglassen bzw wie soll ich nun weiter machen?

tobiasfaust, 2024-10-25:
Weglassen geht nicht, ich schreibe sie um. Lasse sie erstmal weg und konfiguriere dein Setup.
Ich melde mich spätestens Montag mit einer alternativen Zeile 805

tobiasfaust, 2024-10-26:
probier mal bitte den aktuellen Fix aus. 
Ich habe es mit aktiviertem Debugmode (modbus.h) und Solax-X1 config getestet.

Falls es immer noch nicht geht bitte
* Zeile 770-776 auskommentieren sodass es wieder läuft
* von der RAW Data Seite brauche ich deinen kompletten Hex-String mit dem du den ESP-Crash verursachst

Cracker0815, 2024-10-27:
Hi also die änderung funktioniert 1a! 👍 
Ich hab den fix eben an einem SPH und an einem MIC WR erfolgreich getestet.

Brauchst du den Hex-String trotzdem?

Vielen dank für deinen großartigen Einsatz! 💯

tobiasfaust, 2024-10-28:
Super, perfekt. Brauche dann den Hexstring natürlich nicht mehr

closed 2024-10-28 by tobiasfaust