tobiasfaust/SolaxModbusGateway #85: Firmware Update?

ronk74, 2024-10-10:
Hallo,

bei mir läuft noch FW3.1.1. Ich möchte den ESP32 gern auf die aktuelle Firmware updaten. Reicht es aus, die Firmware über das Webinterface hochzuladen? Oder muss der Weg über das Flash Download Tool gegangen werden? Weiterhin weiß ich nicht genau, welche ESP32-Version ich habe, ist das irgendwo zu erkennen?

Vielen Dank!

MagicSven81, 2024-10-10:
hier musst du über das Flash Download Tool gehen, da du auch die littlefs.bin hochladen musst.
Vermutlich hast du einen ESP32 (normal) - wichtig, die Settings wie von Tobias im Wiki geschrieben verwenden.

ronk74, 2024-10-10:
Danke sehr. Ich probiere das morgen. Viele Grüße!

ronk74, 2024-10-11:
Update ist geglückt. Danke für die Hilfe!