tobiasfaust/SolaxModbusGateway #90: Gateway forgets Item Config with every restart

benhubert, 2024-10-27:
First of all, thanks for your work and this great project!

Here's the issue with my gateway:
Every time I power cycle my gateway all Items are unchecked again (except for one). I have to walk through all items and tick the checkbox to receive updates again. All other configuration (Basis Config, Modbus Config) is restored after reboot.

My Inverter: Solax X3 G4
Inverter Type in use: Solax-X3
Release in use: 3.2.3 (@development) of Oct 8 2024 22:15:16

If you need any other information, let me know.

tobiasfaust, 2024-11-01:
Hi,
sorry, but i cannot reproduce your incident. I tested on a fresh ESP32, select Solax-X3, checked some items, store it and reboot. After reboot all items are still checked. Please see the log.

You can check with files section the file "modbusitemconfig.json" if all your checked items are stored sucessfully. 

<pre>
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Start of Solar Inverter MQTT Gateway
Starting BaseConfig
[    65][E][vfs_api.cpp:105] open(): /littlefs/baseconfig.json does not exist, no permits for creation

ready
Starting Wifi and MQTT
WiFi Start
WiFi interface ready
WiFi client started
Connected to access point
WiFi connected with local IP: 10.0.2.102
Starting MQTT (test.mosquitto.org:1883)
[  5365][E][vfs_api.cpp:105] open(): /littlefs/modbusconfig.json does not exist, no permits for creation
[  9558][E][vfs_api.cpp:105] open(): /littlefs/modbusitemconfig.json does not exist, no permits for creation
MQTT aktiviert
Starting WebServer
WebServer started...
Attempting MQTT connection as Solax-48071
connected... 
reading config file
opened config file
{
  "data": {
    "mqttroot": "Solax",
    "SelectConnectivity": "wifi",
    "mqttserver": "test.mosquitto.org",
    "mqttport": "1883",
    "mqttbasepath": "home",
    "UseRandomClientID": "yes",
    "serial_rx": 3,
    "serial_tx": 1,
    "sel_auth": "off",
    "debuglevel": 3
  }
}Request queue data to inverter: 0x01 0x03 0x01 0xF4 0x00 0x77 0x45 0xE2 
Read Data from Queue: 
no response from client
Request queue data to inverter: 0x01 0x03 0x02 0x6B 0x00 0x41 0xF5 0x9E 
Read Data from Queue: 
no response from client
Request queue data to inverter: 0x01 0x03 0x01 0xF4 0x00 0x77 0x45 0xE2 
Read Data from Queue: 
no response from client
Request queue data to inverter: 0x01 0x03 0x02 0x6B 0x00 0x41 0xF5 0x9E 
Read Data from Queue: 
no response from client
Request queue data to inverter: 0x01 0x03 0x01 0xF4 0x00 0x77 0x45 0xE2 
Read Data from Queue: 
no response from client
Request queue data to inverter: 0x01 0x03 0x02 0x6B 0x00 0x41 0xF5 0x9E 
Read Data from Queue: 
no response from client
Request queue data to inverter: 0x01 0x03 0x01 0xF4 0x00 0x77 0x45 0xE2 
Read Data from Queue: 
[... 411 lines cut ...]
WebServer started...
Request queue data to inverter: 0xFF 0xFF 
Read Data from Queue: 
no response from client
Request queue data to inverter: 0xFF 0xFF 
Read Data from Queue: 
no response from client
Attempting MQTT connection as Solax-17258
connected... 
Publish home/Solax/IP: 10.0.2.102
Publish home/Solax/version: 3.2.3(@development)
Publish home/Solax/state: Online
Request queue data to inverter: 0xFF 0xFF 
Read Data from Queue: 
no response from client
Request queue data to inverter: 0xFF 0xFF 
Read Data from Queue: 
no response from client
Request queue data to inverter: 0xFF 0xFF 
Read Data from Queue: 
no response from client
Request queue data to inverter: 0x01 0x04 0x00 0x00 0x00 0x78 0xF0 0x28 
Read Data from Queue: 
no response from client
</pre>

MagicSven81, 2024-11-03:
sorry for my stupid question, but did you "save" the settings after you activate the items?

tobiasfaust, 2024-11-03:
Yes, of course. Save Button does Save it.

benhubert, 2024-11-26:
Absolutely not a stupid question! :tada: It is very likely that I never saw this save button before. Now, also after a reboot, all my items are activated again. Many thanks @MagicSven81 !

But I have to add ... this button is quite far away from the table and it only appears when I scroll down to the very bottom. This could be improved from UX perspective:
[image]

closed 2024-11-26 by benhubert