tobiasfaust/SolaxModbusGateway #91: Flash to adafruit_feather_esp32s3 not working

rgrae81, 2024-10-30:
HI,

I have an adafruit_feather_esp32s3 and tried to flash the S3 bin files.
after flashing it won't boot.
Output on Serial:
invalid header: 0xffffffff
invalid header: 0ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048836
invalid header: 0xffffffff

another firmware works on the board.

regards
Roman

tobiasfaust, 2024-10-31:
Hi
It depends on your board type. Please use gitpod (see my wiki page) and change the parameters of esp-s3 section according your board type.

here are some examples:

https://docs.platformio.org/en/latest/boards/espressif32/adafruit_feather_esp32s3_nopsram.html

https://docs.platformio.org/en/latest/boards/espressif32/adafruit_feather_esp32s3.html

rgrae81, 2024-10-31:
Thanks, will have a look.
I had one left from another Project, so it's cheaper to use this ;-)

tobiasfaust, 2024-11-01:
if you found the right config, please share it to us ;) so maybe i will add this to platformio configfile

closed 2024-12-16 by tobiasfaust