tobiasfaust/SolaxModbusGateway #93: Serial Output RX und TX Pin D1

aceron76, 2024-11-07:
Hallo, ich habe mir diesen MODBUSGateway nach Anleitung nachgebaut für Growatt SPH, leider komunizieren die beiden aber nicht
Nach dem ESP Flash ist in den basic config alles so wie es sein soll wenn ich aber meine daten eintrage dann steht bei  serial output RX und TX auf Pin D1 und ich kann es auch nicht korrigieren hat jemand eine Idee
Danke
[image]

MagicSven81, 2024-11-07:
Serial Log Output ist erst einmal irrelevant.
Viel wichtiger ist, was du für einen Growatt Wechselrichter hast.
Diese werden je nach Typ (TL3-BH(UP) unterschiedlich angeschlossen (Pins des RJ45 Steckers)

aceron76, 2024-11-07:
Hallo Danke für deine Rückmeldung, wusste ich nicht das diese Serial Log Output  nicht relevant sind, ich habe eine growatt sph 10000tl3 bh-up inverter und Pin1= B, Pin5=A hast du ander Informationen?

tobiasfaust, 2024-11-07:
Die serial log Output Pins muss man eigentlich nie ändern, es sei denn das man einen anderen esp anschließt um das serial log zu lesen wenn das Gateway nicht im Wlan ist und dessen Default Pins belegt sind. ;)

wenn man sie ändert bekommt man auch kein serielles log mehr per USB raus ;) weil diese fix auf D1 und D3 lauschen

tobiasfaust, 2024-11-08:
Bekommst du nur mit der Doku deines WR heraus

MagicSven81, 2024-11-08:
[image]

aceron76, 2024-11-10:
Hallo Magicsven81, danke für dein Unterstützung, dieses Bild habe ich auch schon gefunden. ich habe eine HW97 TTL converter, verbunden habe ich D16--RO, D17--DI und D5 --DE+RE nach Anleitung 
[image]

[image]

MagicSven81, 2024-11-12:
Baudrate auf 9600 setzen. Der Growatt arbeitet nicht mit 19200

aceron76, 2024-11-13:
Danke für den Tip ändert aber leider nichts 😒

aceron76, 2024-11-13:
Muss ich irgendwas im Wechselrichter einstellen bin am verzweifeln

tobiasfaust, 2024-12-10:
Hast du es hin bekommen?
Die Modbus ClientID und Baudrate müssen mit dem des WR exakt übereinstimmen.
Die Modbus Kontakte A- und A+ dürfen auch nicht vertauscht sein. 
Zum ersten Test kannst Du die beiden Dataframe Tests auch erstmal ausschalten.
Ansonsten kann ich nur auf die FAQ‘s verweisen und du musst probieren bis es klappt.

saschoctm, 2025-01-04:
Hi, gerade gesehen, schau mal hier, man muss was in den settings setzen, ist schon lange her bei mir und ich habe einen SPH10000
[image]

tobiasfaust, 2025-01-04:
@saschoctm ist das etwas für mein FaQ Seite?

saschoctm, 2025-01-04:
Ja sicherlich, das ist aus der BDA vom Growatt SPH TL3 BH-UP Series, ist nen hybrid WR, ohne diese Einstellung am WR geht es nicht.

MagicSven81, 2025-01-04:
Das Thema, dass der VPP Mode am Wechselrichter gesetzt werden muss betrifft alle Growatt Wechselrichter.
Bitte beim Anschluss des RS485 auf den exakten Wechselrichter Typ achten.

Es gibt den Growatt SPH-1000 TL3 BH UP und den normalen TL3 BH  (ohne UP)
Diese haben eine andere Pinbelegung am RJ45 Stecker.

[image]

aceron76, 2025-01-13:
Hallo Danke für die vielen Ratschläge leider hat nichts zum Erfolg geführt. Als Info: ich habe mit der ShineBus SW und Port3 mit RS485-USB eine Verbindung aufbauen können aber das wars dann auch schon.

aceron76, 2025-01-15:
Also mit USB kommend laufend Daten im SOLAX kommt nichts an ich habe Release 3.3.0 gibt es noch andere Releases? Danke für eure Unterstützung

closed 2025-01-17 by aceron76

tobiasfaust, 2025-01-18:
Hi
alle releases funktionieren. Es muss definitiv an deiner Verkabelung und/oder Konfiguration liegen.

aceron76, 2025-01-19:
Hallo Tobias, Danke für die Info, habe das Problem gelöst. Es war das RS485 Board HW-97, mit einem HW-0519 funktionierts.
Alles so wie es sein soll top Danke dafür. Eine Frage: wie bekomme ich den Batteriezustand SOC in die Readings, ich verwende Fhem.
sG
Erwin

[image]

[image]

tobiasfaust, 2025-01-19:
Auf der Item Seite musst du nachsehen wie das korrekte mqtt topic ausschaut und dieses dann genau so in FHEM definieren. 

Wenn dir das Projekt gefällt und es jetzt bei dir gut funktioniert würde ich mich immer über eine kleine Spende freuen ;)

aceron76, 2025-01-19:
Hallo Tobias, habe ich hinbekommen. Jetzt ist es aber so das sich der Solax Gateway  in Fhem ca alle 3 Minuten neu anmeldet und immer mit einer anderen ID wieso ist das?

[image]

[image]

aceron76, 2025-01-24:
Hallo Tobias, Danke ich habs hinbekommen. Ich habe in Fhem das device manuell angelegt somit bleibt es stabil.
Andere Frage, ich habe zum Growatt-SPH.json etwas hinzugefügt, nach einem Absturz musste ich die files neu laden aber dann waren meine Änderungen auch wieder weg. Wie kann ich mein .json file fix einspielen?
Danke sG Erwin

tobiasfaust, 2025-01-24:
Manuelle Anpassungen musst du nach einem neu flashen der Files immer wieder über die files section selbst updaten. 
Manuelle json Anpassungen an der regfiles sollte immer nur temporär sein wenn möglich ;)

aceron76, 2025-01-29:
Hallo Tobias, ich habe versucht nach deiner Anleitung über gitpod das Growatt-SPH anzupassen, aber dann komme ich nicht weiter, reicht es wenn ich die Growatt-SPH.json bearbeite, wo kann ich das command eingeben? platformio run --target buildfs to build the littlefs.bin und wie bekomme ich dann das neue file? Oder mir würde das Growatt-SPH-V124.json auch schon reichen, wie bringe ich das auf den ESP?
Danke
sG Erwin

tobiasfaust, 2025-01-29:
wenn du nur das json anpassen willst musst du das nicht im gitpod tun.
Im WebUI, files section kannst du das betreffende file laden, anpassen und wieder speichern.

Wenn möglich vor dem speichern über jsonlint.com prüfen ob das json korrekt ist

aceron76, 2025-01-29:
Alles klar Danke, aber ich dachte das sollte man so nicht machen.

_Manuelle json Anpassungen an der regfiles sollte immer nur temporär sein wenn möglich ;)_

aceron76, 2025-01-29:
Ok, Danke kenn mich aus
sG
Erwin