tobiasfaust/SolaxModbusGateway #95: Some suggestions to cherry pick :-)

Obihoernchen80, 2024-11-20:
Hi Tobias,
I have been using your great project for over a year now.
Thanks for settings this up.

I have had some minor changes going locally and recently put them into a fork repo here on github.
Not being an avid user of github, I created a fork of your project. I hope thats alright and I dont want the fork to be around for long.

Here are some commits I would propose to take over:
- [GetLiveDataAsJson: use c_str() to allow for longer names in mqtt_root](https://github.com/Obihoernchen80/SolaxModbusGateway/commit/6f3bd44cd333b915f7f2d889007d13963d7d7c73)
When I configure the mqtt_root to be e.g. "Solax123456789" the modbus item page fails (root cause being an invalid json response). I had no issues with the codee from aproxx one year ago. The commit fixes that issue.

- [add to wifi BSSid to status](https://github.com/Obihoernchen80/SolaxModbusGateway/commit/ad6b54cef0404153f36d53692a09887ff6595746)
I have some wifi repeaters around, and sometimes the esp32 connects to a far away one and has a bad connection. The BSSID can be used to trace which access point is being used.

- [Solax-X3: add GridPower register](https://github.com/Obihoernchen80/SolaxModbusGateway/commit/404117f6877a0645fdccf59a8b25c90f5caf469e)
I have been sucessfully using my QCells QVolt HYB-G3 with the Solax-X3 profile for over a year know. After all, its just rebranding of the same device. I noticed one register missing and added it.
Also I would suggest to rethink having a seperate register file for the QCells QVolt HYB-G3  (called "QVolt HYP-G3.json") since it contains less registers and the Solax-X3 works just as well.

Looking forward to hear back from you!
Regards,
Chris

tobiasfaust, 2024-11-21:
Hi
That sounds great, feel free to open a pull request.

tobiasfaust, 2024-12-18:
fixes implemented

closed 2024-12-18 by tobiasfaust