tobiasfaust/SolaxModbusGateway #96: Growatt SPH3000-RS485-ESP32-MQTT

eckhard58, 2024-11-26:
Good morning Tobias,
i managed to register here on GitHub. I'm not very good at English.

I use the ESP32 on a Growatt SPH3000. (Year of manufacture 10/24, ShineWiFi is now on the USB port)
The connection is only established if I disconnect the RS485 connection and reconnect it after the MQTT connection has been established.

Here is the desired HTerm output from the ESP32
[mit WR.txt](https://github.com/user-attachments/files/17914614/mit.WR.txt)
[ohne WR.txt](https://github.com/user-attachments/files/17914635/ohne.WR.txt)
If I see it correctly, there are a few ERRORs

Regards, Eckhard

eckhard58, 2024-11-26:
PS
After an indefinite period of time, the MQTT connection goes offline. After briefly disconnecting the RS485 connection, the MQTT connection goes back online

tobiasfaust, 2024-11-28:
Aus den Logs kann ich leider keinen Fehler herauslesen, sieht alles normal aus. Ich finde auch kein Hinweis auf einen MQTT dis- oder reconnect.
Wie sieht es aus wenn RS485 steckt und der ESP neu startet? Es gibt eigentlich keine Abhängigkeit zwischen WLAN/MQTT und der RS485 Verbindung

eckhard58, 2024-11-28:
Hallo Tobias,wenn die MQTT-Verbindung stirbt läuft ja die RS485-Verbindung weiter, das sieht man im HTerm.Um MQTT wieder zum Laufen zu bringen trennen ich kurz die RS485 und nach kurzer Zeit sind alle Werte für unbestimmte Zeit wieder da.Ich hab jetzt mal einen anderen Wandler benutzt,  leider ohne Verbesserung. Könnte es sein das der ESP32 mit der Live Datatransmission von 5sec überfordert ist?Ich hab sie jetzt mal auf 30sec gestellt. Mal sehen wie das damit läuft.Gruß Eckhard

tobiasfaust, 2024-11-28:
nein, der ESP ist  nicht überfordert, bei mir läuft es im 3sec Intervall sauber seit Monaten durch.

- Hast du einen "normalen" ESP32? Also keinen S2, S3, C3 o.ä.? 
- Das WLAN Signal ist stabil und stark genug? (RSSI < 75)
- Poste mal ein Log mit Loglevel 4 von einem ESP Neustart und angesteckten RS485. (bitte lesbar, die ersten beiden Logs waren nicht wirklich lesbar)
- Poste mal Screenshots der Baseconfig und der Modbusconfig. 
- Ist GND zwischen WR und ESP32 verbunden? Ist eher optional, bei Problemen sollte man es aber verbinden.

Ich habe keine Ahnung was bei dir herum zickt :(

eckhard58, 2024-11-28:
die nenne sich ESP32-WROM-32. ich habe zwei verschiedene davon.
Und beim flashen habe ich auch die normalen dateien genommen.

Das were ich mal probieren, leider geht gerade mein HTerm nicht mehr connecten will auf dem Laptop. Einmal abgestürzt und nun spinnt das HTerm.

Ich weis auch nicht was da zickt

MagicSven81, 2024-11-29:
hängen bei dir "Wlan Repeater" zwischen Router und Modbus Modul?
Falls ja, Intervall einmal testweise auf 10s stellen.
Evtl. liegt es auch an deinem MQTT Server.

Ich benutze ebenalls den ESP32-WROOM-32  (ESP32) seit über 1,5 Jahren ohne Probleme.
Außer, dass er sich alle 20-30 Tage mal neu startet - aber das stört mich nicht.

eckhard58, 2024-11-29:
ja es ist ein Router dazwischen.
Den Intervall habe ich schon auf 20s stehen.
Der MQTT2 Server ist der von Fhem.
Ich werde mal paar LOGs machen.

tobiasfaust, 2024-11-29:
Probier mal testweise den Test.mosquitto.org zu benutzen

eckhard58, 2024-11-29:
Hm, das mit Test,mosquitto haut bei mir nicht hin, hab ich noch nie benutzt.
Da muss ich mich erst noch besser belesen.

hier noch mal die LOGs
[WR 131.txt](https://github.com/user-attachments/files/17963305/WR.131.txt)
[FhemLOG 131.txt](https://github.com/user-attachments/files/17963308/FhemLOG.131.txt)

tobiasfaust, 2024-11-30:
Der esp stürzt ein paar mal ab bevor er läuft. Häng mal ein stärkeres Netzteil dran.

eckhard58, 2024-12-01:
Die Netzteile haben alle mindestens 3A bei 5V. 
Die ESP`s brauchen maximal 70mA.
Es kann sein das ich noch mal Reset gemacht habe.
Ich hab jetzt auch mal GND mit durchgezogen.
Abschlußwiderstand von 120 Ohm hab ich auch noch eingebaut.

Es will einfach nicht durchlaufen.
Der ESP hängt sich irgend wann auf und es hilft nur ein kurzes trennen der RS485 Verbindung. Genau so verhält es sich bei Spannug einschalten und Reset.

tobiasfaust, 2024-12-02:
Ich habe ein Vermutung, kannst du den esp mal ohne Repeater direkt in die Nähe deines AP bringen? Ggf. Die RS-485 Verbindung über ein Netzwerkkabel verlängern.

Es kann sein das der Webserver starten will, die Wifi Verbindung aber noch nicht steht.

eckhard58, 2024-12-03:
Hallo Tobias,
gestern habe ich mal auf einen Test RbPi5 den Home Assistent installiert und MQTT eingebunden. Der ESP32 verhält sich genau so wie mit dem Fhem MQTT Server.
Sobald ich den Growatt mit dem ESP32 verbinde läuft er nur noch eine gewisse Zeit.
Ich hab mittlerweile für mein neueren Growatt SPH 3000 die Einstellungen für den Growatt MOD 4000 verwendet. Damit ist der ESP32 seit 17 Stunden am laufen. Leider sind da paar Einstellungen nicht ganz richtig. z.B. werden die Batteriewerte nicht angezeigt, die Anteige PV1 und PV2 müssten angepasst werden und drei Phasen hat mein WR auch nicht.
Ich hab mal in den Files gestöbert. Wo und was ich da tun kann endzieht sich zur Zeit meiner Kenntnis. Dafür bin ich leider nur ein Anwender.
Wenn Du dazu noch einen Tip hast?  Wie ich welche Progamme anpassen könnte/müsste . Und was brauche ich für Daten von meinem WR?

Gruß Eckhard

tobiasfaust, 2024-12-03:
Hier findest du eine Anleitung: https://github.com/tobiasfaust/SolaxModbusGateway/wiki/adding-new-inverters-or-items#find-out-live-data-get-new-items

Probier mal bitte die neueste Development version über den web-installer: https://tobiasfaust.github.io/SolaxModbusGateway/
Jetzt wird (hoffentlich) der webserver erst gestartet wenn der ESP mit dem WLAN verbunden ist.

Anscheinend liefert sein WR irgendetwas zurück was den ESP zum Absturz bringt, deshalb brauche ich ein Log mit Loglevel 4 zum Zeitpunkt des Absturzes um die Antworten des WR zu sehen.

eckhard58, 2024-12-03:
Deine Anleitung hab ich schon gelesen. Für mich noch etwas verwirrend.

Das Release: 3.3.0 läuft jetzt auf einem ESP32. schauen wir mal obs das ist.

Welchen Log meinst Du? Den über ESP32-USB mit HTerm?

tobiasfaust, 2024-12-03:
Ja, mit hterm.
Wenn die Anleitung für dich verwirrend ist wäre super wenn du diese anpassen könntest das jemand der nicht im Thema steckt diese auch versteht. Da bin ich wohl etwas betriebsblind:(

eckhard58, 2024-12-03:
Dazu muss ich erstmal erfolgreich zusätzliche Werte hinzu gefügt haben, dann ist ein anpassen der Aleitung möglich. Es geht schon los das ich fünf verschiedene Datenbankschemen und Register habe. Welches passt zu meinem neuen WR.
Ja ich werd mir das angucken.

Bis jetzt sieht es aber schon mal besser aus,
Release 3.2.2 startet und läuft mit Growatt-MOD 
Release 3.3.0 startet und läuft mit Growatt-SVG

eckhard58, 2024-12-05:
Hmmm, jetzt hab ich dich zu getextet, tut mit leid.
Bin schon ein bisschen weiter gekommen.

Eine Frage habe ich noch.
Wo kann mann das machen?
- request: 01 03 00 00 00 07 08 04
- Response: 01 03 .....

Ich hab die Inverter SerialNumber im Status nur einmal gesehen und weis aber nicht wie so.

Eckhard

tobiasfaust, 2024-12-05:
In der files section kannst du natürlich auch anpassen .
Einfach die gewünschte Datei in der Navileiste anklicken, dann wird der Inhalt ins textfenster geladen.
Dann ändern und unten wieder auf den Speicherbutton klicken. 
also wirklich einfach

eckhard58, 2024-12-06:
Wo mache ich das examples?

eckhard58, 2024-12-08:
Zur Info:
in der neuen Release 3.3.0 werden in Raw Data keine Raw Data of ID-Data angezeigt.
Die alte Releases  3.2.2 tut es noch.

eckhard58, 2024-12-09:
Hast du nicht noch eine Idee zum den nicht vorhandenen Raw Data of ID-Data?

StefanNouza, 2024-12-09:
Moin !

Das Schlüsselwort für mich fiel letzte Woche: „fhem“. Diese Heimautomatisierung nutze ich auch (und gern), Probleme wie die hier beschriebenen hatte ich auch.
Für mich konnte ich das lösen und wollte es (neben einigen anderen Kleinigkeiten) hier auch gern an Tobias zurückgeben - nur so richtig Zeit hatte ich nicht gefunden und mit git(hub) kenn ich mich (noch) nicht aus. Ich hab halt das ZIP geladen (3.2.0) und lokal mit PlatformIO gearbeitet.
Deshalb hier meine Frage: kann mir einer von Euch ein kurzes Rezept geben, wie ich meine Anpassungen in dies Projekt einbringen kann?

Grüße,
  Stefan

eckhard58, 2024-12-09:
Hallo,
ich habe schon die Release 3.3.0, im Grunde funktioniert alles auch stabil, nur das mir die ID-Daten vom WR nicht mehr angezeigt werden.
Thobias hat seit kurzen auch einen Web Installer. Damit ist das flashen vom ESP32 recht einfach. (siehe aktuelle Wiki-Beschreibung.
Habe auch die Growatt-SPH.json mittlerweile etwas für meinen  WR (Live-Data) erweitert. Lokal nehme ich auch PlatformIO und git(hub) ist auch recht neu für mich.
Was hast du für einen WR? Siehst du die ID-Daten?

Gruß
Eckhard

StefanNouza, 2024-12-09:
Hallo Eckhard!

Das hatte ich vergessen, zu schreiben… tatsächlich auch einen Growatt, hier den SPH10000TL. Ich hab mich reichlich lang mit dessen Modbus-Schnittstelle auseinandergesetzt. Laut mehrerer Quellen soll die Spezifikation V1.20 am vollständigsten sein, bei mir passt die auf mein 2024er Gerät, ich verwende 40 Datenpunkte von den ca.70, die ich definiert habe - läuft super.
Auch den Datensatz wollte ich Tobias zukommen lassen.

Gruß,
  Stefan

eckhard58, 2024-12-09:
Ich habe beide Growatt .json(MOD und  SPH) benutzt und daraus meinen Datensatz erstellt. Mein WR hatte von beiden was und ich weiß noch nicht welche Spezifikation passt. Ich habe auch eine V1.24 und bin da noch am durchfitzen. Ist alles noch neu für mich.
Die Problemchen die ich hatte, liest du weiter oben.
Mein Sohn hat einen zwei Jahre alten Growatt SPH3000, meiner hat schon wieder andere Hardwareanschlüsse. Ging schon mit den RS485 PIN's los.

tobiasfaust, 2024-12-10:
> Hast du nicht noch eine Idee zum den nicht vorhandenen Raw Data of ID-Data?

Ich gehe mal davon aus das es einen Impact hierzu gibt: 
new feature: defining multiple requests for id-data (thanks to @Lazgar) (#89)

muss ich mir ansehen, ich bin gerade mit dem WebInstaller und OTA komplett beschäftigt.

tobiasfaust, 2024-12-10:
> Das Schlüsselwort für mich fiel letzte Woche: „fhem“. Diese Heimautomatisierung nutze ich auch (und gern), Probleme wie die hier beschriebenen hatte ich auch. Für mich konnte ich das lösen und [...]

Dazu muss man mit seinem eigenen Github-Account angemeldet sein, von meinem Development Branch einen Fork ziehen. Ein Fork ist praktisch nur eine Repository Kopie im eigenen account. Danach in seinem eigenen Account in dem Fork die Anpassungen machen. Anschließend über die eigene Github Website einen PullRequest erstellen.
Wichtig ist, das kein Spezial-Code der nur für das eigene Projekt funktioniert eingebaut wird. Es muss generalistisch für alle nutzbar sein.

Lazgar, 2024-12-10:
> > Hast du nicht noch eine Idee zum den nicht vorhandenen Raw Data of ID-Data?
> 
> Ich gehe mal davon aus das es einen Impact hierzu gibt: new feature: defining multiple requests for id-data (thanks to @Lazgar) (#89)
> 
> muss ich mir ansehen, ich bin gerade mit dem WebInstaller und OTA komplett beschäftigt.

Damit die ID-Daten wieder ausgelesen werden können, muss man die abzufragenden Register in ein Array packen. 

@tobiasfaust wenn du möchtest kann ich das gerne anpassen und pushen

eckhard58, 2024-12-10:
Hallo Lazgar und Tobias,
vielen Dank für die Rückmeldung.

Wie so ist das auslesen der ID-Daten Releases abhängig?
Was benötigst du dafür, nur die SGH.json File?
Oder was muss ich machen? Würde es ja auch selber können wollen. 
Scheint aber nicht so einfach zu sein, für mich als Anwender.
Das Problem ist zum Beispiel, das mein WR für die Inverter SerialNumber  die Position": [181 - 190] und nicht wie erwartet Position": [49 -58] benutzt.
Hab ich rausbekommen mit Verwendung der Files Growatt-MOD.json in der Realeases 3.2.2.
Leider komme ich mit den "Growatt Inverter Modbus RTU Protocol.pdf" die ich habe, im Zusammenhang mit dem Beispiel von Tobias, nicht klar.

Gruß
Eckhard

Lazgar, 2024-12-10:
@eckhard58 habe bereits einen Pull Request mit den Anpassungen eröffnet, hänge es hier mal an...
[Growatt-SPH.json](https://github.com/user-attachments/files/18079291/Growatt-SPH.json)

Das hat sich deshalb geändert weil man jetzt mehrere Register-Bereich für die ID-Daten angeben kann.
Bei meinem Solax sind auch viele unwichtige Daten bei der Abfrage dabei. Mit dem Array kann ich mehrere kleinere Bereich abfragen und verhindern dass mir der ESP wegen der Flut an Daten aussteigt...

Wenn die Register nicht zusammen müsstest du dir eventuell eine eigene Version bauen. Kann gut sein dass der Hersteller bei neueren oder älteren Version die Register anders aufbaut.

Können uns das ja mal gemeinsam ansehen.

tobiasfaust, 2024-12-10:
Habe den PR #99 gemerged. @Lazgar , vielen Dank dafür!
wenn ihr ein neues json baut, benennt den WR bitte genau damit andere User das json besser zuordnen können.

eckhard58, 2024-12-11:
Hallo Tobias,
leider hat das noch nicht geklappt. Die neue Growatt-SPH.json zeigt auch keine ID-Daten am. In der alten Releases funktionierte auch nur die Growatt-MOD_4000_TL3-XH.

eckhard58, 2024-12-11:
Hallo Lazgar,
werde ich machen, melde mich
Anfänger Frage, wie bekome ich den Screenshot hier rein?

Hallo Tobias,
nein kann nicht sein, ich habe die Ausgabedaten mit meine Sereinnummer verglichen.

Lazgar, 2024-12-11:
es gibt ein kleines Büroklammer Symbol, damit kann man eine Datei anhängen:
[image]

Das war keine Frage von Tobias, mit @tobiasfaust habe ich eine Frage an ihn gerichtet.

eckhard58, 2024-12-11:
Danke.
Das sieht besser aus,
nur das meine Position für die SN-Nummer die Positionen 181 - 190 hat.
[image]
[image]

Lazgar, 2024-12-11:
Hab hier eine neue Version... die wird auch nicht funktionieren, mir aber helfen es zu finden.
[Growatt-SPH-custom.json](https://github.com/user-attachments/files/18093797/Growatt-SPH-custom.json)

Bitte probier die mal aus und häng dein Raw Data hier wieder an + die Seriennummer die den WR hat.

eckhard58, 2024-12-11:
Sieht genau so aus.
Wie gesagt in der alten Releases war die MOD_4000 passender für meinen WR.
Da war meine SN-Nummer auf den Positionen 181 - 190 zu finden.

[image]

Lazgar, 2024-12-11:
Bitte unter Modbus Config -> Select Inverter Type den Eintrag "Growatt-SPH-custom" verwenden. Eigentlich müsste da mehr kommen bei RawData of ID-Data

eckhard58, 2024-12-11:
OK, wie mache ich das dann, wenn ich mehrere Files testen will?
Ändern lassen sich im ESP32 Files nicht, nur speichern unter anderm Namen.
Die  SN Nummer :  DTM4E5J00N          - sieht gut aus-
Und wie so jetzt ab Position 49 ?

[image]

eckhard58, 2024-12-11:
Und hier mal die Ausgaben vom alten Release 3.2.2  MOD_4000-TL3-XH

[image]

[image]

Lazgar, 2024-12-11:
Hm, eigentlich sollte es das sein laut Raw Data. Du kannst im json bei der Seriennummer mal die ,58 raus nehmen und schauen was passiert. Sieht für mich so aus als müsste man es um zwei Stellen kürzen (57 und 58)

eckhard58, 2024-12-11:
Haut leider nicht hin und so sieht das aus 

[image]

[image]

Lazgar, 2024-12-11:
[image]
Kannst du mal die Positionen 49 bis 58 eingeben und testen? Bei den Optionen aber String auswählen... Es sollte nur die Seriennummer angezeigt werden ohne exotische Zeichen...

eckhard58, 2024-12-11:
Passt

[image]

Lazgar, 2024-12-11:
Super, aber dann ist es somit kein Problem im json. Vielleicht hat Tobias eine Idee wo diese Zeichen herkommen können am weg in die weboberfläche.

eckhard58, 2024-12-12:
Sooo, ich hab mal das json-File für den Growatt SPH 3000 TL BL-UP Baujahr 2024 erweitert, soweit ich finden konnte. Alle Werte sind plausibel.
[Growatt-SPH5.json](https://github.com/user-attachments/files/18114390/Growatt-SPH5.json)
Könnte man bestimmt noch ausbauen. Leider habe ich keine brauchbare/aktuelle Growatt Modbus-Register Doku. Habt ihr ev. was brauchbares?

Der kleine Fehler ist natürlich geblieben. Wenn ich helfen kann sagt mir wie.
Bin kein Profi aber zum Testen brauchbar.

Lazgar, 2024-12-12:
Tobias hat geschrieben was zu tun wäre um den Fehler weiter einzugrenzen. Ich kann es leider nicht testen, da der Fehler bei mir nicht auftritt. Vielleicht können wir das gemeinsam, in einer gemeinsamen zoom Besprechung bewerkstelligen. andreas@markl.biz kannst gerne Kontakt aufnehmen. 😊

StefanNouza, 2024-12-14:
Moin!

Leider fehlt gerade die Zeit, hier komplett einzusteigen - aber meine lauffähige JSON häng ich hier mal dran. Sie läuft unter dem Projektstand 3.2.0 und einem SPH10000TL aus Ende 2023. Grundlage ist die Growatt-Doku zum Stand V1.24, die lt. anderer Foren von Growatt empfohlen wird - die PDF
 suche ich nochmal raus... Ich habe ca.70 Werte bereitgestellt, die bei mir auch funktionieren.

Mal sehen, ob es Euch hilft.

Gruß,
  Stefan
[Growatt-SPH-V124.json](https://github.com/user-attachments/files/18136116/Growatt-SPH-V124.json)

StefanNouza, 2024-12-14:
So, das ging schneller als gedacht… hier die Growatt-ModBus-Doku V1.24. Viel Spaß beim lesen und testen und Euch auch einen schönen 3.Advent.
[Modbus.RS485.RTU.Protocol.-1.2.4.pdf](https://github.com/user-attachments/files/18136148/Modbus.RS485.RTU.Protocol.-1.2.4.pdf)

eckhard58, 2024-12-14:
Hallo Steffan,

da ich hier in der Runde der einzige mit einem Growatt WR bin, werde ich deine Json mal ausprobieren.
Ich habe mehrere ESP32 zum testen.
Übers WE bitte aber etwas Geduld, wir haben die Enkelkinder zu besuch, da ist LEGO wichtiger ;-)

Gruß
Eckhard

eckhard58, 2024-12-15:
Moin Stefan,

habe gestern mal meine ESP´s mit deiner Json geladen, läuft unter der Relaises 3.2.2 stabil.
Unter der Relaises 3.3.0 ging fast nix. Der ESP meldet sich zwar aber schon beim öffnen der Item Config ging das Active setzen nicht mehr.
Ich würde deine Json gerne nachnutzen und vielleicht kann Torsten deine Version auf Relaise 3.3.0 zum laufen bringen wenn du nichts dagegen hast.
Es wäre eine gute Basis für alle Growatt SPH einphasig und dreiphasig.
Bei den Batterie Daten Temperatur und Spannung stimmen die Kommastellen nicht. 
Und ob die Batterie nur 2 bzw 20°C hat? 
Aber sonst alles super

Gruß Eckhard

Hier mal ein paar Bilder von meiner und deiner Json.
[image]
[image]

Lazgar, 2024-12-15:
@eckhard58 da sind wir wieder bei dem Thema dass ab Version 3.3.0 die Abfrage der ID-Daten ein Array angegeben werden muss...
Probier mal dieses json mit der Version 3.3.0, sollte dann laufen.
[Growatt-SPH-V124-MGW-3.3.0.json](https://github.com/user-attachments/files/18139497/Growatt-SPH-V124-MGW-3.3.0.json)

StefanNouza, 2024-12-15:
Moin!

Lego geht immer vor!! Aber danke, daß Du meine JSON mal testest.

Klar darf meine JSON verwendet werden, wir wollen ja alle dies Projekt voranbringen. Die Batterie-Items passen bei mir - da hat dann Growatt wohl etwas geändert, siehe meinen Screenshotausschnitt:

[image]

eckhard58, 2024-12-15:
Also auf der 3.3.0 läuft deine Version noch nicht, auch die von Lazgar nicht.
Siel läuft offensichtlich so lange bis man versucht LiveDaten anzuwählen, dann hängt sich der ESP auf.
Nur mit Spannungsreset kommt man wieder ran. und kann eine andere Json wählen.
UND man kann sehen, dass die Inverter SerialNummer diese ominösen Zeichen hat wie meine Json auch hat.

Was hast du für eine Batterie? meine hat grad mal um die 50V und denkst du das die 20,7°C echt sind?

StefanNouza, 2024-12-15:
20kWh stehen im Keller, ist ein deckenhoher Turm. Die Items highest/lowest Cell temperature sagen 17,6 und 20,7°C, ohne Sonne heute ist alles leer und nix bewegt sich im Akku. Daher sind 18°C am Boden und 21°C oben im Keller plausibel, passt auch zur Raumtemperaturmessung.
Eine Batteriegesamtspannung von ca.500V ist bei der Größe auch plausibel - imho wird die Spannung auch am Display des Growatt angezeigt.

eckhard58, 2024-12-15:
Ich habe 4,8kWh mit zwei US2000 C, die haben eine Nennspannung von 48V. Da bin ich erstaunt, dass du so hohe Spannung an der Batterie hast. 
Bei den Temperaturen, die Kommastelle verschoben und die WR-Temperatur hinzu gefügt.
So läuft deine Json in der Release 3.2.2 bei mir auch sicher. Super !!!
Schauen wir mal ob Lazgar oder Thomas was machen können für die Release 3.3.0.
Ich würde das dann testen.

StefanNouza, 2024-12-15:
wenn ich Zeit finde, werde ich mir die 3.3.0 ansehen, da ich in der 2.2.0 Verbesserungen vorgenommen hatte, die ich gern hier zurückgeben möchte - aber dazu muß ich sie selbst erstmal mit der 3.3.0 zusammenbringen.
In dem Zuge werde ich selbst sicher auch über das Problem mit meiner JSON stolpern…

Lazgar, 2024-12-16:
kannst du bitte mal die Verison probieren mit 3.3.0.
[Growatt-SPH-V124-MGW-3.3.0-Lazgar.json](https://github.com/user-attachments/files/18144984/Growatt-SPH-V124-MGW-3.3.0-Lazgar.json)

tobiasfaust, 2024-12-16:
@Lazgar , bist du sicher das die Positionen der InverterSN passen? Ungetestet, nur reingeschaut: du fragst die IDDaten von 0x00 bis 0x1E ab, das sind 15 Register - also 30 Werte. In der InverterSN ist aber Position 49-58 referenziert.....

[112 more replies]

closed 2025-04-14 by tobiasfaust