tobiasfaust/SolaxModbusGateway #100: OpenWb 2.0 mqtt topics change

saschoctm, 2024-12-12:
The new Version OpenWb 2.0 is already some time out now. The mqtt topics changed and the modbus gateway is not working.
It would be good to add this but also to stay compatible with current Version: 
- latest and currently working 1.9 
- New 2.0

tobiasfaust, 2024-12-12:
Can you please post the new topics?
Old topic -> new topic

saschoctm, 2024-12-12:
Here the topics
Old 1.9 -> New 2.0
Please try to stay compatible with 1.9 Version too.

XXX = Modul ID,  integer 0.....??? (see topic) example: openWB/set/bat/2/get/power
The ID number is assigned when the modules (devices) are set up, so it can always be different. I don't know what the maximum number is.
Battery and Inverter are 2 different modules, therefore this ID number must be able to be assigned to the battery and inverter separately by the user

---------------------------------
OpenWB Battery:
Battery Power
openWB/set/houseBattery/W -> openWB/set/bat/XXX/get/power
Speicherleistung in Watt, Zahl mit oder ohne Nachkommastelle (float, Integer) und einem Punkt als Dezimaltrennzeichen, positiv Ladung, negativ Ladung

Battery Capacity
openWB/set/houseBattery/%Soc -> openWB/set/bat/XXX/get/soc
Ladezustand des Speichers Zahl mit oder ohne Nachkommastelle (float, Integer) und einem Punkt als Dezimaltrennzeichen, gültige Werte 0 bis 100

Output Energy Charge
openWB/set/houseBattery/WhExported -> openWB/set/bat/XXX/get/exported
Entladene Energie in Wh Zahl mit oder ohne Nachkommastelle (float, Integer) und einem Punkt als Dezimaltrennzeichen, nur positiv

Input Energy Charge Erzeugte Energie
openWB/set/houseBattery/WhImported -> openWB/set/bat/XXX/get/imported
Geladene Energie in Wh Zahl mit oder ohne Nachkommastelle (float, Integer) und einem Punkt als Dezimaltrennzeichen, nur positiv

---------------------------------
OpenWB Inverter:

Power DC 2
openWB/set/pv/2/W -> openWB/set/pv/XXX/get/power
PV Leistung in Watt Zahl mit oder ohne Nachkommastelle (float, Integer) und einem Punkt als Dezimaltrennzeichen, Produzierte Leistung muss ein negatives Vorzeichen haben, In bestimmten Konstellationen wenn ein Hybridsystem über einen zweiten Wechselrichter geladen wird hat die Leistung ein positives Vorzeichen

Total Energy to Grid
openWB/set/pv/WhCounter -> openWB/set/pv/XXX/get/exported
Erzeugte Energie in Wh, Zahl mit oder ohne Nachkommastelle (float, Integer) und einem Punkt als Dezimaltrennzeichen, nur positiv

tobiasfaust, 2024-12-28:
there are 2 topics missing, 
topics to push the generated Watt of PV-String 1 (setpv1w) and PV-String 2 (setpv2w). Can you tell me those topic?
(setpv1w is the key which will refer in new inverter json files)

Thats the new mapping table:
<pre>
[
    {
        "version": "1.9",
        "topics": [
            { "setpvw": "openWB/set/pv/W" },
            { "setpv1w": "openWB/set/pv/1/W"},
            { "setpv2w": "openWB/set/pv/2/W" },
            { "setbatw": "openWB/set/houseBattery/W"},
            { "setbatsoc": "openWB/set/houseBattery/%Soc"},
            { "setbatexpwh": "openWB/set/houseBattery/WhExported"},
            { "setbatimpwh": "openWB/set/houseBattery/WhImported"},
            { "setcounterwh": "openWB/set/pv/WhCounter"},
            { "setpv1counterwh": "openWB/set/pv/1/WhCounter"},
            { "setpv2counterwh": "openWB/set/pv/2/WhCounter"}

        ] 
    },
    {
        "version": "2.0",
        "topics": [
            { "setpvw": "openWB/set/pv/#InverterID#/get/power" },
            { "setpv1w": ""},
            { "setpv2w": "" },
            { "setbatw": "openWB/set/bat/#BatteryID#/get/power"},
            { "setbatsoc": "openWB/set/bat/#BatteryID#/get/soc"},
            { "setbatexpwh": "openWB/set/bat/#BatteryID#/get/exported"},
            { "setbatimpwh": "openWB/set/bat/#BatteryID#/get/imported"},
            { "setcounterwh": " openWB/set/pv/#InverterID#/get/exported"}

        ] 
    }
]
</pre>

saschoctm, 2024-12-29:
I am not sure if this is correct at all.
Openwb 1.9/2 only receive 2 values for 1 inverter. The description I send in my last post was for the 2.0.

The 1.9 Version only was able to connect to 2 inverter in sum.
Therfore this topics: openWB/set/pv/1/W / openWB/set/pv/2/W.
This are not the strings of the inverter. This depends on how you setup the inverter at Pos 1 or Pos 2.
The Version 2.0 is completely different and you can set up multiple "moduls".

The Version I use from your ModbusGateway is from beginning of 2024 and I changed that time the register and still use it together with OWB 1.9.
I am using for example only openWB/set/pv/2/W, because the inverter I connect with MQTT is the 2nd on pos 2.
I  checked the register from the inverter and can confirm its the same value you use at current reg: openWB/set/pv/W.
Therefore I think this is not correct, can this work with Openwb 1.9? From my understanding not. They always use
openWB/set/pv/1/W / openWB/set/pv/2/W.

tobiasfaust, 2024-12-29:
ok, new adjusted version, i deleted everywhere setpv1w and setpv2w.
Please test. (modbus webpage -> OpenWB Support)
<pre>
[
    {
        "version": "1.9",
        "topics": [
            { "setpvw": "openWB/set/pv/#InverterID#/W" },
            { "setbatw": "openWB/set/houseBattery/W"},
            { "setbatsoc": "openWB/set/houseBattery/%Soc"},
            { "setbatexpwh": "openWB/set/houseBattery/WhExported"},
            { "setbatimpwh": "openWB/set/houseBattery/WhImported"},
            { "setcounterwh": "openWB/set/pv/#InverterID#/WhCounter"},
        ] 
    },
    {
        "version": "2.0",
        "topics": [
            { "setpvw": "openWB/set/pv/#InverterID#/get/power" },
            { "setbatw": "openWB/set/bat/#BatteryID#/get/power"},
            { "setbatsoc": "openWB/set/bat/#BatteryID#/get/soc"},
            { "setbatexpwh": "openWB/set/bat/#BatteryID#/get/exported"},
            { "setbatimpwh": "openWB/set/bat/#BatteryID#/get/imported"},
            { "setcounterwh": " openWB/set/pv/#InverterID#/get/exported"}

        ] 
    }
]
</pre>

saschoctm, 2024-12-29:
Is that build already available at the web installer? 
I have a few problems here with the wifi and that all newest build and stable versions not really working.
Maybe its my esp32 board version? Its a AZ-delivery ESP32 WROOM 32.
Its running since december/january with that old version from that time. So far no problems.

When I try the V3.2.2 what I think has still the wifi AP the AP didnt start. I cannot access or find.
The latest version always hangs/stuck mainly when i try to go to items config. Not always but most of the time. the 2 red circles always turning...
Also I cannot change the wifi settings with the web installer....only when i do fresh install than it workd but not changing wifi settings.

More or less I cannot test all new version. Already tested the last days a lot but cannot find the problems, already tried another board but same problems always.

Thats the board...
(https://www.az-delivery.de/products/esp32-d1-r32-board)

What board you recommend?

tobiasfaust, 2024-12-29:
I tested the last version with webinstaller and OTA with a standard ESP32. All of them works fine. I cannot confirm your problems. Maybe the FaQ wiki section can help.

saschoctm, 2024-12-30:
I checked the FAQ already, I tried to flash with the tool with the web installer and compiled by my own.
Also the very old FW from beginning this year what works fine for me I want to reflash but the AP doesnt came up.
Is it maybe the flash adresses changed?
I cant remember this settings/addresse from time before littlefs? Can this be the reason?
[image]

tobiasfaust, 2024-12-30:
Did you noticed that an AP is not using anymore? 
pls see: https://github.com/tobiasfaust/SolaxModbusGateway/wiki/start-and-integration-into-locale-wifi

and yes, thats a real ESP32

saschoctm, 2024-12-30:
Yes sure iI know, I am talking about a version from january 2024 what is still working at my setup, I want to flash it but the AP doesnt appear, so  was asking if this adresses maybe were different or can be the reason...
bootloader @1000
partition @8000
firmware @10000

tobiasfaust, 2024-12-30:
The adresses are the same.
But it’s recommended to use the web-installer. The new stable version is working well.

saschoctm, 2024-12-31:
So far I need to say the problem is not the board.
Whenever I boot with connected Inverter the Gateway will hang/stuck.
When the device is booted and I connect the inverter than it works fine. Any Idea?
The inverter is a Growatt SPH

About OpenWB:
At the moment I have just connected 1.9, but its not working, the gateway is connected to openwb that I can see with the MQTT Explorer but the values get not updated., I can not see the set topics only the connected gateway.

[image]

Another thing is, that the Total Energy is missing in the Growatt SPH, again you have it for both strings but not the total what is needed for OpenWB, here the change with registers, I added the first "TotalEnergyPV"

**{
  	  "position": [
            185,
            186,
            187,
            188
           ],
	   "name": "TotalEnergyPV",
	   "realname": "Erzeugte Energie PV",
	   "openwbtopic": "setcounterwh",
	   "datatype": "integer",
	   "factor": 100,
	   "unit": "Wh"
	},**
        {
          "position": [
            125,
            126,
            127,
            128
          ],
          "name": "TotalEnergyWhPv1",
          "realname": "Erzeugte Energie Pv1 in Wh",
          "datatype": "integer",
          "factor": 100,
          "unit": "Wh"
        },
        {
          "position": [
            125,
            126,
            127,
            128
          ],
          "name": "TotalEnergyKwhPv1",
          "realname": "Erzeugte Energie Pv1 in Kwh",
          "datatype": "float",
          "factor": 0.1,
          "unit": "KWh"
        },
        {
          "position": [
            133,
            134,
            135,
            136
          ],
          "name": "TotalEnergyWhPv2",
          "realname": "Erzeugte Energie Pv2 in Wh",
          "datatype": "integer",
          "factor": 100,
          "unit": "Wh"
        },
     {
          "position": [
            133,
            134,
            135,
            136
          ],
          "name": "TotalEnergyKwhPv2",
          "realname": "Erzeugte Energie Pv2 in Kwh",
          "datatype": "float",
          "factor": 0.1,
          "unit": "KWh"
        },

saschoctm, 2024-12-31:
Live Data Section also is empty

[image]

closed 2025-01-02 by tobiasfaust